margelo/react-native-vision-camera #2669: 💭 Short & Quick question about the documentation.

cropyai, 2024-03-20:
### Question

Hi, 

There is a section in [the format page of the documentation ](https://react-native-vision-camera.com/docs/guides/formats), 

[image]

Is `videoResolution: { width: 3048, height: 2160 }` correct while `videoAspectRatio: 16 / 9`? Is it supposed to be `3840`?

### VisionCamera Version

.

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Dingenis, 2024-04-04:
Hi @cropyai, it could be incorrect. It could also be intentional, since the filters are ordered by priority (descending) this would illustrate that the `videoAspectRatio` takes priority over the `videoResolution`.  So that it is clear, that it is not a guarantee that you get exactly what you want.

mrousavy, 2024-04-22:
Video Resolution doesn't matter as width is just multiplied by height, but aspect ratio does. I think this is correct because sensors are always in landscape orientation.

closed 2024-04-22 by mrousavy

cropyai, 2024-05-26:
@mrousavy what do you mean by "as width is just multiplied by height"? is it to find the number of pixels or smth? 

-> "I think this is correct because sensors are always in landscape orientation."
why is 3048x2160 correct? The default resolution for 16:9 is 3840x2160 as far as I know. I am just curious.

mrousavy, 2024-05-27:
> I am wondering where "3048" comes from instead of "3840"

From the Camera sensor. They don't have to have the same resolution as a TV does for 4k. It defines whatever resolution it can stream at.

> what do you mean by "as width is just multiplied by height"? is it to find the number of pixels or smth?

Yes, it just multiplies both numbers so the _order_ doesn't matter, but that's probably not what you asked for anyways.