margelo/react-native-vision-camera #2670: 💭Not supported in Expo Go!

fneufken, 2024-03-20:
### Question

I want to use vision camera to capture frames and run inference on them using Tensorflow Movenet on an Android tablet. To do this, I am using Expo Go to get the application on the tablet. However when starting the application, I get the following error: 

 ERROR  system/camera-module-not-found: react-native-vision-camera is not supported in Expo Go! Use EAS/expo prebuild instead (`expo run:android`). For more info, see https://docs.expo.dev/workflow/prebuild/., js engine: hermes
 ERROR  Invariant Violation: "main" has not been registered. This can happen if:
* Metro (the local dev server) is run from the wrong folder. Check if Metro is running, stop it and restart it in the current project.
* A module failed to load due to an error and `AppRegistry.registerComponent` wasn't called., js engine: hermes

### What I tried

I followed all the steps in the "Getting started" guide. I am currently working with expo@50.0.13 and react native version 0.73.5. There is nothing to find online on how to fix this. Hopefully you can help me!

### VisionCamera Version

3.9.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

jkaufman, 2024-03-20:
I would not expect React Native Vision Camera to work in Expo Go as it depends on native code.

jonjamz, 2024-03-21:
You need to use development builds instead of Expo Go.

https://docs.expo.dev/develop/development-builds/introduction/

ganeshmohane, 2024-09-18:
for detecting face, expo-face-detector i was using it but now it says it is depreacated and tellling to use react native vision camera but it is not supported in expo so i dont know what to do, is there other solution to detect face in expo app

mrousavy, 2024-09-23:
@ganeshmohane please read the above comments before posting.

dlogvin, 2024-10-08:
> @ganeshmohane please read the above comments before posting.

Hey! I basically am getting the same error message on a development build. 

It's very odd.

flikQ, 2024-10-18:
Only 'solution' i've found so I can continue working with Expo Go/Dev without interruption is to add the following to your top file, usually App.xxx, to suppress the error, still annoyingly appears in debugger etc tho.

`LogBox.ignoreLogs(['react-native-vision-camera is not supported in Expo Go!']);`

ganeshmohane, 2024-10-28:
> @ganeshmohane please read the above comments before posting.

yes

ganeshmohane, 2024-10-28:
> Only 'solution' i've found so I can continue working with Expo Go/Dev without interruption is to add the following to your top file, usually App.xxx, to suppress the error, still annoyingly appears [...]

is it working then?
are u able to use react native vision camera with expo?

ganeshmohane, 2024-10-28:
for face detection i am sending image to server and using deepface i am detecting face, it is not that good but works perfectly!!