margelo/react-native-vision-camera #2673: 💭 How to mirror a camera with external USB cameras

niallel, 2024-03-21:
### Question

Hi,

I'm using 2 external USB cameras, and I want one of them to be mirrored like the front camera.

Is there a way to do this with the library, or does anyone have a recommendation?

Thanks,

### VisionCamera Version

3.9.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

guilhermeshibuya, 2024-04-05:
Hi, were you able to detect and use the USB camera with the library? I tried once and I couldn't

niallel, 2024-04-07:
USB cameras works on some Android builds, but not on others - so I think it's down to if they included support for USB camera's in the Android build. This was on a Raspberry Pi by the way.

mrousavy, 2024-04-22:
Hey - it is not possible to access the Camera stream directly for such stuff, or at least extremely complex.

You could just mirror the view with `scaleX: -1`.

closed 2024-04-22 by mrousavy

niallel, 2024-04-22:
Thanks, much appreciated.

mrousavy, 2024-07-02:
Hey all! I just spent some time to build a new feature: I created a PR to add a `isMirrored` prop to Camera: https://github.com/mrousavy/react-native-vision-camera/pull/3036

With this prop you can enable or disable output mirroring 🎉🥳

If you appreciate my time and dedication towards building new features and constantly improving this library, please [**consider sponsoring me on GitHub**](https://github.com/sponsors/mrousavy/)  :)