margelo/react-native-vision-camera #2678: Pinch to zoom doesn't work after adding enableZoomGesture💭

Sfeeen, 2024-03-22:
### Question

I wanted to  add the option to zoom in with the camera, and added 
`enableZoomGesture={true}`. My camera setup now becomes:
`      {cameraLoaded && isFocused && (
          // <GestureDetector gesture={gesture}>
          <Camera
          device={device}
          enableZoomGesture={true}
          ref={myCamera}
          photo={true}
          isActive={cameraLoaded}
          style={{height: camera_aspect_ratio * screen.width, justifyContent: 'center'}}
          codeScanner={codeScanner}
          format={format}
          // onPress={_handleFocus}
          onError={handleCameraError}
          onInitialized={() => {
              console.log("Camera initiliazed")
              setIsCameraReady(true)
            }
          }

        />
          // </GestureDetector>
      )}`

But when I pinch to zoom  nothing happens, I guess  I'm overseeing something?

### What I tried

I went from the v3.9.1 to beta version v4.0.0 (zoom seemed not to work in both)
I also tested on 2 android devices (2 different types of samsung). 
Since I  seem  the only  one with this problem I asume the fault is in my code, what checks can I do to further diagnose?

### VisionCamera Version

v4.0.0-beta.7

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

Sfeeen, 2024-03-26:
Update: I used another RN camera library that supports zoom and the issue remained. I don't know why this is but I guess its related to that I'm using a navigation drawer which makes use of gestures and maybe also overrides the pinch gesture. 

I don't have a solution for now

ritathesaver, 2024-03-28:
how does your gesture look like?

Sfeeen, 2024-03-29:
The gesture of my drawer navigation is a swipe from left to right as can be seen in the animation on [this page](https://reactnavigation.org/docs/drawer-based-navigation/). The gesture I perform for the zoom is pinch to zoom, but I tried several possible movements though. I guess the problem lays in the native function 'SimpleOnScaleGestureListener' that maybe gets overriden by another library or gets block by something.

mrousavy, 2024-04-02:
Maybe you have a view above your Camera that needs something like `pointerEvents="none"` to correctly let touches through.

Either way, not a VisionCamera bug 👍

Sfeeen, 2024-04-09:
Indeed that seemed to be the issue and `pointerEvents="none"` was the solution!