margelo/react-native-vision-camera #2680: How to run a simple rendering test using jest ? 💭

HND-dev, 2024-03-22:
### Question

Hello everyone, I apologize in advance for my English and the fact that I could possibly ask a question that may seem simple to most of you.
My question is: how do I launch a rendering test with jest using the VisionCamera library?

Im trying to run a simple test like this:
```jsx
/**
 * @format
 */

import 'react-native';
import React from 'react';
import App from '../App';

// Note: import explicitly to use the types shipped with jest.
import {it} from '@jest/globals';
import {jest} from '@jest/globals';
// Note: test renderer must be required after react-native.
import renderer from 'react-test-renderer';

it('renders correctly', () => {
  renderer.create(<App />);
});
```

And in my App component, I render another component that uses Camera of VisonCamera : 
```jsx
...
import {VisionCamera} from '../vision-camera/vision-camera';

import {
  useCameraDevice,
  useCameraPermission,
  useCodeScanner,
} from 'react-native-vision-camera';

const ScanPage = () => { // this component is after imported and used in my  App component 
  const navigation = useNavigation<StackNavigationProp<any>>();
  const {hasPermission, requestPermission} = useCameraPermission();
  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: codes => {
      console.log(`Scanned ${codes.length} codes!`);
      for (const code of codes) {
        console.log(code.value);
      }
    },
  });
  const device = useCameraDevice('back');

  if (device == null)
    return (
      <View style={styles.camera}>
        <Text> No device found</Text>
      </View>
    );
  return (
    <VisionCamera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
      codeScanner={codeScanner}
    />
  );
};
```

Now when i run my test i'm getting this error :
```bash
export RN_SRC_EXT=e2e.js &&  npm run test                                                                                                                                                                                      ─╯

> MyTestApp@0.0.1 test
> jest

 FAIL  __tests__/App.test.tsx
  ● Test suite failed to run

    system/camera-module-not-found: Failed to initialize VisionCamera: The native Camera Module (`NativeModules.CameraView`) could not be found.
    * Make sure react-native-vision-camera is correctly autolinked (run `npx react-native config` to verify)
    * Make sure you ran `pod install` in the ios/ directory.
    * Make sure you rebuilt the app.

      1 | // vision-camera.js
      2 |
    > 3 | import { Camera, sortDevices } from 'react-native-vision-camera'
        | ^
      4 |
      5 | export const VisionCamera = Camera
      6 |

      at Object.<anonymous> (node_modules/react-native-vision-camera/src/NativeCameraModule.ts:41:9)
      at Object.require (node_modules/react-native-vision-camera/src/Camera.tsx:7:1)
      at Object.require (node_modules/react-native-vision-camera/src/index.ts:1:1)
      at Object.require (vision-camera/vision-camera.js:3:1)
      at Object.<anonymous> (Pages/ScanPage.tsx:11:1)
      at Object.<anonymous> (App.tsx:19:1)
      at Object.<anonymous> (__tests__/App.test.tsx:7:1)

Test Suites: 1 failed, 1 total
Tests:       0 total
Snapshots:   0 total
Time:        3.708 s
Ran all test suites.
```

I'm running this on Kubuntu 22.04

### What I tried

I tried to follow the instructions [Mocking Guide](https://react-native-vision-camera.com/docs/guides/mocking) of VisonCamera but nothing worked for me.

### VisionCamera Version

3.9.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

closed 2024-04-22 by mrousavy

ydlooming, 2024-06-11:
Why is this closed? I am struggling to find a solution

andreszib, 2024-12-26:
I came across this question while searching if there was a Jest mock available for this library (there isn't), so I figured I'd answer your question in case anyone else runs into this.

The error you are getting is very common when trying to render Native Modules using Jest. This is because Jest runs in a Node env and cannot execute native code. You'll find that the only solution is to mock the library, like so:

```js
jest.mock('react-native-vision-camera', () => ({
  Camera: jest.fn(() => 'Camera'),
  useCameraDevice: jest.fn(),
  useCameraPermission: () => ({
    hasPermission: true,
    requestPermission: jest.fn(),
  }),
  useCodeScanner: jest.fn(),
}));
```

That's just a simple mock that will prevent the error from happening. However, if you want to test the code scanned callback, you'll need to create a mock Camera component, which should allow you to trigger the event during the test.