margelo/react-native-vision-camera #2681: 🔧 App not run when i install react native vision camera

ZiaImdad, 2024-03-24:
### How were you trying to build the app?

i am using react native cli...

### Full build logs

```tsx
PS D:\Reactnative projects\testcameraapp> npm run android

> testcameraapp@0.0.1 android
> react-native run-android

info A dev server is already running for this project on port 8081.
info Installing the app...

> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core found, Frame Processors enabled!
5 actionable tasks: 5 up-to-date

info 💡 Tip: Make sure that you have set up your development environment correctly, by running npx react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor 

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':react-native-vision-camera'.
> [CXX1101] NDK at C:\Users\ziazia\AppData\Local\Android\Sdk\ndk\25.1.8937393 did not have a source.properties file

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 12s
error Failed to install the app. Command failed with exit code 1: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081 FAILURE: Build failed with an exception. * What went wrong: A problem occurred configuring project ':react-native-vision-camera'. > [CXX1101] NDK at C:\Users\ziazia\AppData\Local\Android\Sdk\ndk\25.1.8937393 did not have a source.properties file * Try: > Run with --stacktrace option to get the stack trace. > Run with --info or --debug option to get more log output. > Run with --scan to get full insights. > Get more help at https://help.gradle.org. BUILD FAILED in 12s.
info Run CLI with --verbose flag for more details.
PS D:\Reactnative projects\testcameraapp>
```

### Project dependencies

```json
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.73.6",
    "react-native-vision-camera": "^3.9.1",
    "react-native-worklets-core": "^0.4.0"
  },
```

### VisionCamera Version

ex. 3.9.1

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ZiaImdad, 2024-03-24:
please help me and give me solution about it

CHAOSdonut, 2024-03-27:
The exception states there is something wrong in your Android SDK, and you cant run the example app either.

Are you sure that your android dev environment is set up correctly? [react native environment setup](https://reactnative.dev/docs/environment-setup)
Are you able to run the default app using android?

DianaMGalindo, 2024-04-01:
@ZiaImdad I got the same error, and I still haven't been able to solve it: 

`* What went wrong:
A problem occurred configuring project ':react-native-vision-camera'.
 [CXX1101] NDK at C:\Users\Diana\AppData\Local\Android\Sdk\ndk\25.1.8937393 did not have a source.properties file`

I have looked at my ndk folder, and well, it doesn't have a source.properties file but there isn't clear documentation on how to add this file if needed. I thought it was generated once the react native project was created.

This is my current SDK status, yet I can't get to build the app. Help ! :`-( 

ext {
        buildToolsVersion = "34.0.0"
        minSdkVersion = 26
        compileSdkVersion = 34
        targetSdkVersion = 34
        ndkVersion = "25.1.8937393"
        kotlinVersion = "1.8.0"
    }

ZiaImdad, 2024-04-02:
Change your NDK version try with 23 version

DianaMGalindo, 2024-04-02:
@ZiaImdad well, it seems that at least I have moved from that error, to face the next one: 
`* What went wrong:
Execution failed for task ':react-native-vision-camera:configureCMakeDebug[arm64-v8a]'.
> [CXX1428] exception while building Json A problem occurred starting process 'command 'C:\Users\Diana\AppData\Local\Android\Sdk\cmake\3.22.1\bin\cmake.exe''`

Are you using a specific version of CMake? Did you face the same error?