margelo/react-native-vision-camera #2682: 🐛 Static background audio noise in the recorded video.

prasan2421, 2024-03-24:
### What's happening?

The video recorded has a static background audio noise. Compared to the iphone camera app and there are no noise. 

### Reproduceable Code

```tsx
<ReanimatedCamera
                        ref={cameraRef}
                        device={device}
                        fps={fps}
                        isActive={isActive}
                        format={format}
                        onError={onError}
                        enableZoomGesture={true}
                        animatedProps={animatedProps}
                        orientation="portrait"
                        video={true}
                        audio={true}>
                      </ReanimatedCamera>
```

### Relevant log output

```shell
{}
```

### Camera Device

```json
{}
```

### Device

iphone 11 pro max (ios 17.3.1)

### VisionCamera Version

3.9.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

mcshiz, 2025-04-04:
I've noticed this issue as well, sounds like the recorded audio is staticy / underwater.
`"react-native-vision-camera": "^4.6.4",`