margelo/react-native-vision-camera #2686: 🐛 takePhoto() generate a fully green image - An unknown error occurred while creating the Camera Session, but the Camera can recover from it

Kohys58, 2024-03-25:
### What's happening?

### Question

I'm using react-native-vision-camera: 4.0.0-beta.10, react-native: 0.72.6 (Target SDK has been upgraded to 34).

**Device is : Samsung TAB A8 (Android 14)**

My code is based on the one provided in the example App (pretty much a copy paste).

Just removed the qualityPrioritization has it not available in the method anymore.

### What would be the issue?

When I try to take the picture it gives me the following error (in landscape view) : 

[session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]

This doesn't show when being in portrait, but in both case it resultats in generating a fully green image instead of the correct picture :

[image]

### What I tried
Tried to enable precapture on Android

### Reproduceable Code

```tsx
const photo = await camera.current.takePhoto({
    flash: flash,
    enableShutterSound: false,
});
```

### Relevant log output

```shell
2024-03-25 15:00:41.875 17079-17079 ViewRootIm...nActivity] com.ag_consulting                    I  ViewPostIme pointer 0
2024-03-25 15:00:41.897 17079-17388 ReactNativeJS           com.ag_consulting                    D  state: BEGAN
2024-03-25 15:00:41.957 17079-17079 ViewRootIm...nActivity] com.ag_consulting                    I  ViewPostIme pointer 1
2024-03-25 15:00:41.965 17079-17079 unknown:ReactNative     com.ag_consulting                    W  Can't cancel already finished gesture. Is a child View trying to start a gesture from an UP/CANCEL event?
2024-03-25 15:00:41.975 17079-17388 ReactNativeJS           com.ag_consulting                    D  state: ACTIVE
2024-03-25 15:00:41.976 17079-17388 ReactNativeJS           com.ag_consulting                    D  state: END
2024-03-25 15:00:41.976 17079-17388 ReactNativeJS           com.ag_consulting                    I  Taking photo...
2024-03-25 15:00:41.981 17079-17079 CameraView              com.ag_consulting                    D  Finding view 2463...
2024-03-25 15:00:41.981 17079-17079 CameraView              com.ag_consulting                    D  Found view 2463!
2024-03-25 15:00:41.982 17079-20690 CameraView.takePhoto    com.ag_consulting                    I  Taking photo... Options: {enableShutterSound=false, flash=off}
2024-03-25 15:00:41.986 17079-17407 UseCaseAttachState      com.ag_consulting                    D  Active and attached use case: [androidx.camera.core.ImageCapture-23d35658-291b-4022-a2df-9a459c0dbc6a11304672, androidx.camera.core.Preview-381b6c7e-3dad-4070-a988-be4f0c93d17d148722915] for camera: 0
2024-03-25 15:00:41.990 17079-17079 ImageCapture            com.ag_consulting                    D  takePictureInternal
2024-03-25 15:00:41.991 17079-17079 CameraOrientationUtil   com.ag_consulting                    D  getRelativeImageRotation: destRotationDegrees=0, sourceRotationDegrees=90, isOppositeFacing=true, result=90
2024-03-25 15:00:41.991 17079-17079 TakePictureManager      com.ag_consulting                    D  Issue the next TakePictureRequest.
2024-03-25 15:00:41.995 17079-17407 CaptureSession          com.ag_consulting                    D  Attempting to submit CaptureRequest after setting
2024-03-25 15:00:41.995 17079-17407 CaptureSession          com.ag_consulting                    D  Issuing request for session.
2024-03-25 15:00:41.995 17079-17407 Camera2Cap...estBuilder com.ag_consulting                    D  createCaptureRequest
2024-03-25 15:00:42.159 17079-17407 ZslControlImpl          com.ag_consulting                    E  dequeueImageFromBuffer no such element
2024-03-25 15:00:42.160 17079-17407 Camera2CameraImpl       com.ag_consulting                    D  {Camera@8c68ed9[id=0]} Issue capture request
2024-03-25 15:00:42.160 17079-17407 CaptureSession          com.ag_consulting                    D  Issuing capture request.
2024-03-25 15:00:42.163 17079-17407 Camera2Cap...estBuilder com.ag_consulting                    D  createCaptureRequest
2024-03-25 15:00:42.193 17079-17094 gralloc4                com.ag_consulting                    I  register: id=410000001a8, handle:0xb4000073c4004e00, vaddr=0x72ea57d000
2024-03-25 15:00:42.298 17079-20693 TrafficStats            com.ag_consulting                    D  tagSocket(122) with statsTag=0xffffffff, statsUid=-1
2024-03-25 15:00:42.458 17079-17079 CameraView              com.ag_consulting                    I  invokeOnShutter(PHOTO)
2024-03-25 15:00:42.501 17079-17079 gralloc4                com.ag_consulting                    W  Could not find component description for FourCC value 0, alloc_format 0x21
2024-03-25 15:00:42.501 17079-17079 gralloc4                com.ag_consulting                    I  lockBuffer: id=410000001a8, vaddr=0x72ea57d000
2024-03-25 15:00:42.510 17079-17405 DeferrableSurface       com.ag_consulting                    D  use count-1,  useCount=1 closed=false androidx.camera.core.SurfaceRequest$2@d2d9437
2024-03-25 15:00:42.511 17079-17405 DeferrableSurface       com.ag_consulting                    D  use count-1,  useCount=0 closed=false androidx.camera.core.impl.ImmediateSurface@516f709
2024-03-25 15:00:42.511 17079-17405 DeferrableSurface       com.ag_consulting                    D  Surface no longer in use[total_surfaces=3, used_surfaces=1](androidx.camera.core.impl.ImmediateSurface@516f709}
2024-03-25 15:00:42.512 17079-17079 TakePictureManager      com.ag_consulting                    D  Issue the next TakePictureRequest.
2024-03-25 15:00:42.512 17079-17079 TakePictureManager      com.ag_consulting                    D  No new request.
2024-03-25 15:00:42.513 17079-17405 Camera2CameraImpl       com.ag_consulting                    D  {Camera@8c68ed9[id=0]} Camera receive onErrorCallback
2024-03-25 15:00:42.514 17079-17405 Camera2CameraImpl       com.ag_consulting                    D  CameraDevice.onError(): 0 failed with ERROR_CAMERA_DEVICE while in OPENED state. Will attempt recovering from error.
[... 272 lines cut ...]
  "hasTorch": false,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung TAB A8 (Android 14)

### VisionCamera Version

4.0.0-beta.10

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-03-27:
Did you try to reproduce that in the example app on the `v4` branch?

robertyulisman, 2024-03-28:
i get an issue to on xiamoy and realmi device, the screen si black and showing this error 

[image]

working on another devices

this is my code 

```
const camera = React.useRef<Camera>(null);
  const isFocused = useIsFocused();
  const appState = useAppState();
  const [isCameraActive, setIsCameraActive] = React.useState(false);
  const { hasPermission, requestPermission } = useCameraPermission();

  const [deviceDir, setDeviceDir] = React.useState<CameraPosition>('front');
  const devices = Camera.getAvailableCameraDevices();
  const device = devices.find((d) => d.position === deviceDir);

React.useEffect(() => {
    if (hasPermission) {
      if (isFocused && appState === 'active' && showCamera) {
        setIsCameraActive(true);
      }
    }
    requestPermission().then((value) => {
      if (value === true) {
        setIsCameraActive(true);
      }
    });
  }, []);

  return (
    <SafeAreaView style={StyleSheet.absoluteFill}>
      <StatusBar translucent backgroundColor={COLORS.dark[500]} />
{hasPermission && device ? (
        <React.Fragment>
          <Camera
            ref={camera}
            key={device.id}
            style={StyleSheet.absoluteFill}
            device={device}
            isActive={isCameraActive}
            photo
            onError={(err) => {
              console.log('err', err.message);
              Alert.alert('Error Camera', err.message);
            }}
          />

          <LayerCameraSvg />

          <TouchableOpacity
            onPress={closeCamera}
            style={{
              backgroundColor: COLORS.light[50],
              position: 'absolute',
              top: 50,
              right: 20,
              zIndex: 999,
              width: 30,
              height: 30,
              borderRadius: 15,
              alignItems: 'center',
              justifyContent: 'center',
            }}
          >
            <Icon width={24} height={24} name={ICONS.close} />
          </TouchableOpacity>
          <View
            style={{
              zIndex: 999,
              position: 'absolute',
              bottom: 10,
              alignItems: 'center',
              alignSelf: 'center',
            }}
          >
            <Text style={{ color: COLORS.light[50], textAlign: 'center' }} variant="heading-2">
              Sejajarkan Wajahmu
            </Text>

            <TextBody
              title="Sejajarkan wajah sesuai pola diatas dan ambil foto selfiemu"
              style={{
                color: COLORS.light[50],
                textAlign: 'center',
                width: '70%',
                marginVertical: 10,
              }}
            />
            <TouchableOpacity
              onPress={handleTakeCamera}
              // onPress={onRandomZoomPress}
              style={{ marginVertical: 20, marginTop: 20 }}
            >
              <Image
                source={require('@asset/images/shutter.png')}
                style={{ width: 80, height: 80 }}
              />
            </TouchableOpacity>
            <TouchableOpacity
              onPress={handleSwitchCamera}
              // onPress={onRandomZoomPress}
              style={{
                marginVertical: 20,
                marginTop: 40,
                position: 'absolute',
                bottom: 20,
                right: width / 6,
              }}
            >
              <Image source={ICONS.switch} style={{ width: 25, height: 25 }} />
            </TouchableOpacity>
          </View>
        </React.Fragment>
      ) : (
        <View style={{ flex: 1, alignItems: 'center', justifyContent: 'center' }}>
          <StatusBar barStyle="dark-content" />
          <Text
            variant="body-1"
            style={{
              color: COLORS.dark['500'],
            }}
          >
            No camera device or permission
          </Text>
        </View>
      )}
 </SafeAreaView>
  );
```

ChristopherGabba, 2024-04-06:
+1 I'm getting this error on Google Pixel 3 just simply pressing a button that issues a takePhoto()

```json
    "errorType": "CameraRuntimeError",
    "title": "session/recoverable-error",
    "message": "An unknown error occurred while creating the Camera Session, but the Camera can recover from it.",
```

Doing nothing fancy just like example app:
Capture Button:
```tsx
  const takePhoto = useCallback(async () => {
    try {
      if (camera.current == null) throw new Error("Camera ref is null!")

      const photo = await camera.current.takePhoto(takePhotoOptions)
      onMediaCaptured(photo, "photo")
    } catch (error) {
      reportCrash({ error, method: "takePhoto", component: "CaptureButton" })
    }
  }, [camera, onMediaCaptured, takePhotoOptions])
```

CameraScreen:
```tsx
const onMediaCaptured = useCallback(
    async (media: PhotoFile | VideoFile, type: "photo" | "video") => {
      console.log(`Media captured! ${JSON.stringify(media)}`)
    },
    [],
  )
```

It's getting thrown from the Camera's `onError` prop as a Runtime Error. 
My screen is not green, it actually freezes and looks like a photo was taken but the app freezes at that point.

mrousavy, 2024-04-09:
will push a new beta soon hopefully that fixes this

ChristopherGabba, 2024-04-09:
Thanks @mrousavy. I did post a bug request here: https://issuetracker.google.com/issues/332350593 per your request. They did send me a demo test app apk to try out, which I plan on doing when I get off work. https://github.com/mrousavy/react-native-vision-camera/issues/2577#issuecomment-2031851613

ChristopherGabba, 2024-04-10:
@mrousavy I tested the camera freezing issue on the test app that they provide and it seems to work fine on that app. That sort of rules out the issue being with the package.

SveBo, 2024-04-19:
I can reproduce this issue on my Samsung Tab A8  with the 4.0.0beta15 aswell.
Does someone knows where the issue lies / where to investigate ?

mrousavy, 2024-04-20:
@SveBo @ChristopherGabba you can just play around with the settings used in `configureOutputs(..)` in `CameraSession.kt`.

Try to tweak those values, rebuild, and see if that changes anything.

SveBo, 2024-04-22:
@mrousavy In my case (Samsung Tab A8) updating CameraX to 1.4.0-alpha05 fixed the problem for me. This seems to be a bug with some Samsung devices generating too large images.

mrousavy, 2024-04-22:
Could you build alpha05? I couldn't build it and get errors..

SveBo, 2024-04-22:
I just changed the react-native-vision-camera/android/build.gradle to "def camerax_version = "1.4.0-alpha05", 
deleted the android build and .gradle folder, 
invalidated caches in Android Studio.

It did build for me and swapping the camerax_version seems to fix the camera issue.

mrousavy, 2024-04-22:
hm, weird - the CI also fails to build: https://github.com/mrousavy/react-native-vision-camera/pull/2779

SveBo, 2024-04-22:
Mhm.. I have to check later with the example... but maybe it is the Gradle Version? 
The pipeline is using 7.5.1, Im using 8.3.
(Im not too familiar with the Android pipeline, so I could totally be wrong)

SveBo, 2024-04-24:
This seems to work for me #2788 2788

SveBo, 2024-04-24:
On an interesting side note, I am not sure yet how relevant this is:

On camerax1.4.0-alpha04 the error / green image only occures if there is no frameprocessor set in the react part.
So this could either be the CameraSelector in vision-camera or camerax is doing something with ImageAnalysis that im not sure of

(Samsung is working fine on 1.4.9-alpha 5, but I got feedback that Lenovo Tab P11  is not working, but I can not confirm that reliable)

ldalzottomp, 2024-04-25:
I have the same issue on a Galaxy Tab A8. At first I thought that the issue was with devices that don't have flash. But some devices with flash were mentioned here.

SveBo, 2024-04-26:
@mrousavy  I found the fix for the Lenovo Tab 11 now as well. 

The VisionCamera prop has
 _photoQualityBalance="balance" (CAPTURE_MODE_ZERO_SHUTTER_LAG) as default,_
 but this causes the issue even if frameprocessor / video are bound.

CameraX says this falls back  to CAPTURE_MODE_MINIZE_LATENCY if the device doesn't support ZSL. This does not seem to work correctly in this case.

mrousavy, 2024-04-26:
Yea just set it to quality or speed instead until Google fixes this bug in CameraX

SveBo, 2024-04-26:
Thanks again for the V4 release, its way more stable now on Android 👍

mrousavy, 2024-04-26:
great to hear 💪

ShamuXX, 2024-04-26:
I have the same problem, I was on version 3.6 but I had the streched view camera error, I tried updating to v4 to resolve it but now I have a new problem with the camera view, it turns a green background and after a few seconds the camera freezes, this is shown in the logcats:

 E  'Camera.onError(session/recoverable-error): An unknown error occurred while creating the Camera Session, but the Camera can recover from it.', { [session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]
name: 'session/recoverable-error',
_code: 'session/recoverable-error',
_message: 'An unknown error occurred while creating the Camera Session, but the Camera can recover from it.', _cause: undefined }

[image]

dragonmaster784, 2024-05-17:
Current devices I have noticed experiencing this are:
Samsung Galaxy A05s
Samsung Galaxy A03 Core
Oppo model CPH2197
Samsung Galaxy Grand Primo
Samsung Galaxy J4 Core

Some get a black screen, other's get the green screen. I will start debugging where/why they are receiving "recoverable" error.

dragonmaster784, 2024-05-20:
> @mrousavy I found the fix for the Lenovo Tab 11 now as well.
>
> The VisionCamera prop has _photoQualityBalance="balance" (CAPTURE_MODE_ZERO_SHUTTER_LAG) as default,_ but this causes the issue even if [...]

I can confirm that setting the photoQualityBalance prop to "speed" instead of its default value of "balanced", has helped resolve the black and green screens reported on the above devices 👍

mrousavy, 2024-05-20:
@ChristopherGabba @dragonmaster784 could you do me a favor and report that to the Google CameraX team? It's clearly a bug - the doc of the ZSL option (which is what "balanced" is essentially) say that it will fall-back to not use ZSL, but it doesn't apparently - and fails when used in combination with video or ImageAnalysis.

Would be great if you could either tell them that in a new issue report or in the one @ChristopherGabba already created.

Thanks!

SveBo, 2024-05-20:
I also posted an issue report, but it seems like, they can not reproduce it?
Sadly I do not have a device where the error occures myself (maybe its because we need to use it in combination with Video or ImageAnalysis?)

https://issuetracker.google.com/issues/337214687

dragonmaster784, 2024-05-20:
Absolutely @mrousavy 👍  I have a reproducible device, so I'll issue a ticket to the CameraX team with a full report with the stack trace and logs once I clock off here 🫡

mrousavy, 2024-05-21:
Hey - just created a PR to temporarily fix this - can you test if that works for you? https://github.com/mrousavy/react-native-vision-camera/pull/2897 🙏

closed 2024-05-21 by mrousavy

ldalzottomp, 2024-05-23:
I am not reproducing this issue on a device that was previously showcasing it (Samsung galaxy tab A8) using the version 4.0.5 👍
Thanks for the fix