margelo/react-native-vision-camera #2688: 🐛 Not bound to a valid Camera

bglgwyng, 2024-03-25:
### What's happening?

I got this error from Sentry logs and I don't know what was happening. I'll add more info as soon as I get it.

### Reproduceable Code

```tsx
// not yet producible
```

### Relevant log output

```shell
unknown/unknown: Not bound to a valid Camera [ImageCapture:androidx.camera.core.ImageCapture-066850f5-1d8f-4143-99ab-4111b5410e08]
  at construct(native)
  at apply(native)
  at _construct(node_modules/@babel/runtime/helpers/construct.js:4:65)
  at Wrapper(node_modules/@babel/runtime/helpers/wrapNativeSuper.js:17:23)
  at construct(native)
  at _callSuper(app:///index.android.bundle:225:2)
  at constructor(node_modules/react-native-vision-camera/src/CameraError.ts:150:5)
  at construct(native)
  at _callSuper(app:///index.android.bundle:225:2)
  at <global>(node_modules/react-native-vision-camera/src/CameraError.ts:175:32)
  at tryParseNativeCameraError(node_modules/react-native-vision-camera/src/CameraError.ts:215:36)
  at takePhoto(node_modules/react-native-vision-camera/src/Camera.tsx:138:38)
  at throw(native)
  at asyncGeneratorStep(node_modules/@babel/runtime/helpers/asyncToGenerator.js:3:24)
  at _throw(node_modules/@babel/runtime/helpers/asyncToGenerator.js:25:27)
  at tryCallOne(node_modules/promise/setimmediate/core.js:37:14)
  at setImmediate$argument_0(node_modules/promise/setimmediate/core.js:123:25)
  at apply(native)
  at queueReactNativeMicrotask(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:247:23)
  at _callTimer(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:111:15)
  at _callReactNativeMicrotasksPass(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:161:15)
  at callReactNativeMicrotasks(node_modules/react-native/Libraries/Core/Timers/JSTimers.js:415:42)
  at __callReactNativeMicrotasks(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:393:44)
  at __guard$argument_0(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:132:39)
  at __guard(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:368:11)
  at flushedQueue(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:131:17)
  at invokeCallbackAndReturnFlushedQueue(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:127:29)
```

### Camera Device

```json
// I'll get it soon
```

### Device

SM-A326B, Android 13

### VisionCamera Version

4.0.8-beta

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bglgwyng, 2024-04-02:
No, I wasn't able to reproduce it yet.

mrousavy, 2024-04-22:
hey - I think this has been fixed in 4.0.0 :)

closed 2024-04-22 by mrousavy