margelo/react-native-vision-camera #2690: 🐛 Preview stretched / zoomed-in on android

mlecoq, 2024-03-25:
### What's happening?

I'm creating a new issue with the same topic as this one : https://github.com/mrousavy/react-native-vision-camera/issues/2364

because on samsung device, taken picture is still zoomed in / stretched in my case (ratio 1 and 0.625)
 
I have issues with square and the ratio used in the example app.

I have made a small example to have side by side preview and taken picture (preview on the left)

With back camera

[image]

With front camera 

[image]

### Reproduceable Code

```tsx
Here is a repo to reproduce the issue 

https://github.com/mlecoq/VisionCameraPreviewIssue
```

### Relevant log output

```shell
1142  1142 E cdsprpcd: vendor/qcom/proprietary/adsprpc/src/fastrpc_apps_user.c:2707: Error 0x200: open_dev (-1) failed for domain 3 (errno Connection refused)
03-25 17:13:45.447  1142  1142 E cdsprpcd: vendor/qcom/proprietary/adsprpc/src/fastrpc_apps_user.c:1215: Error 0x200: remote_handle_open failed for adsp_default_listener (errno Success)
03-25 17:13:45.447  1142  1142 I cdsprpcd: vendor/qcom/proprietary/adsprpc/src/rpcmem_android.c:191: rpcmem_deinit_internal: closed ION fd 5
03-25 17:13:45.448  1142  1142 I cdsprpcd: vendor/qcom/proprietary/adsprpc/src/fastrpc_apps_user.c:2756: fastrpc_apps_user_deinit done
03-25 17:13:45.449  1142  1142 E cdsprpcd: vendor/qcom/proprietary/adsprpc/src/cdsprpcd.c:47:cdsp daemon will restart after 100ms...
03-25 17:13:45.454   823  1388 E CamX    : [ERROR][STATS_AF ] af_alg_util.cpp:480: af_alg_util_get_fovc_factor() [Jake] pSettings = 0xe776f984
03-25 17:13:45.466  1184  1184 E cnss-daemon: Failed to get sock name Socket operation on non-socket
03-25 17:13:45.474   823  1396 I CHI     : [SS_INFO]: [CHI         ]: chxusecase.cpp: ExecuteCaptureRequest: 2401: [CaptureRequest] stream = 0xeae66dd8 streamType = 0 streamFormat = 0x22 streamWidth = 1280 streamHeight = 960
03-25 17:13:45.474   823  1396 E CHI     : [SS_ERR ]: [CHI         ]: chxusecase.cpp: ExecuteCaptureRequest: 2447: ExecuteCaptureRequest: pMetaData is NULL
03-25 17:13:45.474   823  1396 I CHI     : [SS_INFO]: [CHI         ]: chxusecase.cpp: ExecuteCaptureRequest: 2456: Default usecase capture request for frame 22016 with 1 output buffers
03-25 17:13:45.487  1184  1184 E cnss-daemon: Failed to get sock name Socket operation on non-socket
03-25 17:13:45.506  1184  1184 E cnss-daemon: Failed to get sock name Socket operation on non-socket
03-25 17:13:45.527 22427 25346 I CameraView.takePhoto: Successfully captured 2976 x 2976 photo!
03-25 17:13:45.529 22427  2538 I ReactNativeJS: /data/user/0/com.visioncamerapreviewissue/cache/mrousavy-9182797906249792109.jpg
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy Tab S5e | SM-T720

### VisionCamera Version

4.0.0-beta.10

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mlecoq, 2024-03-26:
I think it's more visible because I'm using a tablet but I also see small differences on my redmi

ezebecke, 2024-03-28:
I'm also experiencing this issue while testing 4.0.0-beta.11 on Galaxy A01

rodgomesc, 2024-03-28:
can you guys reproduce this on the example app in v4 branch?

mlecoq, 2024-03-30:
> can you guys reproduce this on the example app in v4 branch?

I have just tested with example app. First I didn't reproduce the issue (even if I have a small preview with a ratio of 0.625), but then I have tried to have a similar display as my example : preview and result on the same screen. And now I reproduce the issue, particularly when I remove frame Processor 

Here is the CameraPage I use in the example app on v4 branch to reproduce the issue:

https://gist.github.com/mlecoq/94ffbfb4803cddea58860e379d9a5eb6.js

mlecoq, 2024-04-02:
Maybe I am not using correctly react-native-vision-camera but everything is fine on iOS so I was expecting a similar behavior on android

ezebecke, 2024-04-04:
Sorry you all. After more testing, I realized I was setting the wrong `photoAspectRatio`.

mrousavy, 2024-04-04:
@mlecoq how about you - are you using different photo sizes?

mlecoq, 2024-04-04:
@mrousavy in the example I made, I use as aspectRatio the same one as used to display preview and taken picture

In the example of `react-native-vision-camera` I have seen that used aspect-ratio depends on screen size, so I have tried with it (see gist here https://github.com/mrousavy/react-native-vision-camera/issues/2690#issuecomment-2028490665) but I have still the issue

mlecoq, 2024-04-04:
The capture is good, on the preview I have an elongated face

[image]

mrousavy, 2024-04-04:
Yea because photo size doesn't have anything to do with the video stream.

I think what you're seeing is all expected.

mlecoq, 2024-04-05:
Do you mean that there is an issue on how I use <Camera/> ? Because I don't see this preview issue on iOS with the same code and on my Redmi neither.

mlecoq, 2024-04-05:
what also surprise me (but there is maybe a technical limitation), is that given `photoAspectRatio` is not applied on taking a picture.

For example I have set a ratio of 0.625 (like preview in the capture below) and the taken picture has a width of 720 and a height of 1280, so a ratio of 0.5625 (it's why we see black stripes on the captures image)

So what you see with preview is not what you get. For the moment I mitigate this by cropping the image (to have expected ratio) but the result is zoomed in.

[image]

mlecoq, 2024-04-05:
when I use screen aspect ratio, the result looks worse (taken picture has a width of 720 and a height of 1600)

[image]

mlecoq, 2024-04-08:
I am wondering if this https://developer.android.com/media/camera/camerax/configuration?hl=en#crop could be used to solve this issue

ashfaq14060, 2024-04-09:
Any solution about this issue?

TatvaKarnika, 2024-04-10:
Hi everyone,
I'm also facing the same preview stretched issue on stock android device, for other device and for ios it is working fine.
[image]
Did you found any solution?

TatvaKarnika, 2024-04-10:
Switching to beta version as mention in [https://github.com/mrousavy/react-native-vision-camera/issues/2583] solved my issue.

mrousavy, 2024-04-10:
> Switching to beta version solved my issue.

you're welcome 👍

n-ii-ma, 2024-04-15:
Try this [hack](https://github.com/mrousavy/react-native-vision-camera/issues/2583#issuecomment-2056645167) if you don't wanna install the beta version.

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy