margelo/react-native-vision-camera #2693: ✨ chunked recording

fahadyaqub, 2024-03-26:
### What feature or enhancement are you suggesting?

To progressively save or upload chunks of audio/video while recording happens, instead of one big file at the end of recording.

This could work by specifying an interval and a callback, upon which the camera returns the recorded video file/data up to that time, while the recording continues.

What i have in mind is similar to Javascript's MediaRecorder, where you can specify a "timeslice" in MediaRecorder:start method - which triggers the "dataavailable" event on every tick of the timeslice.

https://developer.mozilla.org/en-US/docs/Web/API/MediaRecorder/start

A similar functionality is Mediarecorder's requestData function, which also triggers "dataavailable" event, which can be called at an interval.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

A close parallel to this functionality can be achieved by stoping and starting recording. But I assume this will drop some frames every time start/stop happens. 

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-03-27:
Hey - I think this is gonna be much more complicated than it sounds.

Can't you just start and stop new recordings each time? That's essentially the same as chunking.

fahadyaqub, 2024-03-28:
Yes, thats what i currently thought of doing, but i am pretty sure this might drop frames between a stop and restart of recording - in the merged video file.

Unfortunately, there is no easy way for me to verify if this is actually happens (a frame or 2 were missed out). 

Since you build this plugin, you would have a better idea. :)

I am pretty new to your library, so excuse me if i sound stupid, but i had another idea, which is to use the frame processor functionality?

Basically, push each frame (references) into an array in useFrameProcessor hook, and when the interval reaches, save the current frames to a file, and reset array.

You think this is a viable way of achieving this?

mrousavy, 2024-03-29:
Yea I think Frame Processors are also a good solution for this! I believe @coofzilla built something like this (not opensource) and it works fine!

fahadyaqub, 2024-04-01:
Thanks @mrousavy @coofzilla,
I was hoping of reusing the uploader from web; using arrayBuffers from frames. The major concern is encoding. On web the blobs come pre-encoded in the specific codecs and bitrates. Im guessing i might have to use react native ffmpeg plugin for this.

coofzilla, 2024-04-01:
@fahadyaqub so, I didn't use ffmpeg, I was able to do the encoding/bitrate/etc natively in `ios` and its quite straight forward. 

so `AVAssetWriter` and `AVAssetWriterInput` are gonna be your bread and butter. But when you're configuring the output settings, you can just set it to `hevc` and it'll work.

Something like this:

```swift
    videoWriterInput = AVAssetWriterInput(mediaType: .video, outputSettings: configuration.videoCompressionSettings)
```

```swift
  var videoCompressionSettings: [String: Any] = [
    AVVideoCodecKey: AVVideoCodecType.hevc,
    AVVideoCompressionPropertiesKey: [AVVideoAverageBitRateKey: 6000000]
  ]
 ```
 
 Now in my case, I wanted separated video/audio; so, if you want to do that, you'll need two AVAssetWriters, one for audio and one for video.

mrousavy, 2024-04-02:
Yea I think this can be closed then, as I don't plan on adding chunked recordings to VisionCamera. That should be a frame processor plugin. Good luck!

closed 2024-04-02 by mrousavy

colonelpanic8, 2024-06-26:
> Hey - I think this is gonna be much more complicated than it sounds.
> 
> Can't you just start and stop new recordings each time? That's essentially the same as chunking.

... no its not at all, you will almost certainly drop frames this way.

I have a version of react-native-vision-camera that does chunked recording that lives here

https://dev.railbird.ai/railbird/react-native-vision-camera

It does have a few major downsides, which are:

- its more a of a proof of concept that works for our use case and we have gutted certain things
- because of the switch to the camera api with version 4 here, its no longer possible to merge with this repo
- There is no ios implementation

@coofzilla I'm not really easily able to do ios dev, but it would be really helpful for me to see/maybe use your ios implementation of chunked recording. Do you think you could share that with me?

colonelpanic8, 2024-06-26:
@fahadyaqub Also interested to hear where you ended up, and would be happy to at least get you running with my version of react-native-vision-camera on the android side of things.