margelo/react-native-vision-camera #2696: Hi Im using react-native:0.73.6 after capturing photo or video if I try to close the view I have: FATAL EXCEPTION: mrousavy/VisionCamera.main 💭

Mohamed-Ziyad, 2024-03-26:
### Question

this is the error thrown from android logcat
[image]

do you have a patch for this? 

### VisionCamera Version

3.9.1

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-03-27:
Enable `photo={true}`.

mrousavy, 2024-03-27:
Or `video={true}`

closed 2024-03-27 by mrousavy