margelo/react-native-vision-camera #2699: 🔧 New version "react-native-vision-camera": "^3.9.2" always showing error

vinod1988, 2024-03-27:
### How were you trying to build the app?

Here is version details 

 "react": "18.1.0",
 "react-native": "0.70.14",
"react-native-vision-camera": "^3.9.2",

`
const device = useCameraDevice("back");
 <Camera
              style={StyleSheet.absoluteFill}
              device={device}
              isActive={true}
            />
`

### Full build logs

```tsx
Type 'CameraDevice | undefined' is not assignable to type 'CameraDevice'.
  Type 'undefined' is not assignable to type 'CameraDevice'.ts(2322)
CameraProps.d.ts(31, 5): The expected type comes from property 'device' which is declared here on type 'IntrinsicAttributes & IntrinsicClassAttributes<Camera> & Readonly<CameraProps>'
(property) device: CameraDevice
The Camera Device to use.

See the Camera Devices section in the documentation for more information about Camera Devices.

@example

const device = useCameraDevice('back')

if (device == null) return <NoCameraErrorView />
return (
  <Camera
    device={device}
    isActive={true}
    style={StyleSheet.absoluteFill}
  />
)
```

### Project dependencies

```json
"dependencies":{
 "react": "18.1.0",
 "react-native": "0.70.14",
"react-native-vision-camera": "^3.9.2",
}
```

### VisionCamera Version

3.9.2

### Target platforms

iOS, Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-03-31 by vinod1988

Arsenalist, 2024-09-27:
How was this resolved @vinod1988? I'm getting the same thing using Expo

FedeTommi, 2025-01-08:
I have the same issue as @Arsenalist