margelo/react-native-vision-camera #2704: Execution failed for task ':react-native-vision-camera:compileDebugKotlin'🔧

vinod1988, 2024-03-31:
### How were you trying to build the app?

I tried to build on android simulator and real device but its not working. always giving a build error. Please suggest me whats going wrong with android version 

iOS is working smoothly. 

### Full build logs

```tsx
Duplicate class com.facebook.yoga.YogaPositionType found in modules jetified-react-android-0.74.0-rc-debug.5-debug-runtime (com.facebook.react:react-android:0.74.0-rc.5) and jetified-react-native-0.70-debug.14-debug-runtime (com.facebook.react:react-native:0.70.14)
     Duplicate class com.facebook.yoga.YogaProps found in modules jetified-react-android-0.74.0-rc-debug.5-debug-runtime (com.facebook.react:react-android:0.74.0-rc.5) and jetified-react-native-0.70-debug.14-debug-runtime (com.facebook.react:react-native:0.70.14)
     Duplicate class com.facebook.yoga.YogaStyleInputs found in modules jetified-react-android-0.74.0-rc-debug.5-debug-runtime (com.facebook.react:react-android:0.74.0-rc.5) and jetified-react-native-0.70-debug.14-debug-runtime (com.facebook.react:react-native:0.70.14)
     Duplicate class com.facebook.yoga.YogaUnit found in modules jetified-react-android-0.74.0-rc-debug.5-debug-runtime (com.facebook.react:react-android:0.74.0-rc.5) and jetified-react-native-0.70-debug.14-debug-runtime (com.facebook.react:react-native:0.70.14)
     Duplicate class com.facebook.yoga.YogaValue found in modules jetified-react-android-0.74.0-rc-debug.5-debug-runtime (com.facebook.react:react-android:0.74.0-rc.5) and jetified-react-native-0.70-debug.14-debug-runtime (com.facebook.react:react-native:0.70.14)
     Duplicate class com.facebook.yoga.YogaValue$1 found in modules jetified-react-android-0.74.0-rc-debug.5-debug-runtime (com.facebook.react:react-android:0.74.0-rc.5) and jetified-react-native-0.70-debug.14-debug-runtime (com.facebook.react:react-native:0.70.14)
     Duplicate class com.facebook.yoga.YogaWrap found in modules jetified-react-android-0.74.0-rc-debug.5-debug-runtime (com.facebook.react:react-android:0.74.0-rc.5) and jetified-react-native-0.70-debug.14-debug-runtime (com.facebook.react:react-native:0.70.14)
     
     Go to the documentation to learn how to <a href="d.android.com/r/tools/classpath-sync-errors">Fix dependency resolution errors</a>.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 2s

    at makeError (/Users/vinod/workspace/vino/sk/Vinod/source_code/Vinod/node_modules/execa/index.js:174:9)
    at /Users/vinod/workspace/vino/sk/Vinod/source_code/Vinod/node_modules/execa/index.js:278:16
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async runOnAllDevices (/Users/vinod/workspace/vino/sk/Vinod/source_code/Vinod/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:109:5)
    at async Command.handleAction (/Users/vinod/workspace/vino/sk/Vinod/source_code/Vinod/node_modules/@react-native-community/cli/build/index.js:142:9)
```

### Project dependencies

```json
buildToolsVersion = "31.0.0"
minSdkVersion = 24
compileSdk = 33
targetSdkVersion = 33

kotlinVersion = "1.7.0"

"dependencies": {
    "react": "18.1.0",
    "react-native": "0.70.14", 
    "react-native-reanimated": "^3.7.0",  
    "react-native-vision-camera": "^3.9.2",
}
```

### VisionCamera Version

3.9.2

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-02:
This doesn't look like a VisionCamera error, but rather a react-native issue. Clean your cache and try again. Also, maybe you need to upgrade to RN 0.71.

closed 2024-04-02 by mrousavy

vinod1988, 2024-04-04:
@mrousavy Which version is supported for RN 0.70.14.

Can you please tell the react native vision camera version compatibility for RN 0.70.14

mrousavy, 2024-04-04:
I don't remember - maybe an old V3 version. Either way, I'd recommend you to upgrade to RN 0.71 either way

vinod1988, 2024-04-07:
@mrousavy 
I am getting the below error when using the RN 0.71 and camera vision version 3.0.2

 Duplicate class com.facebook.yoga.YogaValue$1 found in modules jetified-react-android-0.74.0-rc-debug.6-debug-runtime (com.facebook.react:react-android:0.74.0-rc.6) and jetified-react-native-0.70-debug.14-debug-runtime (com.facebook.react:react-native:0.70.14)

     Duplicate class com.facebook.yoga.YogaWrap found in modules jetified-react-android-0.74.0-rc-debug.6-debug-runtime (com.facebook.react:react-android:0.74.0-rc.6) and jetified-react-native-0.70-debug.14-debug-runtime (com.facebook.react:react-native:0.70.14)

  Go to the documentation to learn how to <a href="d.android.com/r/tools/classpath-sync-errors">Fix dependency resolution errors</a>.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 5m 37s