margelo/react-native-vision-camera #2706: 🐛 photoResolution: 'max' not the same as native camera photo.

erssykes, 2024-04-01:
### What's happening?

I wanted to take a picture using react-native-vision-camera that has the same resolution as the native device's camera app.
Currently, I have been experimenting with photoResolution: 'max' and was expecting to see the same resolution as the native camera.  
Currently, the react-native-vision-camera produces photos:  12MP, 3024x4032, whereas the device camera takes photos at 24MP, 4284 x 5712 (both are using the 'main camera').  I had thought the photoResolution: 'max' would force it to use the same high resolution as the native device's camera.  
Device is:  iPhone 15 max

Any insight on how to force it to use the maximum resolution available?

Thank you so much,

### Reproduceable Code

```tsx
const device = useCameraDevice('back');
const format = useCameraFormat(device, [
    { photoResolution: 'max'},  // use max resolution possible.
    { photoHdr: true }  // High Dynamic Range
]);

<Camera
        ref={camera}
        style={StyleSheet.absoluteFill}
        device={device!}
        format={format}
        enableHighQualityPhotos={true}  
        isActive={isActive}
        onTouchStart={handleFocus}
        enableZoomGesture
        photo
      />

const takePhoto = useCallback(async () => {
    const photo = await camera.current?.takePhoto({
      qualityPrioritization: 'quality',  
      flash: 'auto',
      enableShutterSound: true
    });
```

### Relevant log output

```shell
N/A
```

### Camera Device

```json
"id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "position": "back",
  "hardwareLevel": "full",
  "minFocusDistance": 20,
  "hasTorch": true,
  "sensorOrientation": "landscape-right",
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "maxZoom": 123.75,
  "supportsRawCapture": false,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "neutralZoom": 1,
  "minZoom": 1,
  "isMultiCam": false,
  "formats": [],
  "hasFlash": true,
  "name": "Back Camera",
  "minExposure": -8
}
```

### Device

iPhone 15 pro max

### VisionCamera Version

3.9.2 (v4)

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-04-02 by mrousavy

erssykes, 2024-04-03:
hi, 

Thank you for your help.  So, I've experimented with the different physical devices ('ultra-wide-angle-camera', 'wide-angle-camera', 'telephoto-camera' and various multi-cam configurations.  Still can't seem to capture pictures that are of the same high-resolution as a native device's camera.  Here's a brief breakdown:  

Using an iPhone 15 pro max, the example app in 'react-native-vision-camera iOS example', the result is:   8MP, 2160x3840 
Using an iPhone 15 pro max, my react-native-vision-camera app and 
 const device = useCameraDevice('back', {
    physicalDevices: ['wide-angle-camera']
  });
the result is: 12MP, 3024 x 4032

Using an iPhone 15 pro max, my react-native-vision-camera app and 
 const device = useCameraDevice('back', {
    physicalDevices: ['ultra-wide-angle-camera', 'telephoto-camera']
  });
the result is: 12MP, 3024 x 4032

Using an iPhone 15 pro max, my react-native-vision-camera app and 
 const device = useCameraDevice('back', {
    physicalDevices: ['ultra-wide-angle-camera', 'wide-angle-camera', 'telephoto-camera']
  });
the result is:  12MP, 3024x4032

Using an iPhone 15 pro max, my react-native-vision-camera app and 
 const device = useCameraDevice('back', {
     physicalDevices: ['ultra-wide-angle-camera', 'wide-angle-camera']
  });
the result is:  12MP, 3024x4032

Using the iPhone 15 pro max's camera:
the result is: 24MP, 5712 x 4284

[The scene was exactly the same for all of the above picture taking]

So, the question is, how can I configure react-native-vision-camera to use the same high-resolution as the default camera (i.e, 24MP, 5712 x 4284)?

Thank you so much for your help

kind regards

mrousavy, 2024-04-04:
I think the default Camera shoots ProRAW or HEIF or something where it can do higher resolution. VisionCamera doesn't have support for that yet (see https://github.com/mrousavy/react-native-vision-camera/issues/1851)

mrousavy, 2024-04-04:
Uhm.. did you read the issue I linked ?

It is currently a feature request I'm raising money for to fund it - I don't plan on developing this for free in my free time

erssykes, 2024-04-05:
Could you explain how the process works, please?

mrousavy, 2024-04-08:
People donate some funds, and once the pool is full I will implement it.

erssykes, 2024-04-08:
how long does it generally take for the funds to reach the appropriate level?  how long will it take to implement the feature?  what if more funds are needed (ie., it is more complex than originally assessed)?

mrousavy, 2024-04-08:
I don't know, I don't have a lot of experience with that.
But generally people don't like paying money for open source, so it's hard to raise funds.

If it takes longer, no more money will be requested, that's my risk.