margelo/react-native-vision-camera #2709: 💭 minimum required version for React Native

maimai886, 2024-04-03:
### Question

hello everyone. 
If possible, I'd like to ask a question here. 
What is the minimum required version for React Native? 
It seems that my project with React Native 61.2 cannot use this library.
Would it be possible to use it by downgrading the VisionCamera version? 
Thank you all.

### VisionCamera Version

3.9.2

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

athaljen, 2024-04-11:
As they have no minimum react/react-native requirement in package.json, also the initial version i.e 1.0.0-alpha.1 it had 
   "react": "16.13.1",
    "react-native": "0.63.4",
  as dev devDependencies so it should work on 0.61.2

mrousavy, 2024-04-22:
Hey - I think the min version is react-native 0.71 right now.

closed 2024-04-22 by mrousavy

nolife08021, 2024-06-13:
@mrousavy  is it possible to run it with 0.70? would like to use vision-camera to having face detection