margelo/react-native-vision-camera #2711: 💭 Failed to initialize VisionCamera: VisionCamera currently does not work on web.

ajesses, 2024-04-03:
### Question

[image]
when open on the web. There is a error! The Error: Failed to initialize VisionCamera: VisionCamera currently does not work on web. 

It appears to initialize VisionCamera in the web environment as well.

How to resolve, or How to do without initializing VisionCamera in a web environment?

### What I tried

[image]
I try to use Platform API to resolve it, but it don't work

### VisionCamera Version

^3.9.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

willsunnn, 2024-04-18:
I'm also facing this issue, and currently the only workaround I can think of is using JS dynamic imports to try catch loading the module. 

If the package could at least provide a No-op VisionCamera implementation for web that returns null devices or fails on call rather than on import, that could be a potential temporary solution.

mrousavy, 2024-04-21:
As the error message states, VisionCamera does not work on Web.

I am not using Web myself, so I've never seen that error before.

For Web, you should use web camera APIs instead.

closed 2024-04-21 by mrousavy

ishikawa, 2024-07-11:
We also encountered the same issue, and we resolved it by applying patches to `react-native-vision-camera` and `react-native-worklets-core` as follows. Please refer to the following for your reference.

`react-native-vision-camera+4.3.2.patch`

```diff
diff --git a/node_modules/react-native-vision-camera/lib/commonjs/index.web.js b/node_modules/react-native-vision-camera/lib/commonjs/index.web.js

new file mode 100644

index 0000000..b92811b

--- /dev/null

+++ b/node_modules/react-native-vision-camera/lib/commonjs/index.web.js

@@ -0,0 +1,5 @@

+'use strict';

+

+Object.defineProperty(exports, '__esModule', {

+ value: true,

+});

diff --git a/node_modules/react-native-vision-camera/lib/module/index.web.js b/node_modules/react-native-vision-camera/lib/module/index.web.js

new file mode 100644

index 0000000..e69de29
```

`react-native-worklets-core+1.3.3.patch`

```diff
diff --git a/node_modules/react-native-worklets-core/lib/commonjs/index.web.js b/node_modules/react-native-worklets-core/lib/commonjs/index.web.js

new file mode 100644

index 0000000..db7129b

--- /dev/null

+++ b/node_modules/react-native-worklets-core/lib/commonjs/index.web.js

@@ -0,0 +1,5 @@

+"use strict";

+

+Object.defineProperty(exports, "__esModule", {

+ value: true,

+});

diff --git a/node_modules/react-native-worklets-core/lib/module/index.web.js b/node_modules/react-native-worklets-core/lib/module/index.web.js

new file mode 100644

index 0000000..e69de29
```

a-eid, 2025-12-14:
@mrousavy why does it throw an error even thou I am not trying to use it on web.
I have 2 files 

camera.ts // native with vision camera
camera.web.ts // web without vision camera.

and I still see this error.