margelo/react-native-vision-camera #2712: React Native vision camera freezes on Android 14 with frameProcessor

Ch-Anas-Abdullah, 2024-04-03:
### What's happening?

### Issue Description:
I am facing an issue with the React Native vision camera freezing when using frameProcessor={frameProcessor} on Android 14. The camera works fine on Android 12 and below, but on Android 14, it freezes when the frameProcessor is added.

### Steps to Reproduce:

Use the following setup:
React Native version: ^0.73.6
React Native Vision Camera version: ^3.9.0
React Native Reanimated version: ^3.8.0
React Native Vision Camera Face Detector version: ^1.4.0
React Native Worklets Core version: 0.3.0
Add the frameProcessor={frameProcessor} prop to the Camera component.
Run the application on an Android 14 device.

### Expected Behavior:
The camera should function normally with the frameProcessor added, as it does on Android 12 and below.

### Actual Behavior:
The camera freezes when the frameProcessor is added on Android 14

### Reproduceable Code

```tsx
import { Worklets } from 'react-native-reanimated';

const handleFacesDetection = Worklets.createRunInJsFn((result) => {
  if (result.faces.length !== 0) {
    console.log('detection result', result)
    console.log('face bounds', result.faces[0].bounds)
    console.log('face landmarks', result.faces[0].landmarks)
  }
})

const frameProcessor = useFrameProcessor((frame) => {
  'worklet'
  detectFaces(
    frame,
    handleFacesDetection, {
      landmarkMode: "all",
    }
  )
}, [handleFacesDetection])

<Camera
  style={{ width: 320, height: 320, borderRadius: 10 }}
  ref={cameraRef}
  isActive={true}
  device={device}
  frameProcessor={frameProcessor} // Comment out this line to see the camera working
  onError={(error) => console.error(error)}
/>
```

### Relevant log output

```shell
2024-04-03 14:49:38.313 17051-32352 TrafficStats            com.tsg.swivx                        D  tagSocket(133) with statsTag=0x90000, statsUid=-1
2024-04-03 14:49:39.821 17051-17051 PreviewView             com.tsg.swivx                        I  Creating PreviewView...
2024-04-03 14:49:39.822 17051-17051 PreviewView             com.tsg.swivx                        I  PreviewView is 0x0, rendering 1080x1920 content (LANDSCAPE_LEFT). Resizing to: 1080x1920 (COVER)
2024-04-03 14:49:39.822 17051-17051 CameraView              com.tsg.swivx                        I  Updating CameraSession...
2024-04-03 14:49:39.823 17051-17164 CameraSession           com.tsg.swivx                        I  configure { ... }: Waiting for lock...
2024-04-03 14:49:39.823 17051-17164 CameraSession           com.tsg.swivx                        I  configure { ... }: Updating CameraSession Configuration... Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
2024-04-03 14:49:39.823 17051-17164 CameraSession           com.tsg.swivx                        I  Configuring inputs for CameraSession...
2024-04-03 14:49:39.823 17051-17164 Persistent...ureSession com.tsg.swivx                        D  --> setInput(1)
2024-04-03 14:49:39.823 17051-17164 CameraSession           com.tsg.swivx                        I  Destroying previous outputs...
2024-04-03 14:49:39.824 17051-17164 CameraSession           com.tsg.swivx                        I  Creating outputs for Camera #1...
2024-04-03 14:49:39.829 17051-17051 ScrollerOp...ionManager com.tsg.swivx                        D  registerConfigChangedListener
2024-04-03 14:49:39.832 17051-17164 CameraSession           com.tsg.swivx                        I  Adding 1920x1440 Video Output in YUV_420_888...
[... 164 lines cut ...]
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Xiaomi Redmi Note 13

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Timer00, 2024-04-05:
Facing the same problem here (Poco F3 Android 13) ! I believe it might be this [same issue](https://github.com/mrousavy/react-native-vision-camera/issues/2293).

mrousavy, 2024-04-09:
Can you try if VisionCamera V4 fixed the issue?

Timer00, 2024-04-09:
> Can you try if VisionCamera V4 fixed the issue?

Good stuff, it works!

mrousavy, 2024-04-21:
Great that V4 fixes this, and thanks for the sponsorship @Timer00 !

closed 2024-04-21 by mrousavy