margelo/react-native-vision-camera #2713: 🐛  [V4] Android GPS location cache issue

idrisssakhi, 2024-04-03:
### What's happening?

The GPS coordinates are fetched from cache on Android, so we need to disable the cache when streaming location on Android. (Screenshot included, the GPSTimeStamp 07:53, image was taken at 11:20)
The heading value (GPSImageDirection), are not available on android.

### Reproduceable Code

```tsx
import { ReactElement, useCallback, useEffect, useMemo, useRef, useState } from 'react';
import { ImageRequireSource, Pressable, StyleSheet, View, ViewStyle } from 'react-native';

import { faCameraRotate } from '@fortawesome/free-solid-svg-icons/faCameraRotate';
import { faMoon } from '@fortawesome/free-solid-svg-icons/faMoon';
import { FontAwesomeIcon } from '@fortawesome/react-native-fontawesome';
import { writeAsync } from '@lodev09/react-native-exify';
import { GeolocationResponse } from '@react-native-community/geolocation';
import CompassHeading from 'react-native-compass-heading';
import Reanimated, {
  interpolate,
  useAnimatedProps,
  useAnimatedRef,
  useAnimatedStyle,
  useSharedValue,
  withSpring,
  withTiming,
} from 'react-native-reanimated';

import { useSafeAreaInsets } from 'react-native-safe-area-context';
import {
  Camera,
  CameraPosition,
  CameraProps,
  PhotoFile,
  TakePhotoOptions,
  useCameraDevice,
  useCameraDevices,
  useCameraFormat,
} from 'react-native-vision-camera';

import HDRLogo from 'assets/icons/camera-hdr/hdr.svg';

import { filePrefix, isAndroid } from 'src/Common/Constants';
import { usePrevious } from 'src/Common/Hooks';
import { Color, style as styleUtils } from 'src/Common/Types';

import { IconButton } from '../Buttons';
import { FullScreenLoader } from '../Loaders';

export interface CameraViewProps {
  onTakePhoto: (photo: PhotoFile, photoPosition?: GeolocationResponse) => void;
  isActive?: boolean;
  allowFrontCamera?: boolean;
  enableZoomGesture?: boolean;
}

const flashIcon = new Map<TakePhotoOptions['flash'], ImageRequireSource>([
  ['auto', require('assets/icons/camera-flash-auto-icon/camera-flash-auto-icon.png')],
  ['on', require('assets/icons/camera-flash-on-icon/camera-flash-on-icon.png')],
  ['off', require('assets/icons/camera-flash-off-icon/camera-flash-off-icon.png')],
]);

/**
 * Returns next value for flash mode based on current, passed as argument
 *
 * @param flashMode - Current flash mode
 */
const getNextFlashMode = (flashMode: TakePhotoOptions['flash']): TakePhotoOptions['flash'] => {
  const flashModesOrder = Array.from(flashIcon.keys());
  const currentIndex = flashModesOrder.findIndex((mode) => mode === flashMode);
  if (currentIndex > -1) {
[... 358 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

OPPO A17

### VisionCamera Version

4.0.0-beta.11

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-09:
Well it does start updating location, but the initial value is set to the cached value. It should give an update within the first 5 seconds of running, does that never happen? Or is 5 seconds just too slow for your usecase?

idrisssakhi, 2024-04-09:
yes, it does update the location after a while, didn't measure the time. Any way, in my use case I need to get the real actual location. I used exify for android to write metadata directly. if it's a lot of time to improve we can close this, i think there are more important issues than this

mrousavy, 2024-04-18:
Can you maybe help debug this a bit? I'm not sure what values I need to put in here: https://github.com/mrousavy/react-native-vision-camera/blob/5a350c6dc9f8328cadf8982e59b4ae69146952ad/package/android/src/main/java/com/mrousavy/camera/core/MetadataProvider.kt#L38-L40

Those values are `5000L` for the update interval (= 5 seconds) and `5f` for the distance (= 5 meters) - isn't that reasonable?
If I set both to `0` it will stream location quite aggressively, no?

There is also a method to imperatively get the current location ([`LocationManager.getCurrentLocation`](https://developer.android.com/reference/android/location/LocationManager#getCurrentLocation(java.lang.String,%20android.os.CancellationSignal,%20java.util.concurrent.Executor,%20java.util.function.Consumer%3Candroid.location.Location%3E))) but this is API level 30+.

idrisssakhi, 2024-05-11:
@mrousavy seems good I made a sort of loader when opening camera on android to extract location & if I wait a bit before taking the first image the location is good. we can close this

closed 2024-05-11 by idrisssakhi