margelo/react-native-vision-camera #2714: 🐛 CameraVision Example Crash on Tap to Focus on Android.

mdottavi, 2024-04-03:
### What's happening?

Hi, 
I'm integrating react-native-camera-vision in my project. 
I'm getting an app crash ( on my Android 14 - Motorola G84 5G) when Tapping on the screen to set Focus.
I see that the camera responds setting the focus where tapped but after few seconds it crashes.
I tested with metro and with apk debug build. Same thing.

The same test works on OnePlus Nord3 - Android 14 / Doogee T20S - Android 13.

To double check it was not a fault in my code I built CameraVision Example debug apk 
and made the same test and got exact same crash on the same Tap-to-focus action on my phone (but not on the other mentioned phones).

### Reproduceable Code

```tsx
CameraVision Example code without any change.
```

### Relevant log output

```shell
Hereafter LogCat CameraVision Example  (the stack dump on my app is exactly the same):

2024-04-03 17:15:16.640 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D  1920 x 864 Image with format #35. Logging 5 parameters:
2024-04-03 17:15:16.640 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D    -> 42.0 (java.lang.Double)
2024-04-03 17:15:16.640 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D    -> true (java.lang.Boolean)
2024-04-03 17:15:16.641 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D    -> {test=0.0, second=test} (java.util.HashMap)
2024-04-03 17:15:16.641 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D    -> hello! (java.lang.String)
2024-04-03 17:15:16.641 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D    -> [another test, 5.0] (java.util.ArrayList)
2024-04-03 17:15:16.643 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D  1920 x 864 Image with format #35. Logging 5 parameters:
2024-04-03 17:15:16.643 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D    -> 42.0 (java.lang.Double)
2024-04-03 17:15:16.643 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D    -> true (java.lang.Boolean)
2024-04-03 17:15:16.644 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D    -> {test=0.0, second=test} (java.util.HashMap)
2024-04-03 17:15:16.644 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D    -> hello! (java.lang.String)
2024-04-03 17:15:16.644 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D    -> [another test, 5.0] (java.util.ArrayList)
2024-04-03 17:15:16.656 29226-29270 AndroidRuntime          com.mrousavy.camera.example          E  FATAL EXCEPTION: mrousavy/VisionCamera.main
                                                                                                    Process: com.mrousavy.camera.example, PID: 29226
                                                                                                    com.mrousavy.camera.core.CaptureTimedOutError: [capture/timed-out] The image capture was aborted because it timed out.
                                                                                                    	at com.mrousavy.camera.extensions.CameraCaptureSession_setRepeatingRequestAndWaitForPrecaptureKt$setRepeatingRequestAndWaitForPrecapture$2$1.invokeSuspend(CameraCaptureSession+setRepeatingRequestAndWaitForPrecapture.kt:131)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:584)
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:793)
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:697)
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:684)
                                                                                                    	Suppressed: kotlinx.coroutines.internal.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@4d6c394, mrousavy/VisionCamera.main]
2024-04-03 17:15:16.657 29226-29271 VideoPipeline           com.mrousavy.camera.example          I  ImageReader::onImageAvailable!
2024-04-03 17:15:16.657 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D  1920 x 864 Image with format #35. Logging 5 parameters:
2024-04-03 17:15:16.658 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D    -> 42.0 (java.lang.Double)
2024-04-03 17:15:16.658 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D    -> true (java.lang.Boolean)
2024-04-03 17:15:16.658 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D    -> {test=0.0, second=test} (java.util.HashMap)
2024-04-03 17:15:16.658 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D    -> hello! (java.lang.String)
2024-04-03 17:15:16.658 29226-29271 ExamplePlugin           com.mrousavy.camera.example          D    -> [another test, 5.0] (java.util.ArrayList)
2024-04-03 17:15:16.659 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D  1920 x 864 Image with format #35. Logging 5 parameters:
2024-04-03 17:15:16.659 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D    -> 42.0 (java.lang.Double)
2024-04-03 17:15:16.659 29226-29275 ReactNativeJS           com.mrousavy.camera.example          I  440441351674830: 1920x864 yuv Frame (portrait)
2024-04-03 17:15:16.659 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D    -> true (java.lang.Boolean)
2024-04-03 17:15:16.660 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D    -> {test=0.0, second=test} (java.util.HashMap)
2024-04-03 17:15:16.660 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D    -> hello! (java.lang.String)
2024-04-03 17:15:16.660 29226-29271 ExampleKotlinPlugin     com.mrousavy.camera.example          D    -> [another test, 5.0] (java.util.ArrayList)
2024-04-03 17:15:16.664  1663-11286 ActivityTaskManager     system_server                        W    Force finishing activity com.mrousavy.camera.example/.MainActivity
2024-04-03 17:15:16.676 29226-29270 Process                 com.mrousavy.camera.example          I  Sending signal. PID: 29226 SIG: 9
2024-04-03 17:15:16.717  1663-11508 ActivityManager         system_server                        I  appDiedLocked: app=ProcessRecord{b342493 29226:com.mrousavy.camera.example/u0a495} thread=android.os.BinderProxy@f38a39d fromBinderDied=true isKilledByAm=false reason=null
2024-04-03 17:15:16.717  1663-11508 ActivityManager         system_server                        I  Process com.mrousavy.camera.example (pid 29226) has died: vis TOP 
2024-04-03 17:15:16.717  1663-14872 ImeTracker              system_server                        I  com.mrousavy.camera.example:a50d9f5f: onRequestHide at ORIGIN_SERVER_HIDE_INPUT reason HIDE_REMOVE_CLIENT
2024-04-03 17:15:16.717  1663-14872 ImeTracker              system_server                        I  com.mrousavy.camera.example:a50d9f5f: onCancelled at PHASE_SERVER_SHOULD_HIDE
2024-04-03 17:15:16.718  1663-4668  WindowManager           system_server                        I  WIN DEATH: Window{6cb5463 u0 com.mrousavy.camera.example/com.mrousavy.camera.example.MainActivity}
2024-04-03 17:15:16.719  1663-4668  InputManager-JNI        system_server                        W  Input channel object '6cb5463 com.mrousavy.camera.example/com.mrousavy.camera.example.MainActivity (client)' was disposed without first being removed with the input manager!
```
[... 23 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Motorola G84 5G

### VisionCamera Version

3.9.2

[... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Jeanne-1, 2024-04-09:
same on Xiaomi Pad 6

mrousavy, 2024-04-09:
Can you test if VisionCamera V4 beta fixes your issue?

mdottavi, 2024-04-10:
> Can you test if VisionCamera V4 beta fixes your issue?

Hi,
tested RNVC eample app on branch "v4" on the buggy phone (Android 14 - Motorola G84 5G) ...
commit cd9b84d838e744ccd2f9944e447357140ce47a98 (HEAD -> v4, origin/v4)
Author: Marc Rousavy <me@mrousavy.com>
Date:   Sat Apr 6 12:18:26 2024 +0200

and tap-to-focus **was working properly** and Example App **was not crashing** !

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy