margelo/react-native-vision-camera #2716: 🐛

PedroGabrielCB, 2024-04-04:
### What's happening?

The Camera is working fine on most devices. However, there are a few on which it just won't work. The camera will simply try to load and stay black.

i tried add -> const device = useCameraDevices() but does work....

### Reproduceable Code

```tsx
<SafeAreaView style={StyleSheet.absoluteFill}>
                                <Camera
                                    style={{ flex: 1 }}
                                    torch='off'
                                    device={device}
                                    isActive={true}
                                    zoom={0}
                                    photo={true}
                                    exposure={0}
                                    ref={camera}
                                    orientation='portrait'
                                />

                            </SafeAreaView>
```

### Relevant log output

```shell
none
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 24,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -24,
  "name": "1 (FRONT) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "front",
  "id": "1"
}
```

### Device

Poco X4 Pro

### VisionCamera Version

4.0.0-beta.10

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

PedroGabrielCB, 2024-04-17:
Redmi note 14 pro also has the same problem

PedroGabrielCB, 2024-04-17:
@mrousavy do you have any suggestions on how I can fix it? I have an app in production with this problem... @

mrousavy, 2024-04-17:
If you have issues you cannot solve yourself, feel free to contact me for consultancy so I can fix them for you.

PedroGabrielCB, 2024-04-17:
> If you have issues you cannot solve yourself, feel free to contact me for consultancy so I can fix them for you.
But isn't this a library bug as versusvoid put it above?

mrousavy, 2024-04-17:
Yea, might be. I didn't investigate it yet.

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy