margelo/react-native-vision-camera #2720: 🐛 v4.0.0-beta.13 - Android preview freeze/crash

samuelbeaulieu, 2024-04-05:
### What's happening?

It doesn't happen all the time, but sometime when taking a picture I get this error "Failed to take photo! [unknown/unknown: Camera is closed.]". When taking multiple pictures one after the other, the error will happen. The example app seems to handle the error better than my implementation because on mine the preview freeze.

https://github.com/mrousavy/react-native-vision-camera/assets/64789082/0c0697d9-ef18-4bdb-aa11-d7806edf1fae

This issue, as far as I know, is only affecting Android devices. All my tests on iOS didn't show any problems.

On device, the issue happens after `2024-04-05 12:56:20` in the logcat.

### Reproduceable Code

```tsx
// camera/camera-screen.jsx

  const camera = useRef(null);
  const device = useCameraDevice('back');

  // ...

  takePicture = async () => {
    const { camera, setPreview } = this.props;

    if (camera.current) {
      console.log('camera.current is not null');

      this.setState({ loadingSpinnerVisible: true });

      try {
        const photo = await camera.current.takePhoto({
          enableShutterSound: false,
          qualityPrioritization: Platform.OS === 'android' ? 'speed' : 'balanced',
          flash: 'off',
        });

        // Set the preview picture
        setPreview(photo);
      } catch (error) {
        console.log('error taking photo', error);
      } finally {
        this.setState({ loadingSpinnerVisible: false });
      }
    } else {
      console.log('camera.current is null');
    }
  };

  // ...

  renderCamera() {
    const {
      camera,
      activeAngle,
      showExposureControls,
    } = this.props;

    // ...

    return (
        // ...
          <MainCameraView
            innerRef={camera}
            activeAngle={activeAngle}
            showExposureControls={showExposureControls}
          />
        // ...
    );
  }

  // ...
[... 267 lines cut ...]
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy S21 (Android 14), Android devices

### VisionCamera Version

4.0.0-beta.13

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

samuelbeaulieu, 2024-04-05:
Full logcat because the issue add too many characters. On device, the issue happens after `2024-04-05 12:56:20` in the logcat.

This is a full logcat from the camera screen opening to the issue. The logcat in the main issue is only a few seconds before the issue happens on the device.

```
2024-04-05 12:55:40.770 18277-18277/com.my-bundle.my-app-name I/ViewRootImpl@6cd2404[MainActivity]: ViewPostIme pointer 0
2024-04-05 12:55:40.826 18277-18277/com.my-bundle.my-app-name I/ViewRootImpl@6cd2404[MainActivity]: ViewPostIme pointer 1
2024-04-05 12:55:40.833 18277-10173/com.my-bundle.my-app-name I/ReactNativeJS: inside
2024-04-05 12:55:41.170 18277-10173/com.my-bundle.my-app-name I/ReactNativeJS: 'camera', { current: null }
2024-04-05 12:55:41.186 18277-10173/com.my-bundle.my-app-name I/ReactNativeJS: 'device', { formats: 
       [ { pixelFormats: [ 'yuv', 'native' ],
           videoStabilizationModes: [ 'off' ],
           autoFocusSystem: 'contrast-detection',
           photoWidth: 4032,
           supportsPhotoHdr: false,
           supportsDepthCapture: false,
           maxISO: 3200,
           minISO: 50,
           minFps: 7,
           videoWidth: 3840,
           supportsVideoHdr: false,
           videoHeight: 2160,
           fieldOfView: 80.0605180983776,
           maxFps: 30,
           photoHeight: 3024 },
         { pixelFormats: [ 'yuv', 'native' ],
           videoStabilizationModes: [ 'off' ],
           autoFocusSystem: 'contrast-detection',
           photoWidth: 4032,
           supportsPhotoHdr: false,
           supportsDepthCapture: false,
           maxISO: 3200,
           minISO: 50,
           minFps: 7,
           videoWidth: 3840,
           supportsVideoHdr: false,
           videoHeight: 2160,
           fieldOfView: 80.0605180983776,
           maxFps: 30,
           photoHeight: 2268 },
         { pixelFormats: [ 'yuv', 'native' ],
           videoStabilizationModes: [ 'off' ],
           autoFocusSystem: 'contrast-detection',
           photoWidth: 4032,
           supportsPhotoHdr: false,
           supportsDepthCapture: false,
           maxISO: 3200,
           minISO: 50,
           minFps: 7,
           videoWidth: 3840,
           supportsVideoHdr: false,
           videoHeight: 2160,
           fieldOfView: 80.0605180983776,
           maxFps: 30,
           photoHeight: 1816 },
         { pixelFormats: [ 'yuv', 'native' ],
           videoStabilizationModes: [ 'off' ],
           autoFocusSystem: 'contrast-detection',
           photoWidth: 4032,
           supportsPhotoHdr: false,
           supportsDepthCapture: false,
           maxISO: 3200,
           minISO: 50,
           minFps: 7,
           videoWidth: 3840,
           supportsVideoHdr: false,
           videoHeight: 2160,
           fieldOfView: 80.0605180983776,
           maxFps: 30,
           photoHeight: 1728 },
[... 880 lines cut ...]
2024-04-05 12:56:24.819 18277-18419/com.my-bundle.my-app-name I/ViewRootImpl@7e07a6b[DevMenuActivity]: mWNT: t=0xb40000743584bdf0 mBlastBufferQueue=0xb4000075a44fa670 fn= 1 caller= android.view.ViewRootImpl$8.onFrameDraw:13614 android.view.ThreadedRenderer$1.onFrameDraw:788 <bottom of call stack> 
2024-04-05 12:56:24.820 18277-18419/com.my-bundle.my-app-name D/ViewRootImpl@7e07a6b[DevMenuActivity]: Setting up sync and frameCommitCallback
2024-04-05 12:56:24.821 18277-18410/com.my-bundle.my-app-name I/BLASTBufferQueue: [ViewRootImpl@7e07a6b[DevMenuActivity]#115](f:0,a:0,s:0) onFrameAvailable the first frame is available
2024-04-05 12:56:24.822 18277-18410/com.my-bundle.my-app-name D/ViewRootImpl@7e07a6b[DevMenuActivity]: Received frameCommittedCallback lastAttemptedDrawFrameNum=1 didProduceBuffer=true
2024-04-05 12:56:24.823 18277-18410/com.my-bundle.my-app-name D/OpenGLRenderer: CFMS:: SetUp Pid : 18277    Tid : 18410
2024-04-05 12:56:24.823 18277-18277/com.my-bundle.my-app-name D/ViewRootImpl@7e07a6b[DevMenuActivity]: reportDrawFinished
2024-04-05 12:56:24.824 18277-18277/com.my-bundle.my-app-name I/ViewRootImpl@6cd2404[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 0 0
2024-04-05 12:56:24.827 18277-18277/com.my-bundle.my-app-name I/ViewRootImpl@7e07a6b[DevMenuActivity]: registerCallbackForPendingTransactions
2024-04-05 12:56:24.830 18277-18277/com.my-bundle.my-app-name D/InsetsSourceConsumer: applyRequestedVisibilityToControl: visible=true, type=1
2024-04-05 12:56:24.830 18277-18277/com.my-bundle.my-app-name D/InsetsSourceConsumer: applyRequestedVisibilityToControl: visible=true, type=2
2024-04-05 12:56:24.835 18277-18418/com.my-bundle.my-app-name I/ViewRootImpl@7e07a6b[DevMenuActivity]: mWNT: t=0xb40000743588af30 mBlastBufferQueue=0xb4000075a44fa670 fn= 2 caller= android.view.ViewRootImpl$6.onFrameDraw:5539 android.view.ViewRootImpl$2.onFrameDraw:2103 android.view.ThreadedRenderer$1.onFrameDraw:788 
2024-04-05 12:56:24.855 18277-18277/com.my-bundle.my-app-name I/ViewRootImpl@7e07a6b[DevMenuActivity]: MSG_WINDOW_FOCUS_CHANGED 1 0
2024-04-05 12:56:24.856 18277-18277/com.my-bundle.my-app-name D/ViewRootImpl@7e07a6b[DevMenuActivity]: mThreadedRenderer.initializeIfNeeded()#2 mSurface={isValid=true 0xb40000763517f070}
2024-04-05 12:56:24.857 18277-18277/com.my-bundle.my-app-name D/InputMethodManagerUtils: startInputInner - Id : 0
2024-04-05 12:56:24.857 18277-18277/com.my-bundle.my-app-name I/InputMethodManager: startInputInner - IInputMethodManagerGlobalInvoker.startInputOrWindowGainedFocus
[... cut ...]  2
2024-04-05 12:56:31.572 18277-18427/com.my-bundle.my-app-name E/BufferQueueProducer: [ImageReader-4032x3024f22m9-18277-81](id:4765000000c3,api:4,p:1969,c:18277) dequeueBuffer: waitForFreeSlotThenRelock: timeout. dequeuedCount: 0, acquiredCount: 2
2024-04-05 12:56:32.575 18277-18721/com.my-bundle.my-app-name E/BufferQueueProducer: [ImageReader-4032x3024f22m9-18277-81](id:4765000000c3,api:4,p:1969,c:18277) dequeueBuffer: waitForFreeSlotThenRelock: timeout. dequeuedCount: 0, acquiredCount: 2
```

mrousavy, 2024-04-21:
Hey - can you still reproduce this on the latest V4 beta (18)?

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

spsaucier, 2024-05-10:
My users are still seeing this error just as often in version 4.0.1. We did a release with 4.0.3 today, and I am still seeing some instances of `ZslControlImpl: Failed to acquire latest image IllegalStateException = Maximum image number reached.` there too, unfortunately.

mrousavy, 2024-05-13:
`photoQualityBalance="speed"` or `="quality"` fixes this, apparently ZSL is broken on some devices.

spsaucier, 2024-05-14:
Thanks @mrousavy, adding `photoQualityBalance="speed"` does appear to have fixed this error (though I have only a few hours of data so far).

samuelbeaulieu, 2024-05-14:
I was already using `photoQualityBalance="speed"`, that might be why it fixed the issue for me.