margelo/react-native-vision-camera #2721: 🐛 v4.0.0-beta.13 - Camera.onError(unknown/unknown): already bound to a different lifecycle.

isinuyk, 2024-04-06:
### What's happening?

I wanted to flip the camera from the Front to the Back one but got this error: Camera.onError(unknown/unknown): Use case Preview:androidx.camera.core.Preview-de4f3173-e76f-49cf-b589-73fab23b84b1 already bound to a different lifecycle.

https://github.com/mrousavy/react-native-vision-camera/assets/72970722/74ff6996-4001-4762-954c-2c532a7a9e86

### Reproduceable Code

```tsx
export default function App() {
  const [view, setView] = useState<'front' | 'back'>('back');
  const device = useCameraDevice(view);

  const {hasPermission, requestPermission} = useCameraPermission();
  console.log(JSON.stringify(device, (k, v) => (k === 'formats' ? [] : v), 2));
  useEffect(() => {
    requestPermission();
  }, []);

  if (!hasPermission || !device) {
    return null;
  }
  return (
    <>
      <Camera style={StyleSheet.absoluteFill} device={device} isActive={true} />
      <TouchableOpacity
        onPress={() => setView(prev => (prev === 'back' ? 'front' : 'back'))}
        style={{
          position: 'absolute',
          bottom: 20,
          borderWidth: 2,
          borderColor: 'white',
          backgroundColor: 'transparent',
          right: 20,
          width: 40,
          height: 40,
          borderRadius: 20,
        }}
      />
    </>
  );
}
```

### Relevant log output

```shell
Failed to configure CameraSession! Error: Use case Preview:androidx.camera.core.Preview-de4f3173-e76f-49cf-b589-73fab23b84b1 already bound to a different lifecycle., Config-Diff: Difference(deviceChanged=true, outputsChanged=false, sidePropsChanged=false, isActiveChanged=false, locationChanged=false)
java.lang.IllegalStateException: Use case Preview:androidx.camera.core.Preview-de4f3173-e76f-49cf-b589-73fab23b84b1 already bound to a different lifecycle.
	at androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:587)
	at androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:386)
	at com.mrousavy.camera.core.CameraSession.configureCamera(CameraSession.kt:417)
	at com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:165)
	at com.mrousavy.camera.CameraView$update$1.invokeSuspend(CameraView.kt:156)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8669)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)

'Camera.onError(unknown/unknown): Use case Preview:androidx.camera.core.Preview-de4f3173-e76f-49cf-b589-73fab23b84b1 already bound to a different lifecycle.', { [unknown/unknown: Use case Preview:androidx.camera.core.Preview-de4f3173-e76f-49cf-b589-73fab23b84b1 already bound to a different lifecycle.]
  name: 'unknown/unknown',
  _code: 'unknown/unknown',
  _message: 'Use case Preview:androidx.camera.core.Preview-de4f3173-e76f-49cf-b589-73fab23b84b1 already bound to a different lifecycle.',
  _cause: 
   { stacktrace: 'java.lang.IllegalStateException: Use case Preview:androidx.camera.core.Preview-de4f3173-e76f-49cf-b589-73fab23b84b1 already bound to a different lifecycle.\n\tat androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:587)\n\tat androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:386)\n\tat com.mrousavy.camera.core.CameraSession.configureCamera(CameraSession.kt:417)\n\tat com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:165)\n\tat com.mrousavy.camera.CameraView$update$1.invokeSuspend(CameraView.kt:156)\n\tat kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)\n\tat kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)\n\tat android.os.Handler.handleCallback(Handler.java:938)\n\tat android.os.Handler.dispatchMessage(Handler.java:99)\n\tat android.os.Looper.loopOnce(Looper.java:226)\n\tat android.os.Looper.loop(Looper.java:313)\n\tat android.app.ActivityThread.main(ActivityThread.java:8669)\n\tat java.lang.reflect.Method.invoke(Native Method)\n\tat com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)\n\tat com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)\n',
     message: 'Use case Preview:androidx.camera.core.Preview-de4f3173-e76f-49cf-b589-73fab23b84b1 already bound to a different lifecycle.' } }
```

### Camera Device

```json
{
  "hardwareLevel": "full",
  "minExposure": -20,
  "neutralZoom": 1,
  "minZoom": 1,
  "supportsFocus": true,
  "formats": [],
  "supportsLowLightBoost": false,
  "hasTorch": false,
  "supportsRawCapture": false,
  "minFocusDistance": 0,
  "sensorOrientation": "landscape-right",
  "maxZoom": 8,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "isMultiCam": false,
  "position": "front",
  "id": "1",
  "hasFlash": false,
  "maxExposure": 20,
  "name": "1 (FRONT) androidx.camera.camera2"
}
```

### Device

Samsung Galaxy A11

### VisionCamera Version

v4.0.0-beta.13

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bexelstudio, 2024-04-11:
have the same issue on A9 Samsung no ref found. on 3.9.2 black screen!

moafandi0, 2024-04-21:
having the same issue on OnePlus5

moafandi0, 2024-04-21:
The camera component needs to be unmounted before you can change the camera. Here's a workaround for now:

```
const [cameraPosition, setCameraPosition] = useState("front");
const [isActive, setIsActive] = useState(true);
let device = useCameraDevice(cameraPosition);
const onFlipCameraPressed = useCallback(() => {
    setIsActive(false);
  }, [isActive]);
useEffect(() => {
    if (!isActive) {
      cameraPosition === "front"
        ? setCameraPosition("back")
        : setCameraPosition("front");
      setIsActive(true);
    }
    return () => {};
  }, [isActive]);
```

and add isActive to conditionally render the component:

```
{isActive && (
        <Camera
          style={{ flex: 1 }}
          device={device}
          isActive={true}
        />
      )}
```

mrousavy, 2024-04-21:
Fixed this issue in https://github.com/mrousavy/react-native-vision-camera/pull/2768. 💪

closed 2024-04-21 by mrousavy