margelo/react-native-vision-camera #2723: How can you crop the resulting photo of camera.current.takePhoto()

tomerh2001, 2024-04-07:
### Question

I have the following code:
```
	const photo = await camera.current.takePhoto();
	const base64 = await ReactNativeBlobUtil.fs.readFile(photo.path, 'base64');
	setReviewImage({ autoDetected, base64 ...photo });
```

I need to crop the image I get, I can't use plugins of this library since the resulting photo is not in the same format as the plugins expect (I.E. Frame object), and there doesn't seem to be any way to just manipulate the array of pixels directly.

Even if it's not a feature of this library, the issue is inherently from the library, and it's currently delaying a production issue we need to roll out, so if anyone knows a library I could use to crop the output, that'll be amazing.

### VisionCamera Version

3.9.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

bruno-centanaro, 2024-04-07:
[Expo image manipulator](https://docs.expo.dev/versions/latest/sdk/imagemanipulator/)

tomerh2001, 2024-04-07:
> [Expo image manipulator](https://docs.expo.dev/versions/latest/sdk/imagemanipulator/)

We're not using expo :/

Dingenis, 2024-04-08:
You could use one of these two: 
- https://github.com/callstack/react-native-image-editor
- https://github.com/guhungry/react-native-photo-manipulator

Another approach would be to create a Native Module and write the crop code yourself in the native languages aka Kotlin, Swift or Objective-C. This gives you more control on what you want.

mrousavy, 2024-04-09:
Exactly what @Dingenis / @bruno-centanaro said - A Camera library shouldn't focus on later image manipulation, otherwise this is going to get too big. People will want quick filter solutions, colorspace changes, blurring, etc. etc. - just use a separate library that was made for this.

closed 2024-04-09 by mrousavy