margelo/react-native-vision-camera #2726: 🔧 iOS build failing

Rc85, 2024-04-08:
### How were you trying to build the app?

Building through Xcode on a simulator.

[image]

### Full build logs

```tsx
Build log is too long, lag when I paste and can't submit because of timeout.
```

### Project dependencies

```json
"dependencies": {
    "@eva-design/eva": "^2.1.1",
    "@gorhom/bottom-sheet": "^4.6.1",
    "@mdi/js": "^7.4.47",
    "@react-native-community/datetimepicker": "^7.6.2",
    "@react-navigation/native": "^6.1.15",
    "@ui-kitten/components": "^5.1.2",
    "axios": "^1.6.7",
    "react": "^18.2.0",
    "react-native": "^0.72.6",
    "react-native-config": "^1.4.11",
    "react-native-gesture-handler": "^2.15.0",
    "react-native-image-crop-picker": "^0.40.3",
    "react-native-maps": "^1.10.3",
    "react-native-reanimated": "^3.8.0",
    "react-native-redash": "^18.1.3",
    "react-native-safe-area-context": "^4.9.0",
    "react-native-select-dropdown": "^3.4.0",
    "react-native-sound": "^0.11.2",
    "react-native-svg": "^13.4.0",
    "react-native-toast-notifications": "^3.3.1",
    "react-native-vector-icons": "^10.0.3",
    "react-native-vision-camera": "^3.9.1",
    "socket.io-client": "^4.7.4",
    "vision-camera-code-scanner": "^0.2.0"
  },
```

### VisionCamera Version

3.9.1

### Target platforms

iOS

### Operating system

MacOS

### Can you build the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-22:
hey - I think there's something wrong with your Xcode. Update your Xcode, your macOS, and VisionCamera to V4, and try again.

closed 2024-04-22 by mrousavy

nolife08021, 2024-06-21:
@Rc85 Hi,  whats your Xcode version? did you able to fix it?