margelo/react-native-vision-camera #2736: 💭 How to follow QR position

shacharcohen1997, 2024-04-12:
### Question

Question
I am really struggling with a specific task.. i am trying to follow a qr code with a square
I tried this https://codeshare.io/dekyBR which almost works in 1 device but not in others and not in iphone

### What I tried

```
import { StatusBar } from 'expo-status-bar'
import { useEffect, useRef, useState } from 'react'
import { Dimensions, StyleSheet, View } from 'react-native'
import { Polygon, Svg } from 'react-native-svg'
import {
  Camera,
  Templates,
  useCameraDevice,
  useCameraFormat,
  useCameraPermission,
  useCodeScanner,
} from 'react-native-vision-camera'

export default function App() {
  const [targetFps] = useState(60)

  const { width: viewWidth, height: viewHeight } = Dimensions.get('window')
  const [polygonPoints, setPolygonPoints] = useState('')

  const codeScanner = useCodeScanner({
    codeTypes: ['qr'],
    onCodeScanned: (codes, frame) => {
      if (codes.length > 0 && codes[0]?.frame && codes[0]?.corners) {
        const { x, y, width, height } = codes[0].frame
        const points = codes[0].corners.map((corner) => ({
          x: corner.y,
          y: corner.x,
        }))

        setPolygonPoints(points.map((p) => `${p.x},${p.y}`).join(' '))
      }
    },
  })

  const camera = useRef<Camera>(null)

  const { hasPermission, requestPermission } = useCameraPermission()

  const device = useCameraDevice('back')
  const format = useCameraFormat(device, Templates.Photo)

  const fps = Math.min(format?.maxFps ?? 1, targetFps)

  useEffect(() => {
    !hasPermission && requestPermission()
  }, [hasPermission, requestPermission])

  useEffect(() => {
    // Re-render the Svg component when the polygonPoints state changes
    // This will update the overlay with the QR code position and area
  }, [polygonPoints])

  if (!device || !hasPermission) return null

  return (
    <View style={styles.container}>
      <Camera
        ref={camera}
        fps={fps}
        isActive={true}
        pixelFormat="yuv"
        device={device}
        format={format}
        codeScanner={codeScanner}
        style={StyleSheet.absoluteFill}
        onError={console.error}
      />
      <Svg height="100%" width="100%" viewBox={`0 0 ${viewWidth} ${viewHeight}`}>
        <Polygon points={polygonPoints}  
          fill="lime"
          stroke="green"
          opacity="0.5"
          strokeWidth="1"
        />
      </Svg>
      <StatusBar style="light" />
    </View>
  )
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    backgroundColor: '#fff',
    alignItems: 'center',
    justifyContent: 'center',
  },
})
```

### VisionCamera Version

3.9.2

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

shacharcohen1997, 2024-04-12:
https://github.com/mrousavy/react-native-vision-camera/issues/2436#issuecomment-1925577421 made it perfect but i dont know how she made it

shacharcohen1997, 2024-04-12:
Iphone - going to the sides is moving up/down and moving up/down is moving to the sides
https://github.com/mrousavy/react-native-vision-camera/assets/160487610/31221c05-8ea8-4a58-a94e-5463dbbffd3b
Android
https://github.com/mrousavy/react-native-vision-camera/assets/160487610/c582749b-7199-465f-9a90-9ba22eeaad18

shacharcohen1997, 2024-04-12:
i managed to improve the code by using 
```
// useTransposeFromStream.ts 
/** Determines the offset & scale values to translate stream coordinates to a layouts coordinate */
const getTransposeProperties = (source: Dimension, target: Dimension) => {
  const scaleRatio = Math.max(
    target.width / source.width,
    target.height / source.height,
  );
  // Represents the width and height of the stream, scaled down to fit the screen
  // This is without taking into account the cut off part of the stream (to fit the screen)
  const uncutDimensions = {
    width: source.width * scaleRatio,
    height: source.height * scaleRatio,
  };
  // Reprensents how much of the stream is cut off to fit the screen
  const cutDimensions = {
    width: (uncutDimensions.width - target.width)*(scaleRatio),
    height: (uncutDimensions.height - target.height)*scaleRatio,
  };
  const x = {
    offset: cutDimensions.width/ 2 ,
    scale: scaleRatio,
  };
  const y = {
    offset: cutDimensions.height / 2,
    scale: scaleRatio,
  };
//   console.log("scaleRatio", scaleRatio, "uncutDimensions", uncutDimensions, "cutDimensions", cutDimensions, "x", x, "y", y);

  return { x, y };
};

export const transposeFromStream = (
    { x, y, width, height }: Rectangle,
    source: Dimension, // replace 'any' with the actual type of 'stream'
    target: Dimension
  ) => {
    const transposeFromStreamProperties = getTransposeProperties(source, target);
  
    return {
      x:
        x * transposeFromStreamProperties.x.scale -
        transposeFromStreamProperties.x.offset,
      y:
        y * transposeFromStreamProperties.y.scale -
        transposeFromStreamProperties.y.offset,
      width: width * transposeFromStreamProperties.x.scale,
      height: height * transposeFromStreamProperties.y.scale,
    };
  };

/** Given a set of coordinates in a landscape picture, returns the corresponding coordinates for a vertical plan  */
export const rotateCoordinatesToVertical = (
  { x, y }: Point,
  { height }: Dimension,
) => {
  return {
    x: height - y,
    y: x,
  };
};

export const rotateRectangleToVertical = (
  rectangle: Rectangle,
  frame: Dimension,
) => {
  // Rotating by 90° means the width and height are swapped
  const width = rectangle.height;
  const height = rectangle.width;
  // Rotate the top left corner coordinates
  const rotatedCoordinates = rotateCoordinatesToVertical(rectangle, frame);
  // The top left corner coordinates now represent the top right corner of the rectangle in the new plane
  // So recalculate the new top left corner coordinates
  return {
    x: rotatedCoordinates.x - width,
    y: rotatedCoordinates.y,
    width,
    height,
  };
};

export type Dimension = {
  width: number;
  height: number;
};
export type Point = {
  x: number;
  y: number;
};
export type Rectangle = Point & Dimension;
```

shacharcohen1997, 2024-04-12:
https://github.com/mrousavy/react-native-vision-camera/assets/160487610/4a77146b-079e-4927-b727-eddc3a6d148c

tuannguyen0410, 2024-04-19:
Hi @shacharcohen1997  I have follow your idea and have a same problem. and I have try like this .maybe this will have help you. Viewbox of my camera is '0 0 414 709' you can get is by View Ref
[image]
And my useCodeScanner function is like this.
[image]
By the way I just make a test in Rect . you can try it in Polygon . Many thank for your idea

tuannguyen0410, 2024-04-19:
And here my result 
[image]

tuannguyen0410, 2024-04-19:
Here is my result for this detection
[image]

tuannguyen0410, 2024-04-22:
@shacharcohen1997 I have try to draw code Scanner Border only when detected code

 const topLeftX = codesFrame.x;
            const topLeftY = codesFrame.y;
            const topRightX = codesFrame.x + codesFrame.width;
            const topRightY = codesFrame.y;
            const bottomRightX = codesFrame.x + codesFrame.width;
            const bottomRightY = codesFrame.y + codesFrame.height;
            const bottomLeftX = codesFrame.x;
            const bottomLeftY = codesFrame.y + codesFrame.height;
            const paths = [
                { d: `M${topLeftX + codesFrame.width / 3} ${topLeftY} H${topLeftX} V${topLeftY + codesFrame.height / 3}` },
                { d: `M${topRightX - codesFrame.width / 3} ${topRightY} H${topRightX} V${topRightY + codesFrame.height / 3}` },
                { d: `M${bottomRightX - codesFrame.width / 3} ${bottomRightY} H${bottomRightX} V${bottomRightY - codesFrame.height / 3}` },
                { d: `M${bottomLeftX + codesFrame.width / 3} ${bottomLeftY} H${bottomLeftX} V${bottomLeftY
                
 divide for 3 just my calculate the stroke width:
 The Result is:
 
[image]

mrousavy, 2024-04-22:
Hey I think this can be discussed on the community discordi nstead of GitHub issues.

closed 2024-04-22 by mrousavy

shachar1997, 2024-04-25:
> @shacharcohen1997 I have try to draw code Scanner Border only when detected code
> 
> const topLeftX = codesFrame.x; const topLeftY = codesFrame.y; const topRightX = codesFrame.x + codesFrame.width; const topRightY = codesFrame.y; const bottomRightX = codesFrame.x + codesFrame.width; const bottomRightY = codesFrame.y + codesFrame.height; const bottomLeftX = codesFrame.x; const bottomLeftY = codesFrame.y + codesFrame.height; const paths = [ { d: `M${topLeftX + codesFrame.width / 3} ${topLeftY} H${topLeftX} V${topLeftY + codesFrame.height / 3}` }, { d: `M${topRightX - codesFrame.width / 3} ${topRightY} H${topRightX} V${topRightY + codesFrame.height / 3}` }, { d: `M${bottomRightX - codesFrame.width / 3} ${bottomRightY} H${bottomRightX} V${bottomRightY - codesFrame.height / 3}` }, { d: `M${bottomLeftX + codesFrame.width / 3} ${bottomLeftY} H${bottomLeftX} V${bottomLeftY
> 
> divide for 3 just my calculate the stroke width: The Result is:
> 
> <img alt="image" width="143" src="https://private-user-images.githubusercontent.com/24580746/324374108-0a9e2fc8-75f8-4b7f-b0ea-273a6b056ab1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQwNTQwMzksIm5iZiI6MTcxNDA1MzczOSwicGF0aCI6Ii8yNDU4MDc0Ni8zMjQzNzQxMDgtMGE5ZTJmYzgtNzVmOC00YjdmLWIwZWEtMjczYTZiMDU2YWIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI1VDE0MDIxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA2N2E3ODM2OTg4Y2E5NGExMTNlZDc3ZDcwYmVhN2Q4ZmI0M2Q1M2U3ODJmOWVmYzYyOTI4NzY4ZDM1YTY1MjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4H05FjDEmK6M8VXclDV1kpWuXYNoV31GbTofz17EWvM">

can you send the full snippet  or your code?

tuannguyen0410, 2024-05-02:
@shacharcohen1997 Just a short of my code. I think this will help you

`
    const [paths, setPaths] = useState<{ d: string }[] | null>(null);

const codeScanner = useCodeScanner({
        codeTypes: ['qr', 'ean-13'],
        onCodeScanned: (codes) => {
            runAtTargetFps(5, () => {
                "worklet";
                if (codes && codes.length > 0 && handleCodeScanned) {
                  handleCodeScanned(codes[0].value || '');
                  if (codes[0].frame) {
                    const codesFrame = codes[0].frame;
                    const topLeftX = codesFrame.x;
                    const topLeftY = codesFrame.y;
                    const topRightX = codesFrame.x + codesFrame.width;
                    const topRightY = codesFrame.y;
                    const bottomRightX = codesFrame.x + codesFrame.width;
                    const bottomRightY = codesFrame.y + codesFrame.height;
                    const bottomLeftX = codesFrame.x;
                    const bottomLeftY = codesFrame.y + codesFrame.height;
                    const paths = [
                        { d: `M${topLeftX + codesFrame.width / 3} ${topLeftY} H${topLeftX} V${topLeftY + codesFrame.height / 3}` },
                        { d: `M${topRightX - codesFrame.width / 3} ${topRightY} H${topRightX} V${topRightY + codesFrame.height / 3}` },
                        { d: `M${bottomRightX - codesFrame.width / 3} ${bottomRightY} H${bottomRightX} V${bottomRightY - codesFrame.height / 3}` },
                        { d: `M${bottomLeftX + codesFrame.width / 3} ${bottomLeftY} H${bottomLeftX} V${bottomLeftY - codesFrame.height / 3}` },
                      ];
                    setPaths(paths);
                  }
                } else {
                  setPaths(null);
                }
            });
        },
      });
      
      return (
      <>
      <VisionCamera
                        ref={cameraRef}
                        device={device}
                        isActive={isActive}
                        style={isCodeScan ? styles.cameraQr : styles.camera}
                        onInitialized={onInitialized}
                        onError={onError}
                        enableZoomGesture={false}
                        photo={true}
                        codeScanner={codeScanner}
                        // fps={fps}
                        // lowLightBoost={device.supportsLowLightBoost && enableNightMode}
                        orientation={orientation}
                        {...(deviceFormat?.format && {
                          format: {
                            ...deviceFormat?.format,
                          },
                        })}
                      />
      {isCodeScan && paths && (
                        <WrapCodeDetection>
                          <Svg
                            width="100%"
                            height="100%"
                            viewBox={`0 0 ${cameraSize?.width} ${cameraSize?.height}`}
                            strokeWidth="2"
                            stroke="#f05b28"
                            fill="none"
                            strokeLinecap="round"
                            strokeLinejoin="round"
                          >
                            {paths.map((item, index) => (
                              <Path key={index} d={item.d} fill="none" />
                            ))}
                          </Svg>
                        </WrapCodeDetection>
                      )}
      </>
      )
`

tuannguyen0410, 2024-05-02:
@shacharcohen1997  I think this well help
[image]
[image]

tuannguyen0410, 2024-05-02:
> can you send the full snippet or your code?

@shacharcohen1997  I think this well help
[image]
[image]

inkCrazy, 2024-07-11:
refre to:https://medium.com/@susumtang/improve-user-experience-with-barcode-highlight-box-using-react-native-vision-camera-code-scanner-ff7f67089aab

[image]
[image]

ios device correct√
android device Coordinate shift x

tuannha01, 2024-07-11:
> refre to:https://medium.com/@susumtang/improve-user-experience-with-barcode-highlight-box-using-react-native-vision-camera-code-scanner-ff7f67089aab
> 
> android： ![5731720667763_ pic_hd](https://private-user-images.githubusercontent.com/12253453/347667833-55ec36a1-6e65-4341-badc-3fd7d33d9007.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA2NzM5NDAsIm5iZiI6MTcyMDY3MzY0MCwicGF0aCI6Ii8xMjI1MzQ1My8zNDc2Njc4MzMtNTVlYzM2YTEtNmU2NS00MzQxLWJhZGMtM2ZkN2QzM2Q5MDA3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzExVDA0NTQwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI3N2RlNDM3OGZkY2Q4OWQ4YTRmZGNlNzVlOGI4YzdjMTkyYWVhMzA5NmFjMDE0N2FlMmUxMmMyMjhlYzZmOTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._zmPcrB43t6VqsJmJYSSAeCWim5wfcbkIsgMQkJ2Z1s)
> 
> iOS： ![5741720667766_ pic_hd](https://private-user-images.githubusercontent.com/12253453/347667846-8daefd79-c135-4882-89e4-fc5c5c69a82b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA2NzM5NDAsIm5iZiI6MTcyMDY3MzY0MCwicGF0aCI6Ii8xMjI1MzQ1My8zNDc2Njc4NDYtOGRhZWZkNzktYzEzNS00ODgyLTg5ZTQtZmM1YzVjNjlhODJiLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzExVDA0NTQwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ0ZjViZmNjNGEyMGY1YzE3MjU0NmE5NmNmNTc3NDBkZjQ5ODUzYmY1N2Y2Y2YzYWRkOGIzMTE4MjgyYzU0N2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.LNa5En0SqoihBlfjXN-poA72twmLPOUjwtv-hE9RLOY)
> 
> ios device correct√ android device Coordinate shift x

Can you share your code to discuss more?

inkCrazy, 2024-07-11:
> Can you share your code to discuss more?

```
import {useFocusEffect, useIsFocused} from '@react-navigation/native';

import React, {useEffect, useState, useRef} from 'react';
import {
  StyleSheet,
  Dimensions,
  Text,
  TouchableWithoutFeedback,
  Image,
  NativeModules,
  StatusBar,
  Platform,
  AppState,
  BackHandler,
  DrawerLayoutAndroid,
  Linking,
  PermissionsAndroid,
  View,
  Pressable,
} from 'react-native';
import ImagePicker from 'react-native-image-crop-picker';
import RNQRGenerator from 'rn-qr-generator';
import ToastUtils from '../utils/ToastUtils';
import ComUtils from '../utils/ComUtils';
import BarcodeMask from '../components/BarcodeMask';
import {useHeaderHeight} from '@react-navigation/elements';
import {
  Camera,
  useCameraDevice,
  NoCameraDeviceError,
  useCodeScanner,
} from 'react-native-vision-camera';
import DeviceInfo from 'react-native-device-info';

const BarCodeScanner = ({navigation, route}) => {
  const {StatusBarManager} = NativeModules;
  const statusBarHeight = StatusBarManager.HEIGHT; //状态栏高度：ios54 android32
  const navigationBarHeader = useHeaderHeight(); //ios103, android 56
  const {height, width} = Dimensions.get('window');
  // const CAM_VIEW_HEIGHT = Dimensions.get('window').height;
  const CAM_VIEW_HEIGHT =
    height -
    (Platform.OS === 'ios' ? statusBarHeight : statusBarHeight) -
    navigationBarHeader;
  const CAM_VIEW_WIDTH = Dimensions.get('window').width;
  // The following are based on landscape orientation with home button to the right
  //andorid全面屏幕中 Dimensions.get('window').height 计算屏幕高度时会自动减少StatusBar 的高度
  const maskWidth = width * 0.7;
  // left
  const topMargin =
    (height -
      (Platform.OS === 'ios' ? statusBarHeight : 0) -
      navigationBarHeader -
      maskWidth) /
    2;
  // top
  const leftMargin = (width - maskWidth) / 2;
  // width (height of portrait orientation)
  const frameWidth = maskWidth;
  // height (width of portrait orientation)
  const frameHeight = maskWidth;
  //TM的计算的百分比
  const scanAreaX = leftMargin / CAM_VIEW_HEIGHT;
  const scanAreaY = topMargin / CAM_VIEW_WIDTH;
  const scanAreaWidth = frameWidth / CAM_VIEW_HEIGHT;
  const scanAreaHeight = frameHeight / CAM_VIEW_WIDTH;
  //参考地址：https://github.com/react-native-camera/react-native-camera/pull/2719/files/583507f22363d330f903af0215f20250a7ee3851#diff-518dcc98e7451b35bc3b9acfc686894d2bee38b6ced2d9c59ccec56153283f5d
[... 403 lines cut ...]
Languages:
  Java:
    version: 11.0.22
    path: /Users/inkcrazy/.sdkman/candidates/java/current/bin/javac
  Ruby:
    version: 3.0.0
    path: /Users/inkcrazy/.rvm/rubies/ruby-3.0.0/bin/ruby
npmPackages:
  "@react-native-community/cli": Not Found
  react:
    installed: 18.2.0
    wanted: 18.2.0
  react-native:
    installed: 0.74.2
    wanted: 0.74.2
  react-native-macos: Not Found
npmGlobalPackages:
  "*react-native*": Not Found
Android:
  hermesEnabled: true
  newArchEnabled: false
iOS:
  hermesEnabled: true
  newArchEnabled: false
```