margelo/react-native-vision-camera #2738: 🐛 App Crash in Specific Android Device

ShamuXX, 2024-04-12:
### What's happening?

**What's happening?**

When I open the camera with the QR code reading property (codeScanner), it locks instantly. (See crash log)

On other devices rarely times crash whitout any error crash just show the follow message.

---------------------------- PROCESS ENDED (6814) for package com.example----------------------------

**Device with error**

These are the characteristics of the device that fails with the camera:

Model name: Pixel 6
Android version: Android 12 (SDK 31)
ABI: arm64-v8a

**gradle configuration**

buildscript {
    ext {
        buildToolsVersion = "33.0.0"
        minSdkVersion = 26
        compileSdkVersion = 33
        targetSdkVersion = 33
        ndkVersion = "23.1.7779620"
    }
    

### Reproduceable Code

```tsx
const [enable, setEnable] = useState(isEnable);
const isFocused = useIsFocused();
const [hasPermission, setHasPermission] = useState(false);
const device = useCameraDevice('back');
const isActive = isFocused && enable && hasPermission && device != null;

const format = useCameraFormat(device, [
    { videoAspectRatio: 16 / 9 },
    { videoResolution: { width: 3048, height: 2160 } },
    { fps: 60 },
    { photoAspectRatio: 16 / 9 }
  ]);

<>
      {isActive && (
        <View style={styles.container}>
          <Camera
            style={styles.camera}
            device={device}
            isActive={isActive}
            format={format}
            codeScanner={codeScanner}
            exposure={0}
          />
          <TouchableOpacity style={ifModalCodesQR ? styles.captureButtonModalCodes : styles.captureButton} onPress={() => handle()}>
            <Text style={styles.captureButtonText}>Cerrar Camara</Text>
          </TouchableOpacity>
        </View>
      )}
    </>

const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13', 'codabar', 'aztec'],
    onCodeScanned: (codes) => {
      if (codes && codes.length > 0) {
        const newCodes = codes.map(code => code.value);
        const uniqueNewCodes = new Set(newCodes);
        setUniqueCodesSet(prevSet => new Set([...prevSet, ...uniqueNewCodes]));
        if (uniqueNewCodes.size === 1) {
          setLastScannedCode(newCodes[0]);
        }
      }
    }
  });
```

### Relevant log output

```shell
Native crash of /apex/com.google.pixel.camera.hal/bin/hw/android.hardware.camera.provider@2.7-service-google

backtrace:
  #00  pc 0x000000000004f85c  /apex/com.android.runtime/lib64/bionic/libc.so (abort+168)
  #01  pc 0x0000000000083260  /apex/com.google.pixel.camera.hal/lib64/lib_3a_coordinator.so (std::__1::__throw_bad_optional_access()+8)
  #02  pc 0x000000000009a884  /apex/com.google.pixel.camera.hal/lib64/lib_3a_coordinator.so (g3a_coordinator::AfEngine::SetFocusCommand(g3a_coordinator::AFInputMetadataValues const&, g3a_coordinator::AFInputVendorTagValues const&, gaf::GAfFrameInput*)+844)
  #03  pc 0x000000000009bb60  /apex/com.google.pixel.camera.hal/lib64/lib_3a_coordinator.so (g3a_coordinator::AfEngine::UpdateInputFromMetadata(g3a_coordinator::AFInputMetadataValues const&, g3a_coordinator::AFInputVendorTagValues const&, g3a_coordinator::FrameInfo const&, gaf::GAfRawStats*, gaf::GAfFrameInput*)+192)
  #04  pc 0x000000000009ee4c  /apex/com.google.pixel.camera.hal/lib64/lib_3a_coordinator.so (g3a_coordinator::AfEngine::Execute(g3a_api::PackedData const&, g3a_api::PackedData&)::$_0::operator()() const+404)
  #05  pc 0x0000000000098a64  /apex/com.google.pixel.camera.hal/lib64/lib_3a_coordinator.so (g3a_coordinator::AfEngine::Execute(g3a_api::PackedData const&, g3a_api::PackedData&)+336)
  #06  pc 0x00000000000a9530  /apex/com.google.pixel.camera.hal/lib64/lib_3a_coordinator.so (g3a_coordinator::AfProcessor::Execute(g3a_api::PackedData const&, g3a_api::PackedData&)+180)
  #07  pc 0x00000000000b0580  /apex/com.google.pixel.camera.hal/lib64/lib_3a_coordinator.so (g3a_coordinator::Coordinator::Process(lyric_api::AutoFocusInterface::Input const&, std::__1::vector<lyric::CameraMetadata*, std::__1::allocator<lyric::CameraMetadata*> > const&)+1492)
  #08  pc 0x000000000044164c  /apex/com.google.pixel.camera.hal/lib64/liblyric_hwl.so (lyric::ExternalAutoFocusControllerProxy::Process(lyric::ProcessingContext*, lyric::AutoFocusController::Input&, std::__1::vector<lyric::CameraMetadata*, std::__1::allocator<lyric::CameraMetadata*> > const&, lyric::ActuatorController*, lyric::MotionSensorController::GyroContext*, lyric::MotionSensorController::GravityContext*, lyric::AutoFocusController::FencedOutputs const&)+2608)
  #09  pc 0x000000000024c4d4  /apex/com.google.pixel.camera.hal/lib64/liblyric_hwl.so (lyric::AutoFocusNode::ProcessInput(lyric::FlatHashMap<lyric::InputPort*, std::__1::vector<lyric::RelFrame, std::__1::allocator<lyric::RelFrame> > const, lyric::container_internal::HashEq<lyric::InputPort*, void>::Hash, lyric::container_internal::HashEq<lyric::InputPort*, void>::Eq, std::__1::allocator<std::__1::pair<lyric::InputPort* const, std::__1::vector<lyric::RelFrame, std::__1::allocator<lyric::RelFrame> > const> > > const&)+6016)
  #10  pc 0x000000000074b7dc  /apex/com.google.pixel.camera.hal/lib64/liblyric_hwl.so (lyric::NodeInputGroupHolder::InvokeWhenReady(lyric::Duration)+640)
  #11  pc 0x00000000000818fc  /apex/com.google.pixel.camera.hal/lib64/liblyric_base.so (lyric::CameraPerformanceAnalyzerRef::ProfileNodeInvocation(lyric::CameraPerformanceAnalyzer::PerNodeInputKey const&, std::__1::function<lyric::StatusOr<long> ()>)+96)
  #12  pc 0x0000000000081948  /apex/com.google.pixel.camera.hal/lib64/liblyric_base.so (lyric::CameraPerformanceAnalyzerRef::ProfileNodeInvocation(lyric::CameraPerformanceAnalyzer::PerNodeInputKey const&, std::__1::function<lyric::StatusOr<long> ()>)+172)
  #13  pc 0x0000000000734dd4  /apex/com.google.pixel.camera.hal/lib64/liblyric_hwl.so (lyric::GraphRunner::ThreadFunction(lyric::InputPriorityLevel, lyric::InputProcessingCoreAffinity, lyric::InputProcessingHintGroup, int)+2508)
  #14  pc 0x00000000001ad864  /apex/com.google.pixel.camera.hal/lib64/liblyric_hwl.so (std::__1::__function::__func<std::__1::future<std::__1::result_of<lyric::GraphRunner::StartGraph()::$_0 ()>::type> lyric::ThreadPool::EnqueueWork<lyric::GraphRunner::StartGraph()::$_0>(lyric::GraphRunner::StartGraph()::$_0&&)::'lambda'(), std::__1::allocator<std::__1::future<std::__1::result_of<lyric::GraphRunner::StartGraph()::$_0 ()>::type> lyric::ThreadPool::EnqueueWork<lyric::GraphRunner::StartGraph()::$_0>(lyric::GraphRunner::StartGraph()::$_0&&)::'lambda'()>, void ()>::operator()()+88)
  #15  pc 0x00000000001ac298  /apex/com.google.pixel.camera.hal/lib64/liblyric_hwl.so (lyric::ThreadPool::ThreadFunction()+108)
  #16  pc 0x00000000001ac1f4  /apex/com.google.pixel.camera.hal/lib64/liblyric_hwl.so (void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, lyric::ThreadPool::ThreadPool(int)::'lambda'()> >(void*)+40)
  #17  pc 0x00000000000b1690  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+204)
  #18  pc 0x00000000000510ac  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64)
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 12,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -12,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Pixel 6 (Android 12)

### VisionCamera Version

3.5.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-04-12 by ShamuXX

ShamuXX, 2024-04-12:
Based on LOGCAT I can solve this problem, I check all the warnings and errors that may affect the performance of the camera.
To complete this issue, I checked and rewrote all the metadata that the camera component receives to force the camera to start only if I have all the prop values.

Example: 

 const isActive = isFocused && enable && hasPermission && device != null;
  
<View style={styles.container}>
        <Camera
          style={styles.camera}
          device={device}
          isActive={isActive}
          format={format}
          codeScanner={codeScanner}
          exposure={0}
        />

I also check that the camera lifecycle is working and responding correctly, instead of handling the states individually each time the camera session is opened and closed, I chose to create functions so that I don't have a bad response or call when the camera is opened. camera.

Sharf8351, 2024-04-26:
I'm facing the same issue on the device Google Pixel 7a while using "3.5.1" version. @ShamuXX  kindly provide the solution. Thanks in advance