margelo/react-native-vision-camera #2741: takePhoto not working in android🐛

devsframe, 2024-04-15:
### What's happening?

I wanted to click a picture but when I am trying to click one, its causing the app to crash.
Also when i am checking the logs for camera.current their is no function for takePhoto().

[image]

### Reproduceable Code

```tsx
const device = useCameraDevice('front'); 
const camera = useRef(null);
const [isCameraInitialized, setIsCameraInitialized] = useState(false);

  const visionTakeImage = async () => {
    try {
      console.log('current', camera.current);
      const img = await camera.current.takePhoto();
      console.log('img', img);
    } catch (error) {
      console.log('error',error);
    }
  };

<Camera
    onError={onError}
    ref={camera}
    style={StyleSheet.absoluteFill}
    device={device}
    isActive={true}
    photo={true}
    onInitialized={onCameraInitialized}
  />

  <TouchableOpacity 
   onPress={visionTakeImage}>
  </TouchableOpacity>
```

### Relevant log output

```shell
Crashing the application
```

### Camera Device

```json
'{\n  "formats": [],\n  "sensorOrientation": "landscape-right",\n  "hardwareLevel": "full",\n  "maxZoom": 5,\n  "minZoom": 1,\n  "maxExposure": 12,\n  "supportsLowLightBoost": false,\n  "neutralZoom": 1,\n  "physicalDevices": [\n    "wide-angle-camera"\n  ],\n  "supportsFocus": false,\n  "supportsRawCapture": true,\n  "isMultiCam": false,\n  "minFocusDistance": 0,\n  "minExposure": -12,\n  "name": "FRONT (1)",\n  "hasFlash": false,\n  "hasTorch": false,\n  "position": "front",\n  "id": "1"\n}'
```

### Device

OnePLus NordeCE 5G

### VisionCamera Version

3.9.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

devsframe, 2024-04-16:
@mrousavy  Can you please help

mrousavy, 2024-04-16:
no logs provided. closing

closed 2024-04-16 by mrousavy