margelo/react-native-vision-camera #2744: 🐛  ERROR  TypeError: 0, _$$_REQUIRE(_dependencyMap[6](...)n-camera").useSkiaFrameProcessor is not a function (it is undefined)

dw444, 2024-04-16:
### What's happening?

I attempted to integrate Skia into react-native-vision-camera V4 following the guidelines in #2727. However, the useSkiaFrameProcessor function isn't functioning properly. Despite clearing the metro cache, reinstalling node-modules, and cleaning the build folder, the problem persists. It's worth mentioning that useFrameProcessor functions as expected (without Skia).

### Reproduceable Code

```tsx
import {
    Camera,
    useSkiaFrameProcessor,
    useCameraDevice,
} from 'react-native-vision-camera';
import {Skia,Paint, Color} from '@shopify/react-native-skia';

function App(): React.JSX.Element {
    const [hasPermission, setHasPermission] = useState(false)
    const device = useCameraDevice("front");
    useEffect(() => {
        Camera.requestCameraPermission().then((p) =>
                                              setHasPermission(p === 'granted')
                                              )
    }, []);

    const frameProcessor = useSkiaFrameProcessor((frame) => {
        'worklet'
            frame.render()
            const redPaint = Skia.Paint()
            const red = Skia.Color('red')
            redPaint.setColor(red)
            const rect = Skia.XYWHRect(frame.width/2, frame.height/2, 150, 150)
            frame.drawRect(rect, redPaint)
    });
   

    return (
            <View style={StyleSheet.absoluteFill}>
            {!hasPermission && <Text style={styles.text}>No Camera Permission.</Text>}
            {hasPermission && device != null && (
                                                 <Camera
                                                 style={StyleSheet.absoluteFill}
                                                 device={device}
                                                 isActive={true}
                                                 frameProcessor={frameProcessor}
                                                 pixelFormat="rgb"
                                                 />
                                                 )}
            </View>
          );
}
```

### Relevant log output

```shell
ERROR  TypeError: 0, _$$_REQUIRE(_dependencyMap[6](...)n-camera").useSkiaFrameProcessor is not a function (it is undefined)

This error is located at:
    in App
    in RCTView (created by View)
    in View (created by AppContainer)
    in RCTView (created by View)
    in View (created by AppContainer)
    in AppContainer
    in camerav4(RootComponent), js engine: hermes
```

### Camera Device

```json
LOG  {
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:1",
  "isMultiCam": false,
  "hardwareLevel": "full",
  "position": "front",
  "supportsLowLightBoost": false,
  "maxExposure": 8,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "neutralZoom": 1,
  "supportsRawCapture": false,
  "supportsFocus": false,
  "formats": [],
  "hasTorch": false,
  "hasFlash": true,
  "minExposure": -8,
  "name": "Front Camera",
  "minZoom": 1,
  "sensorOrientation": "landscape-right",
  "maxZoom": 16,
  "minFocusDistance": 0
}
```

### Device

iPhone 11

### VisionCamera Version

4.0.0-beta.13

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Dingenis, 2024-04-16:
Hey @dw444, version 4.0.0-beta.13 does not yet include `useSkiaFrameProcessor` and the other code for Skia. The PR you mentioned has not been merged yet and doesn't have a version yet (as seen from[ this comment](https://github.com/mrousavy/react-native-vision-camera/pull/2727#issuecomment-2049722092)). 

You can test it cloning the project, checking out the branch `v4-skia` and including that in your project. Or by waiting for the beta release like the comment mentions.

dw444, 2024-04-16:
Hey @Dingenis, thanks for the quick response, appreciate it!

closed 2024-04-17 by dw444