margelo/react-native-vision-camera #2748: 🐛 Flip camera is not working in Android

lv-saikrishna, 2024-04-17:
### What's happening?

I am using the 4.0.0-beta.15 version,  Flip camera is not working in Android. The same code was working fine in V 3.9.2

### Reproduceable Code

```tsx
const camera = useRef<Camera>(null)
  const isFocussed = useIsFocused()
  const isForeground = useIsForeground()
  const isActive = isFocussed && isForeground
  const [cameraPosition, setCameraPosition] = useState<'front' | 'back'>('back')
  
  let device = useCameraDevice(cameraPosition)

  const onFlipCameraPressed = useCallback(() => {
    setCameraPosition((p) => (p === 'back' ? 'front' : 'back'))
  }, [])
  
    const codeScanner: CodeScanner = {
    codeTypes: ['code-39'],
    onCodeScanned: useCallback((codes) => {
      const scanResult = codes[0]?.value?.trim() ?? '';	  
	  ///log scanResult
      }
    }, []),
  };
  

  return (
    <View style={styles.container}>
      {device != null && (
        <Camera ref={camera} style={styles.Camera} device={device} isActive={isActive} codeScanner={codeScanner} enableZoomGesture={true}></Camera>
      )}

      <View style={styles.rightButtonRow}>
        <PressableOpacity style={styles.button} onPress={onFlipCameraPressed} disabledOpacity={0.4}>
          <IonIcon name="camera-reverse" color="white" size={24} />
        </PressableOpacity>
      </View>
    </View>
  )
```

### Relevant log output

```shell
I don't have the logs to share
```

### Camera Device

```json
{}
```

### Device

samsung galaxy s23

### VisionCamera Version

4.0.0-beta.15

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

EmanuilM, 2024-04-18:
I have same issue
### Relevant log output
```
Camera.onError(unknown/unknown): Use case Preview:androidx.camera.core.Preview-1fdaff89-d5cf-44e3-9c72-75d09de454d2 already bound to a different lifecycle. [unknown/unknown: Use case Preview:androidx.camera.core.Preview-1fdaff89-d5cf-44e3-9c72-75d09de454d2 already bound to a different lifecycle.]
```
### Camera Device
```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "limited",
  "maxZoom": 1,
  "minZoom": 1,
  "maxExposure": 6,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "ultra-wide-angle-camera"
  ],
  "supportsFocus": false,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 0,
  "minExposure": -6,
  "name": "10 (BACK) androidx.camera.camera2",
  "hasFlash": false,
  "hasTorch": false,
  "position": "back",
  "id": "10"
}
```
### Device
Samsung A71 - Android 13, Android Emulator - Android 13,14
### VisionCamera Version
4.0.0-beta.14

moafandi0, 2024-04-21:
see this
https://github.com/mrousavy/react-native-vision-camera/issues/2721#issuecomment-2068177951

mrousavy, 2024-04-21:
Duplicate of #2721

mrousavy, 2024-04-21:
Also without sharing logs I cannot help you @lv-saikrishna.

Thanks for extending this and sharing logs @EmanuilM though.

mrousavy, 2024-04-21:
Fixed this issue in https://github.com/mrousavy/react-native-vision-camera/pull/2768. 💪

closed 2024-04-21 by mrousavy