margelo/react-native-vision-camera #2756: 💭 Doc for useSkiaFrameProcessor appears to be wrong

EkaanshArora, 2024-04-18:
### Question

The doc for [skia-frame-processors](https://react-native-vision-camera.com/docs/guides/skia-frame-processors) states:

```js
const invertColorsFilter = Skia.RuntimeEffect.Make(``)
const paint = Skia.Paint(invertColorsFilter) // this method doesn't accept an argument

const frameProcessor = useSkiaFrameProcessor((frame) => {
  'worklet'
  frame.render(paint)
}, [])
```
`Skia.Paint()` doesn't accept any arguments. Does the doc need to be updated to call something like `paint.setShader` instead?

### VisionCamera Version

4.0.0-beta.15

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-04-19:
Yea they aren't up to date yet.

EkaanshArora, 2024-04-19:
Happy to make a PR to fix this if you can describe the process to use a shader within the worklet.

mrousavy, 2024-04-19:
I'm already on it

mrousavy, 2024-04-22:
yes, I just released 4.0.0 and updated the docs.

Space6188, 2024-04-22:
Thank you for pretty library, that really cool
Unfortunately, this not working:
const invertColorsFilter = Skia.RuntimeEffect.Make(`
  uniform shader image;
  half4 main(vec2 pos) {
    vec4 color = image.eval(pos);
    return vec4((1.0 - color).rgb, 1.0);
  }
`);
  const paint = Skia.Paint(invertColorsFilter);

  const frameProcessor = useSkiaFrameProcessor(
    frame => {
      'worklet';
      frame.render(paint);
    },
    [paint],
  );

I have this error - 'Expected arguments: 0, received: 1.ts(2554)' inside Skia.paint(invertColorsFilter)

i working with android and this is my versions of packeges:
    "react-native-vision-camera": "^4.0.1",
    "@shopify/react-native-skia": "1.2.3",
    "react-native-reanimated": "^3.5.4",

Maybe you know how to fix it, maybe i do something wrong

Space6188, 2024-04-22:
This is working now!, thank you so much

mrousavy, 2024-04-22:
nice