margelo/react-native-vision-camera #2757: 🐛 It is not possible to record video and take photos at the same time on Android

huuthongdev, 2024-04-19:
### What's happening?

It is not possible to record video and take photos at the same time on Android. i wanna implement taking photo when recording

### Reproduceable Code

```tsx
<ReanimatedCamera
  ref={camera}
  device={device}
  format={format}
  fps={fps}
  photoHdr={enableHdr}
  videoHdr={enableHdr}
  photo={true}
  video={true}
/>
```

### Relevant log output

```shell
LOG  🔴 [ERROR] [12:44:17 4/19/2024]   Camera Error [session/recoverable-error: An unknown error occurred while creating the Camera Session, but the Camera can recover from it.]
[ERROR] [12:44:18 4/19/2024]   Take photo failed [unknown/unknown: Camera is closed.]
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 12,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 9.200000175476077,
  "minExposure": -12,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

LGE LM-V600

### VisionCamera Version

4.0.0-beta.15

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-19:
This should work flawlessly, not sure why it breaks for you.

I currently don't have time to investigate this, so what you can do is:

- either you can try to hunt down the bug in VisionCamera yourself by first trying to create a native Java Android app with CameraX, and then seeing if the issue can be reproduced there
- or pay me to fix the issue for you

But either way, the logs you posted are definitely not enough for this and you need to pull native adb logcat logs. Not sure why everyone misses that

mrousavy, 2024-04-22:
Can you still reproduce this in VisionCamera 4.0.0?

closed 2024-04-22 by mrousavy

huuthongdev, 2024-04-22:
> Can you still reproduce this in VisionCamera 4.0.0?

I tested on V 4.0.0 but it doesn't work. I see the problem in Android Studio's logs

[image]

mrousavy, 2024-04-22:
hm, really weird. That sounds like a CameraX bug.

jkaufman, 2024-09-10:
I encountered a similar issue in production. Same device.

```# Crashlytics - Stack trace
# Application: com.***.***
# Platform: android
# Version: 1.3.0 (149)
# Issue: 4e648d4455f2a8c193254d888a0eb86b
# Session: 66E049240050000123E213B233D61E60_DNE_1_v2
# Date: Tue Sep 10 2024 07:28:38 GMT-0600 (Mountain Daylight Time)

Non-fatal Exception: io.invertase.firebase.crashlytics.JavaScriptError: Failed to configure the Camera Session because the output/stream configurations are invalid!
       at .<unknown>(Camera Error)
       at .construct((native):0:0)
       at .apply((native):0:0)
       at ._construct(b9d939a775424bd9c9095691b72c513f:1:148214)
       at .Wrapper(b9d939a775424bd9c9095691b72c513f:1:147868)
       at .construct((native):0:0)
       at ._callSuper(b9d939a775424bd9c9095691b72c513f:1:747805)
       at .CameraError(b9d939a775424bd9c9095691b72c513f:1:748209)
       at .construct((native):0:0)
       at ._callSuper(b9d939a775424bd9c9095691b72c513f:1:747805)
       at .CameraRuntimeError(b9d939a775424bd9c9095691b72c513f:1:748721)
       at .onError(b9d939a775424bd9c9095691b72c513f:1:745078)
       at .apply((native):0:0)
       at .invokeGuardedCallbackImpl(b9d939a775424bd9c9095691b72c513f:1:388317)
       at .apply((native):0:0)
       at .invokeGuardedCallback(b9d939a775424bd9c9095691b72c513f:1:388374)
       at .apply((native):0:0)
       at .invokeGuardedCallbackAndCatchFirstError(b9d939a775424bd9c9095691b72c513f:1:388406)
       at .executeDispatch(b9d939a775424bd9c9095691b72c513f:1:388535)
       at .executeDispatchesAndReleaseTopLevel(b9d939a775424bd9c9095691b72c513f:1:391933)
       at .call((native):0:0)
       at .forEachAccumulated(b9d939a775424bd9c9095691b72c513f:1:390011)
       at .anonymous(b9d939a775424bd9c9095691b72c513f:1:392296)
       at .batchedUpdatesImpl(b9d939a775424bd9c9095691b72c513f:1:459215)
       at .batchedUpdates$1(b9d939a775424bd9c9095691b72c513f:1:391849)
       at ._receiveRootNodeIDEvent(b9d939a775424bd9c9095691b72c513f:1:392134)
       at .receiveEvent(b9d939a775424bd9c9095691b72c513f:1:453955)
       at .apply((native):0:0)
       at .__callFunction(b9d939a775424bd9c9095691b72c513f:1:136772)
       at .anonymous(b9d939a775424bd9c9095691b72c513f:1:135212)
       at .__guard(b9d939a775424bd9c9095691b72c513f:1:136153)
       at .callFunctionReturnFlushedQueue(b9d939a775424bd9c9095691b72c513f:1:135170)```

ollyde, 2024-12-21:
I have the same issue, want to snap photos or take videos, simple user interface request. Doesn't seem possible in Expo-camera 😢

I seriously regret choosing expo as a framework, seems like it's not maintained.