margelo/react-native-vision-camera #2759: 🐛 system/frame-processors-unavailable: Frame Processors are not available, react-native-worklets-core is not installed!

sekhuat, 2024-04-20:
### What's happening?

I tried to use the Frame Processors on both Expo and pure React Native app and both give me the error "system/frame-processors-unavailable: Frame Processors are not available, react-native-worklets-core is not installed!" when the camera is displayed. However the `react-native-worklets-core` module is already installed and properly configured according to the documentation. Both expo and pure react native app is a blank app created to test the react-native-vision-camera.

When the app launch and the camera is loaded then the app will throw the above mentioned error. This happened on both iOS and Android device.

### Reproduceable Code

```tsx
Expo App Package.json:

{
  "name": "expo cam",
  "version": "1.0.0",
  "main": "node_modules/expo/AppEntry.js",
  "scripts": {
    "start": "expo start",
    "android": "expo run:android",
    "ios": "expo run:ios",
    "web": "expo start --web",
    "ts:check": "tsc"
  },
  "dependencies": {
    "expo": "~50.0.14",
    "expo-status-bar": "~1.11.1",
    "react": "18.2.0",
    "react-native": "0.73.6",
    "react-native-vision-camera": "^3.9.2",
    "react-native-worklets-core": "^1.1.1"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@types/react": "~18.2.45",
    "typescript": "^5.1.3"
  },
  "private": true
}
```

Pure React Native Package.json:
```
{
  "name": "rncam",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "lint": "eslint .",
    "start": "react-native start",
    "test": "jest"
  },
  "dependencies": {
    "react": "18.2.0",
    "react-native": "0.73.7",
    "react-native-vision-camera": "^3.9.2",
    "react-native-worklets-core": "^1.1.1"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@babel/preset-env": "^7.20.0",
    "@babel/runtime": "^7.20.0",
    "@react-native/babel-preset": "0.73.21",
    "@react-native/eslint-config": "0.73.2",
    "@react-native/metro-config": "0.73.5",
    "@react-native/typescript-config": "0.73.1",
    "@types/react": "^18.2.6",
    "@types/react-test-renderer": "^18.0.0",
    "babel-jest": "^29.6.3",
    "eslint": "^8.19.0",
[... 343 lines cut ...]
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "hardwareLevel": "full",
  "hasTorch": true
}
```

### Device

iPhone 7 iOS 15.8.2, Huawei Mate30 Pro Android 12

### VisionCamera Version

3.9.2

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

alized559, 2024-04-20:
same for me

ExoticCoder16, 2024-04-21:
same issue, while my Frame Processor was still working, it logged some frames ; then starting from yesterday, it said it’s not installed 
[image]

sahinboga, 2024-04-21:
same issue for me
"react-native": "0.73.6",
 "react-native-vision-camera": "^3.9.2",
 "react-native-worklets-core": "^1.1.1",

matheuscaet, 2024-04-21:
Same issue here

    "react-native-vision-camera": "^3.9.2",
    "react-native-vision-camera-face-detector": "^1.5.0",
    "react-native-worklets-core": "^1.1.1"
    "react-native": "0.72.4",
    
    ndkVersion = "25.1.8937393"
    kotlinVersion = "1.8.10"
    gradleVersion = "7.4.2"

mrousavy, 2024-04-21:
Hey - react-native-worklets-core 1.1.1 introduced a breaking change. You need to stay on 0.x.x for VisionCamera V3, or upgrade to VisionCamera V4 to use 1.x.x of react-native-worklets-core.

closed 2024-04-21 by mrousavy

sekhuat, 2024-04-25:
Finally get it work! For those who faced the same issue, you need to install the `react-native-reanimated` for it to work.

Here are the packages I installed. Make sure the react-native-worklets-core is 0.x.x if you using VisionCamera V3.

```
"dependencies": {
    "expo": "~50.0.14",
    "expo-status-bar": "~1.11.1",
    "react": "18.2.0",
    "react-native": "0.73.6",
    "react-native-vision-camera": "^3.9.2",
    "react-native-worklets-core": "0.5.0",
    "react-native-reanimated": "~3.6.2"
  },
```

And here is my babel.config.js

```
module.exports = function(api) {
  api.cache(true);
  return {
    plugins: [
      ["react-native-worklets-core/plugin"],
      'react-native-reanimated/plugin'
    ],
    presets: ['babel-preset-expo'],
  };
};
```

Good luck!!

sekhuat, 2024-04-25:
> You don't need Reanimated for Frame Processors, only Worklets Core (which is a much lighter / smaller dependency). Of course it doesn't hurt if you already have Reanimated installed, but you don't _need_ it for frame processors.
> 
> Maybe you just forgot to add Worklets to the babel config or didn't clear cache?

If I didn't install the `react-native-reanimated` module, I will get the `ERROR  Camera.onError(unknown/unknown): [unknown/unknown] Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29 and has protection mode(s): rw-p [unknown/unknown: [unknown/unknown] Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29 and has protection mode(s): rw-p]` error. After I installed the `react-native-reanimated` module, everything works fine.

akash-r-98, 2024-05-03:
I am still facing this issue

`"dependencies": {
    "react": "18.2.0",
    "react-native": "0.71.10",
    "react-native-camera": "^4.2.1",
    "react-native-easy-toast": "^2.3.0",
    "react-native-reanimated": "~3.6.2",
    "react-native-svg": "^15.1.0",
    "react-native-worklets-core": "0.5.0",
    "react-native-vision-camera": "3.9.2"
  },`

babel.config.js

`module.exports = {
  presets: ['module:metro-react-native-babel-preset'],
  plugins: [
    ['react-native-worklets-core/plugin'],
    'react-native-reanimated/plugin',
  ],
};
`

[image]

mrousavy, 2024-05-03:
try upgrading to react-native-vision-camera V4 and react-native-worklets-core 1.x.x.

akash-r-98, 2024-05-03:
I am still facing this issue

```
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.71.10",
    "react-native-camera": "^4.2.1",
    "react-native-easy-toast": "^2.3.0",
    "react-native-reanimated": "~3.6.2",
    "react-native-svg": "^15.1.0",
    "react-native-worklets-core": "0.5.0",
    "react-native-vision-camera": "3.9.2"
  },
```

babel.config.js

```
module.exports = {
  presets: ['module:metro-react-native-babel-preset'],
  plugins: [
    ['react-native-worklets-core/plugin'],
    'react-native-reanimated/plugin',
  ],
};
```
[image]

lin826, 2024-05-27:
Here is a combination works!

```json
  "dependencies": {
    "@react-native-camera-roll/camera-roll": "^7.4.0",
    "@react-native-community/blur": "^4.4.0",
    "@react-navigation/core": "^6.4.10",
    "@react-navigation/native-stack": "^6.9.17",
    "@shopify/react-native-skia": "0.1.221",
    "react": "18.2.0",
    "react-native": "0.73.6",
    "react-native-fast-image": "^8.6.3",
    "react-native-fast-tflite": "^1.2.0",
    "react-native-gesture-handler": "~2.14.0",
    "react-native-mmkv": "^2.11.0",
    "react-native-pressable-opacity": "^1.0.10",
    "react-native-reanimated": "~3.6.2",
    "react-native-safe-area-context": "4.8.2",
    "react-native-screens": "~3.29.0",
    "react-native-static-safe-area-insets": "^2.2.0",
    "react-native-vector-icons": "^10.0.3",
    "react-native-video": "^5.2.1",
    "react-native-vision-camera": "^4.0.5",
    "react-native-worklets-core": "^1.3.3",
    "vision-camera-resize-plugin": "^3.1.0",
    "expo": "~50.0.17"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@types/react": "~18.2.79",
    "eas-cli": "^9.1.0",
    "expo": "^50.0.18",
    "expo-dev-client": "~3.3.11",
    "expo-status-bar": "~1.11.1",
    "expo-system-ui": "~2.9.4",
    "tsd": "^0.31.0",
    "typescript": "~5.3.3"
  },
```

imvedanshmehra, 2024-05-31:
I am still having issues with this combination; I am not sure, but is "vision-camera-resize-plugin" also required to make it work?