margelo/react-native-vision-camera #2762: 🐛 [^4.0.0-beta.16] - Camera is not aligned properly when using useSkiaFrameProcessor

karventhan13, 2024-04-20:
### What's happening?

🐛 [^4.0.0-beta.16] - Camera is not aligned properly when using useSkiaFrameProcessor  but it's work with [image]

### Reproduceable Code

```tsx
return (
    <SafeAreaView style={styles.container}>
      {!hasPermission && <Text>No Camera Permission.</Text>}
      {hasPermission && device != null && (
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={true}
          photo={true}
          video={true}
          format={format}
          onError={(err)=> { console.log("err **",err)}}
          fps={fps}
          enableFpsGraph={true}
          frameProcessor={frameProcessor}  
          videoHdr={format.supportsVideoHdr}
          photoHdr={format.supportsPhotoHdr}
          enableZoomGesture={false}
        />
      )}
    </SafeAreaView>
  );
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
  },
});
```

### Relevant log output

```shell
none
```

### Camera Device

```json
none
```

### Device

vivo y21

### VisionCamera Version

[^4.0.0-beta.16]

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-21:
I will never understand why people ignore the fact that I need log output. I will write a bot in the future to auto-close issues that don't have valid log output.

mrousavy, 2024-04-21:
Maybe play around with `useSkiaFrameProcessor` hook, specifically the `render()` function in there. If you can figure out what's wrong, please submit a PR.
I'm on vacation this week

Space6188, 2024-04-22:
This happened when you using           resizeMode={'contain'}

Try to delete this from Camera props

mrousavy, 2024-04-22:
@Space6188 does it properly display without the `resizeMode`?

mrousavy, 2024-04-22:
I can also reproduce this when using `resizeMode="cover"`...

mrousavy, 2024-04-22:
Try playing around with the code from `useSkiaFrameProcessor`: https://github.com/mrousavy/react-native-vision-camera/blob/3ff02cc6b4fc34154ec6fb5e650e074e34408e47/package/src/skia/useSkiaFrameProcessor.ts#L165-L175

Maybe we're doing something wrong.

Also, the `ImageProxy` (which is what the native `Frame` type holds) contains a [`getSensorToBufferTransformMatrix()`](https://developer.android.com/reference/androidx/camera/core/ImageInfo#getSensorToBufferTransformMatrix()) method - maybe we can use this Matrix to transform the Frame safely to view dimensions? cc @rodgomesc if you wanna look into this.

rodgomesc, 2024-04-22:
> I will never understand why people ignore the fact that I need log output. I will write a bot in the future to auto-close issues that don't have valid log output.

then you will have zero issues in vc 😅

rodgomesc, 2024-04-22:
i tried all resizeMode possibilities, also all fit possibilities, the problem is not there

mrousavy, 2024-04-22:
then it's probably the sensor to buffer transform matrix.

Space6188, 2024-04-22:
> canvas.save() 
>  const rotation = getRotationDegrees(frame.orientation) 
>  canvas.rotate(rotation, frame.width / 2, frame.height / 2) 
>  // render the Camera Frame to the Canvas 
>  if (paint != null) canvas.drawImage(image, 0, 0, paint) 
>  else canvas.drawImage(image, 0, 0) 
>   
>  // restore transforms/rotations again 
>  canvas.restore()

No, my bad, sorry 
That also not working

Space6188, 2024-04-22:
> @Space6188правильно ли он отображается без `resizeMode`?

No,my bad,sorry, it is not working too

rodgomesc, 2024-04-29:
i think that i found a fix for this, should send a pr some time tomorrow

rodgomesc, 2024-04-30:
it seems that `canvas.drawImage` draws the image in a clip and matrix that we don't know what the default is? or maybe it get messed up by the canvas rotation, if we use `canvas.drawImageRect`,  did some tests using it and it seems ok to me

<details>
<summary>Mock Diff</summary>

```diff
diff --git a/package/src/skia/useSkiaFrameProcessor.ts b/package/src/skia/useSkiaFrameProcessor.ts
index 3feb336..fd69ca1 100644
--- a/package/src/skia/useSkiaFrameProcessor.ts
+++ b/package/src/skia/useSkiaFrameProcessor.ts
@@ -6,8 +6,34 @@ import { VisionCameraProxy, wrapFrameProcessorWithRefCounting } from '../FramePr
 import type { DrawableFrameProcessor } from '../types/CameraProps'
 import type { ISharedValue, IWorkletNativeApi } from 'react-native-worklets-core'
 import { WorkletsProxy } from '../dependencies/WorkletsProxy'
-import type { SkCanvas, SkPaint, SkImage, SkSurface } from '@shopify/react-native-skia'
+import { type SkCanvas, type SkPaint, type SkImage, type SkSurface, ColorType, AlphaType, Skia } from '@shopify/react-native-skia'
 import { SkiaProxy } from '../dependencies/SkiaProxy'
+import { InteractionManager } from 'react-native'
+
+// Mocked Frame Data for testings
+const mockedImageWidth = 1920
+const mockedImageHeight = 1080
+
+const pixels = new Uint8Array(mockedImageWidth * mockedImageHeight * 4)
+pixels.fill(255)
+let i = 0
+for (let x = 0; x < mockedImageWidth; x++) {
+  for (let y = 0; y < mockedImageHeight; y++) {
+    pixels[i++] = (x * y) % 255
+  }
+}
+
+const data = Skia.Data.fromBytes(pixels)
+const image = Skia.Image.MakeImage(
+  {
+    width: mockedImageWidth,
+    height: mockedImageHeight,
+    alphaType: AlphaType.Opaque,
+    colorType: ColorType.RGBA_8888,
+  },
+  data,
+  mockedImageWidth * 4,
+)!
 
 /**
  * Represents a Camera Frame that can be directly drawn to using Skia.
@@ -155,7 +181,7 @@ export function createSkiaFrameProcessor(
 
     // Convert Frame to SkImage/Texture
     const nativeBuffer = (frame as FrameInternal).getNativeBuffer()
-    const image = Skia.Image.MakeImageFromNativeBuffer(nativeBuffer.pointer)
+    // const image = Skia.Image.MakeImageFromNativeBuffer(nativeBuffer.pointer)
 
     return new Proxy(frame as DrawableFrame, {
       get: (_, property: keyof DrawableFrame) => {
@@ -165,11 +191,36 @@ export function createSkiaFrameProcessor(
             'worklet'
             // rotate canvas to properly account for Frame orientation
             canvas.save()
-            const rotation = getRotationDegrees(frame.orientation)
-            canvas.rotate(rotation, frame.width / 2, frame.height / 2)
-            // render the Camera Frame to the Canvas
-            if (paint != null) canvas.drawImage(image, 0, 0, paint)
-            else canvas.drawImage(image, 0, 0)
+
+            const rotationAngle = 0 //getRotationDegrees('portrait')
+            const { width: frameWidth, height: frameHeight } = getPortraitSize(frame)
+
+            // center of the canvas
+            const centerX = frameWidth / 2
+            const centerY = frameHeight / 2
+
+            // define the paint for the rectangle's border
+            const currentPaint = paint ?? Skia.Paint()
+
+            // only draw the rect by 40% of the frame size so we can vizualize the whole thing to debug :)
+            const reductionFactor = 0.4
+            const rectWidth = frameWidth * reductionFactor
+            const rectHeight = frameHeight * reductionFactor
+
+            // apply rotation around the center of the canvas
+            canvas.rotate(rotationAngle, centerX, centerY)
+
+            const rectX = centerX - rectWidth / 2
+            const rectY = centerY - rectHeight / 2
+
+            const srcRect = Skia.XYWHRect(0, 0, image.width(), image.height())
+            // Define the destination rectangle on the canvas
+            const dstRect = Skia.XYWHRect(rectX, rectY, rectWidth, rectHeight)
+            // Draw the image on the canvas
+
+            canvas.drawImageRect(image, srcRect, dstRect, currentPaint)
 
             // restore transforms/rotations again
             canvas.restore()
@@ -185,7 +236,7 @@ export function createSkiaFrameProcessor(
           return () => {
             'worklet'
             // dispose the Frame and the SkImage/Texture
-            image.dispose()
+            // image.dispose() // we do not dispose our mocked image :)
             nativeBuffer.delete()
           }
         }

```
</details>
<table>
<tr>
    <td>
        <p align="center"><b>portrait 0deg</b></p>
        <p align="center">[image]</p>
    </td>
    <td>
        <p align="center"><b>landscape-left 90deg</b></p>
        <p align="center">[image]</p>
    </td>
</tr>
<tr>
    <td>
        <p align="center"><b>portrait-upside-down 180deg</b></p>
        <p align="center">[image]portrait-upside-down" width="80%"/></p>
    </td>
    <td>
        <p align="center"><b>landscape-right 270deg</b></p>
        <p align="center">[image]</p>
    </td>
</tr>
</table>

> **Note:** I'm doing some tricks to print the preview centered and at 40% of its original size. However, at the end of the day, this code simplifies to:
>
> ```tsx
> const srcRect = Skia.XYWHRect(0, 0, frameWidth, frameHeight);
> const dstRect = Skia.XYWHRect(0, 0, frameWidth, frameHeight);
> canvas.drawImageRect(image, srcRect, dstRect, currentPaint);
> ```

rodgomesc, 2024-04-30:
Oh wait, this solves the misalignment of the frame inside the canvas; however, it causes another issue: the frame preview is now rotated 90 degrees clockwise for me. I think this is the default since we are lacking orientation support for Android, @mrousavy, right?

mrousavy, 2024-04-30:
I am applying the orientation inside the `render(..)` function, so actually it is expected that the Preview looks correct.

rodgomesc, 2024-04-30:
> I am applying the orientation inside the `render(..)` function, so actually it is expected that the Preview looks correct.

i mean, this always returns `landscape-right` since[ orientation support not being complete on android](https://github.com/mrousavy/react-native-vision-camera/issues/1891) so i can't visualize the full picture here, of how did you got the image buffers always in `portrait` without a counter-clockwise rotation of `-90deg`

https://github.com/mrousavy/react-native-vision-camera/blob/a53166b01dd08678a354ad81fc04bf9f67bee9f3/package/android/src/main/java/com/mrousavy/camera/frameprocessors/Frame.java#L89-L93

pweglik, 2024-05-08:
Any news on this? I'm also experiencing this misplacement. I observed that it depends on format and video resolution:

1. Without passing format props to Camera:
[image]

2. Default format got from `useCasmeraFormat()`:
```
  const format = useCameraFormat(device, []);
  console.log(format);
```
Log:
```
{"autoFocusSystem": "contrast-detection", "fieldOfView": 80.98175244802063, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 100, "photoHeight": 2160, "photoWidth": 4096, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": ["off", "cinematic"], "videoWidth": 3840}

```
[image]

3. 
```
const format = useCameraFormat(device, [
    { videoResolution: { width: 600, height: 400 } },
  ]);
  console.log(format);
```
Log:
```
{"autoFocusSystem": "contrast-detection", "fieldOfView": 80.98175244802063, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 100, "photoHeight": 2160, "photoWidth": 4096, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 480, "videoStabilizationModes": ["off", "cinematic"], "videoWidth": 640}
```
[image]

4.
```
const format = useCameraFormat(device, [
    { videoResolution: { width: 1200, height: 800 } },
  ]);
  console.log(format);
```
Log:
```
{"autoFocusSystem": "contrast-detection", "fieldOfView": 80.98175244802063, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 100, "photoHeight": 2160, "photoWidth": 4096, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": ["off", "cinematic"], "videoWidth": 1280}
```
[image]
[image]

So in case of lower resolutions (default or set manually) only bottom part is cut off, while in higher resolutions entire frame is misplacement. Maybe that somehow helps you

OleksiiZdaly, 2024-05-14:
> Є новини з цього приводу? Я також відчуваю це неправильне розташування. Я помітив, що це залежить від формату та роздільної здатності відео:
> 
> 1. Без передачі параметрів формату в камеру:
> 
> <img alt="" width="280" height="640" src="https://private-user-images.githubusercontent.com/36445788/328806213-13f1365a-f7df-4302-bc45-8cfdacba39cc.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MTI3OTcsIm5iZiI6MTcxNTYxMjQ5NywicGF0aCI6Ii8zNjQ0NTc4OC8zMjg4MDYyMTMtMTNmMTM2NWEtZjdkZi00MzAyLWJjNDUtOGNmZGFjYmEzOWNjLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDE1MDEzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY0MTkyOTk3MTI1NjIxMzZlNGZiMDg0YzI3Yjc3YmRmYjlmMTE3YTA5NWExMWVlZjE2ZjRlYzZiYzkwOTBmYjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IVp416q9eMBFFSaxJgL7rBQE2ud8Z0nteOhHGpuSlnc">
> 2. Формат за замовчуванням отримано з `useCasmeraFormat()`:
> 
> ```
>   const format = useCameraFormat(device, []);
>   console.log(format);
> ```
> 
> Журнал:
> 
> ```
> {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.98175244802063, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 100, "photoHeight": 2160, "photoWidth": 4096, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 2160, "videoStabilizationModes": ["off", "cinematic"], "videoWidth": 3840}
> ```
> 
> <img alt="" width="280" height="640" src="https://private-user-images.githubusercontent.com/36445788/328806257-e1271238-78cb-412b-80e9-0d1903eb9c9d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MTI3OTcsIm5iZiI6MTcxNTYxMjQ5NywicGF0aCI6Ii8zNjQ0NTc4OC8zMjg4MDYyNTctZTEyNzEyMzgtNzhjYi00MTJiLTgwZTktMGQxOTAzZWI5YzlkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDE1MDEzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk5ZGNhN2Q1MjAzNTc1OWVhMWQwMTNjZGNmN2VlNzVkMmY0YTcyNDExN2MzYjU1M2FiOTIyOGRkMWU5ZTJiOWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.hYKmv9iCeszlCZaZp_-TOuijFLgfhkcQ4iq2-OzpyH4">
> 
> ```
> const format = useCameraFormat(device, [
>     { videoResolution: { width: 600, height: 400 } },
>   ]);
>   console.log(format);
> ```
> 
> Журнал:
> 
> ```
> {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.98175244802063, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 100, "photoHeight": 2160, "photoWidth": 4096, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 480, "videoStabilizationModes": ["off", "cinematic"], "videoWidth": 640}
> ```
> 
> <img alt="" width="280" height="640" src="https://private-user-images.githubusercontent.com/36445788/328806292-1e1908dc-9496-4acc-946c-10a743038a89.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MTI3OTcsIm5iZiI6MTcxNTYxMjQ5NywicGF0aCI6Ii8zNjQ0NTc4OC8zMjg4MDYyOTItMWUxOTA4ZGMtOTQ5Ni00YWNjLTk0NmMtMTBhNzQzMDM4YTg5LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDE1MDEzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRhZDFlY2VlYmNjZGQ0NDQ1MzJlNmUzZmZiNTc0YjZhMmRiNzBhMzQwZWIxMzliZGNiZDhjYmQzZjEzZmJmNjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.aLhaheRaa0jyh9kvKWy6zpdwgQHPJY41zgNWMaKviWY">
> 
> ```
> const format = useCameraFormat(device, [
>     { videoResolution: { width: 1200, height: 800 } },
>   ]);
>   console.log(format);
> ```
> 
> Журнал:
> 
> ```
> {"autoFocusSystem": "contrast-detection", "fieldOfView": 80.98175244802063, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 100, "photoHeight": 2160, "photoWidth": 4096, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": ["off", "cinematic"], "videoWidth": 1280}
> ```
> 
> <img alt="" width="280" height="640" src="https://private-user-images.githubusercontent.com/36445788/328806327-cbe4d179-3c88-44d2-8b3d-8c58b819580c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MTI3OTcsIm5iZiI6MTcxNTYxMjQ5NywicGF0aCI6Ii8zNjQ0NTc4OC8zMjg4MDYzMjctY2JlNGQxNzktM2M4OC00NGQyLThiM2QtOGM1OGI4MTk1ODBjLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDE1MDEzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVhNjBmM2JjMzJkM2E4ZTc4MTZkMjQzOTAxNTgzOTE1MDUzMmJiYjc4YzI1MTk1Nzk1ZWY0Zjc2NzQxNzY4ZDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.b72ZUFQv-03-GrYV_e0tx5uxy0dL_TdqMewCeI3RFcQ"> [image]
> Таким чином, у разі нижчої роздільної здатності (за замовчуванням або встановленої вручну) обрізається лише нижня частина, тоді як у вищій роздільній здатності зміщується весь кадр. Можливо, це якось тобі допоможе

For some reason, the format parameters don't change anything for me, the behavior is the same as yours in the second example. And the same problem with useSkiaFrameProcessor. On the emulator, the screen is completely black, and on real devices, part of it is. Another strange thing is that the frame itself processes frames rotated 90 degrees, I noticed this when my model started drawing rectangle around object on a black screen turned to the horizontal position. In IOS all ok.

mrousavy, 2024-05-21:
> Another strange thing is that the frame itself processes frames rotated 90 degrees

That's because of how Cameras work. Sensors are in 90deg orientation. See `frame.orientation`.

tomerh2001, 2024-05-29:
I also encountered this when using `useSkiaFrameProcessor,` but `useFrameProcessor` works

EnzoDomingues, 2024-06-02:
Same issue

mrousavy, 2024-06-03:
Created a fix for this here; https://github.com/mrousavy/react-native-vision-camera/pull/2931

Can you guys test if this works for you? :)

EnzoDomingues, 2024-06-03:
> Created a fix for this here; #2931
> 
> Can you guys test if this works for you? :)
@mroysavy
Hey, for me, it's even worse 
[image]

I'm using Moto G84

EnzoDomingues, 2024-06-03:
@mrousavy 
I was using this format before
`  const format = useCameraFormat(device, [
    {
      videoResolution: {
        width: 480,
        height: Platform.OS === 'ios' ? 640 : 720,
      },
    },
  ]);`
  
Using this format, even the colors are changed

[image]

rodgomesc, 2024-06-03:
@EnzoDomingues can you leave the format, aspect ration configurations at the default or use them exactly like in this example app and send a new screenshot?

EnzoDomingues, 2024-06-03:
> @EnzoDomingues can you leave the format, aspect ration configurations at the default or use them exactly like in this example app and send a new screenshot?

The first screenshot is default, without any format

mrousavy, 2024-06-03:
Hm. Matrixes are really hard to get right.

rodgomesc, 2024-06-04:
> > @EnzoDomingues can you leave the format, aspect ration configurations at the default or use them exactly like in this example app and send a new screenshot?
> 
> The first screenshot is default, without any format

yup, i can confirm, it's worse for me as well

[image]

rodgomesc, 2024-06-04:
@EnzoDomingues can you confirm if applying this patch https://github.com/mrousavy/react-native-vision-camera/issues/2762#issuecomment-2084282889 the frame keeps centered for you?

closed 2024-06-04 by mrousavy

mrousavy, 2024-06-04:
Try again now - I think it now works :)

EnzoDomingues, 2024-06-04:
@mrousavy I added your changes locally and the video is not showing anymore

[image]

EnzoDomingues, 2024-06-04:
> @EnzoDomingues can you confirm if applying this patch [#2762 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/2762#issuecomment-2084282889) the frame keeps centered for you?

@rodgomesc, with your patch
[image]

mrousavy, 2024-06-04:
@EnzoDomingues can you log `frame.orientation`?

EnzoDomingues, 2024-06-04:
@mrousavy  sure
```
 LOG  landscape-right
 LOG  landscape-right
 LOG  landscape-right
```

mrousavy, 2024-06-04:
Ah that's why! Can you try to adjust this code here: https://github.com/mrousavy/react-native-vision-camera/blob/cd183680753837ed5155ca850d7f403ba8ce723c/package/src/skia/useSkiaFrameProcessor.ts#L75-L78

I don't have a phone where it's sensor orientation is `landscape-right`, so I cannot test this myself. Maybe just switch out the translations, use `0` or whatever - if you find a fix pls let me know.

mrousavy, 2024-06-04:
oh lol after posting this I just realized I used the wrong function.... lmao see https://github.com/mrousavy/react-native-vision-camera/commit/38c6983f367524aaefb6bbc2557fe161d7db99ad

now that should work

EnzoDomingues, 2024-06-04:
@mrousavy  
```
   // rotate two flips on (0,0) origin and move X + Y into view again
        canvas.translate(0, frame.width)
        canvas.rotate(270, 0, 0)
```
This works for me instead `canvas.translate(frame.height, frame.width) `
Could you update?

EnzoDomingues, 2024-06-04:
[image]

<3

mrousavy, 2024-06-04:
aha interesting... can you send a PR?

EnzoDomingues, 2024-06-04:
@mrousavy MR created :)

tomerh2001, 2024-06-05:
Is there a release containing this fix?

mrousavy, 2024-06-05:
Not yet.

tomerh2001, 2024-06-05:
> Not yet.

Waiting on this for prod release

pke, 2024-06-08:
I wonder @mrousavy why your sample app does not seem to have this issue? My app is in Portrait mode and OS is locked in rotation too but the SKIA preview is a zoomed rotated one.

mrousavy, 2024-06-08:
@pke are you using the current main branch? As I just said above; there is no release for this yet. It's on main, but not on npm.

pke, 2024-06-09:
Tried to use `"react-native-vision-camera": "mrousavy/react-native-vision-camera#bd955b5"` in package file, which removes the POD when running `bundle exec pod install` :/

opassion, 2024-06-11:
What's the plan to release this one? @mrousavy

PR #2939 (2024-06-04): fix: Fixed canvas.translate on withRotatedFrame for landscape-right

## What
 This PR fixes `canvas.translate` on `withRotatedFrame` for `landscape-right`

## Changes
* This PR changes the canvas.translate(0, frame.width)

## Tested on
* iPhone SE, iOS 14.4
*  iPhone 13 Pro, iOS 14.4
* Moto G84, Android 13
    
## Related issues
 Fixes #2762