margelo/react-native-vision-camera #2763: 🐛 🐛 [^4.0.0-beta.16] - camera is very slow and lower FPS with useSkiaFrameProcessor and detect faces using react-native-vision-camera-face-detector

karventhan13, 2024-04-20:
### What's happening?

🐛 🐛 [^4.0.0-beta.16] - camera very slow and lower FPS with useSkiaFrameProcessor when using detect faces usingreact-native-vision-camera-face-detector

### Reproduceable Code

```tsx
const { detectFaces } = useFaceDetector({
    performanceMode: 'accurate',
    contourMode: 'all',
    landmarkMode: 'all',
    classificationMode: 'all',
    trackingEnabled : true
  })

 const handleFacesDetection = Worklets.createRunOnJS((
    result : DetectionResult
  ) => { 
    console.log( 'detection result', result )
})
  const frameProcessor = useSkiaFrameProcessor((frame) => {
    'worklet'

    frame.render() 

    const result = detectFaces({
      frame
    },
    ) 
    handleFacesDetection(result) 
 
  }, [])

  return (
    <SafeAreaView style={styles.container}>
      {!hasPermission && <Text>No Camera Permission.</Text>}
      {hasPermission && device != null && (
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={true}
          photo={true}
          video={true}
          format={format}
          onError={(err)=> { console.log("err **",err)}}
          fps={fps}
          enableFpsGraph={true}
          frameProcessor={frameProcessor}  
          videoHdr={format.supportsVideoHdr}
          photoHdr={format.supportsPhotoHdr}
          enableZoomGesture={false}
        />
      )}
    </SafeAreaView>
  );
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
  },
});
```

### Relevant log output

```shell
noe
```

### Camera Device

```json
noe
```

### Device

vivo y21

### VisionCamera Version

[^4.0.0-beta.16]

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

anudit, 2024-04-20:
+1 Bottlenecking on ~15 FPS on Pixel 8

```json
"expo": "~50.0.17",
"react": "18.2.0",
"@shopify/react-native-skia": "^1.2.3",
"react-native": "0.73.7",
"react-native-reanimated": "^3.8.1",
"react-native-vision-camera": "4.0.0-beta.16",
"react-native-vision-camera-face-detector": "^1.5.0",
"react-native-worklets-core": "^1.1.1",
```

mrousavy, 2024-04-21:
No log output, no example video, no camera device, not reproduced in example, nothing.
Closing issue due to lack of information.

closed 2024-04-21 by mrousavy

diogeneshamilton, 2024-08-23:
I am also having this same issue, only using contourMode: "all" (landmarkMode works fast enough but we want to use all of the contours for filters). Any way to get this to render faster? basically, the mask doesn't follow the face fast enough (gets abt 4 FPS): 

```
import { Skia } from "@shopify/react-native-skia";
import { useEffect } from "react";
import { StyleSheet, Text, View } from "react-native";
import {
  Camera,
  runAsync,
  useCameraDevice,
  useSkiaFrameProcessor,
} from "react-native-vision-camera";
import {
  Face,
  useFaceDetector,
} from "react-native-vision-camera-face-detector";
import { useSharedValue } from "react-native-worklets-core";

export default function Index() {
  const faces = useSharedValue<Face[]>([]);

  const { detectFaces } = useFaceDetector({
    landmarkMode: "all",
    contourMode: "all",
    trackingEnabled: true,
  });

  const frameProcessor = useSkiaFrameProcessor((frame) => {
    "worklet";

    frame.render();
    // use Skia to draw dots for each key point in each face's landmarks
    for (const face of faces.value) {
      const landmarks = face.landmarks;
      const contours = face.contours;
      if (landmarks) {
        (Object.keys(landmarks) as Array<keyof typeof landmarks>).forEach(
          (key) => {
            const point = landmarks[key];
            const paint = Skia.Paint();
            paint.setColor(Skia.Color("red"));

            frame.drawCircle(point.x, point.y, 5, paint);
          }
        );
      }
      if (contours) {
        // do the same for contours
        (Object.keys(contours) as Array<keyof typeof contours>).forEach(
          (key) => {
            // for each Point in the contour, draw a circle
            const contour = contours[key];
            const paint = Skia.Paint();
            paint.setColor(Skia.Color("green"));
            for (const point of contour) {
              frame.drawCircle(point.x, point.y, 5, paint);
            }
          }
        );
      }
    }
    runAsync(frame, () => {
      "worklet";
      faces.value = detectFaces(frame);
    });
  }, []);

  const device = useCameraDevice("front");

  useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      console.log({ status });
    })();
  }, [device]);

  return (
    <View style={{ flex: 1 }}>
      {!!device ? (
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          frameProcessor={frameProcessor}
          isActive={true}
          enableFpsGraph={true}
        />
      ) : (
        <Text>No Device</Text>
      )}
    </View>
  );
}

```

Kishore180994, 2024-09-17:
I am having the same issue as well, For the first 15 - 20 seconds, FPS is 2-4, but after that FPS is smooth and consistent with 60 FPS rest of the time.

How to handle the first 15 seconds? 

When I use useFrameProcessor instead of useSkiaFrameProcessor, it is fast as hell. But I need Skia for my project!

These are my settings.
```
 const faceDetectionOptions = useRef<FaceDetectionOptions>({
    performanceMode: 'fast',
    contourMode: 'all', // Using contours for face outline
    landmarkMode: 'none',
    classificationMode: 'none',
  }).current;
  ```

ChristopherGabba, 2024-09-29:
All, I can confirm that this is also happening on my Android Google Pixel 4. 

I'm not using `useSkiaFrameProcessor`, I'm using `useFrameProcessor` + react-native-face-detector and my android video recording frame rate plummets and the preview skips. I'll work on getting all logs, etc. @mrousavy would your preference be to open a fresh issue? 

```typescript
  const frameProcessor = useFrameProcessor(
    (frame) => {
      "worklet"
      const faces = detectFaces(frame)
      // ... chain frame processors
      // ... do something with frame
      handleDetectedFaces(faces)
    },
    [handleDetectedFaces],
  )

              <Camera
                ref={cameraRef}
                video={true}
                audio={true}
                lowLightBoost={device?.supportsLowLightBoost}
                photo={true}
                format={format}
                isActive={isActive}
                device={device}
                exposure={0}
                outputOrientation="preview"
                frameProcessor={frameProcessor} // Commenting out this line of code the preview is way smoother 
                style={$camera}
                onPreviewStarted={onPreviewStarted}
                onError={onCameraError}
              />
```

ska-anonymous, 2025-03-03:
I had the same issue of camera view lagging when using useSkiaFrameProcessor.
In my case the problem is not even due to face detection or other heavy processes running synchronously, I am just rendering the frame but still the camera lags

const frameProcessor = useSkiaFrameProcessor((frame) => {
        'worklet'
        frame.render()
    }, [])

and interestingly what if found that when I changed the pixelFormat to rgb, the problem was resolved and there is no more lag in the cameraview

<Camera
   ......
   pixelFormat='rgb'
/>