margelo/react-native-vision-camera #2773: 🐛

Swikriti0998, 2024-04-22:
### What's happening?

I am trying to read the code using codeScanner, incase that doesn't work I am taking a picture and run OCR on it to fetch the code. My application is crashing when closing the camera and moving to a different screen.
I have tried to disable the isActive, conditionally render the camera, add error function to fix that but nothing seems to work.

### Reproduceable Code

```tsx
<Camera
codeScanner={code_scanner}
onError={handle_error}
onInitialized={onCameraInitialized}
device={device}
isActive={isActive}
photo={true}
ref={cameraRef}
style={styles.camera}
torch={torch ? 'on' : 'off'}
zoom={zoom}
format={format}
/>
```

### Relevant log output

```shell
Error- [unknown/unknown: [unknown/unknown] Image is already closed]
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 10,
  "minZoom": 1,
  "maxExposure": 18,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -18,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

OnePlus 9 Pro

### VisionCamera Version

3.9.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

closed 2024-04-22 by mrousavy

Swikriti0998, 2024-04-23:
It is still crashing with the given error-

Your app just crashed. See the error below.
java.lang.IllegalStateException: Image is already closed
  android.media.Image.throwISEIfImageIsInvalid(Image.java:79)
  android.media.ImageReader$SurfaceImage.getFormat(ImageReader.java:1181)
  com.google.mlkit.vision.common.InputImage.zzb(com.google.mlkit:vision-common@@17.3.0:8)
  com.google.mlkit.vision.common.InputImage.fromMediaImage(com.google.mlkit:vision-common@@17.3.0:1)
  com.mrousavy.camera.core.CodeScannerPipeline._init_$lambda$4(CodeScannerPipeline.kt:53)
  com.mrousavy.camera.core.CodeScannerPipeline.$r8$lambda$J9uS5V_QrF4Y46dODUvYa6w1Wm0(Unknown Source:0)
  com.mrousavy.camera.core.CodeScannerPipeline$$ExternalSyntheticLambda0.onImageAvailable(Unknown Source:4)
  android.media.ImageReader$1.run(ImageReader.java:947)
  android.os.Handler.handleCallback(Handler.java:958)
  android.os.Handler.dispatchMessage(Handler.java:99)
  android.os.Looper.loopOnce(Looper.java:257)
  android.os.Looper.loop(Looper.java:368)
  android.os.HandlerThread.run(HandlerThread.java:67)

mrousavy, 2024-04-29:
I fixed this in #2819 - can you try that and see if the issue is fixed for you @Swikriti0998 ?