margelo/react-native-vision-camera #2776: 🐛 v4.0.0-beta.19 Error: Cannot find module './expo-plugin/withVisionCamera'

pablogdcr, 2024-04-22:
### What's happening?

Hi, I just installed the new beta version and got an issue while running `npx expo install`: Error when installing the new beta version using Expo.

It looks like the path in app.plugin.js isn't correct

### Reproduceable Code

```tsx
Error when installing the package
```

### Relevant log output

```shell
Error: Cannot find module './expo-plugin/withVisionCamera'
Require stack:
- /Users/[...]/node_modules/react-native-vision-camera/app.plugin.js
- /Users/[...]/node_modules/@expo/config-plugins/build/utils/plugin-resolver.js
- /Users/[...]/node_modules/@expo/cli/build/src/install/utils/autoAddConfigPlugins.js
- /Users/[...]/node_modules/@expo/cli/build/src/install/applyPlugins.js
- /Users/[...]/node_modules/@expo/cli/build/src/install/installAsync.js
- /Users/[...]/node_modules/@expo/cli/build/src/install/index.js
- /Users/[...]/node_modules/@expo/cli/build/bin/cli
- /Users/[...]/node_modules/expo/bin/cli
```

### Camera Device

```json
No device
```

### Device

No device

### VisionCamera Version

v4.0.0-beta.19

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

pablogdcr, 2024-04-22:
Just tested with the v4.0.0 and got the same error

mrousavy, 2024-04-22:
fixed in 61ed5f16

closed 2024-04-22 by pablogdcr