margelo/react-native-vision-camera #2777: 🐛 create Capture Session By Output Configurations

AladdinMohamed95, 2024-04-22:
### What's happening?

I'm trying to launch the camera i got this error 
{"error": "Camera.onError(unknown/unknown): [unknown/unknown] No virtual method createCaptureSessionByOutputConfigurations(Ljava/util/List;Landroid/hardware/camera2/CameraCaptureSession$StateCallback;Landroid/os/Handler;)V in class Landroid/hardware/camera2/CameraDevice; or its super classes (declaration of 'android.hardware.camera2.CameraDevice' appears in /system/framework/framework.jar)"}  JSExceptionHandler

It seems like there's an issue with configuring the camera session in an Android application. The error message indicates a NoSuchMethodError related to the method createCaptureSessionByOutputConfigurations in the CameraDevice class.

This error typically occurs when the method being called doesn't exist in the version of the Android framework being used. In this case, it seems that the method createCaptureSessionByOutputConfigurations is not available in the current version of the CameraDevice class.

### Reproduceable Code

```tsx
const device = useCameraDevice("back");
 
{device ? (
            <Camera
              style={{ flex: 1 }}
              ref={cameraRef}
              isActive={true}
              device={device}
              photo={true}
            />
          ) : (
            <ActivityIndicator size="large" color="#1C6758" />
          )}
```

### Relevant log output

```shell
2024-04-22 05:01:44.580 ١٨١٦٦-١٩٠٩٤ CameraSession           com.mas.mashh                        E  Failed to configure CameraSession! Error: No virtual method createCaptureSessionByOutputConfigurations(Ljava/util/List;Landroid/hardware/camera2/CameraCaptureSession$StateCallback;Landroid/os/Handler;)V in class Landroid/hardware/camera2/CameraDevice; or its super classes (declaration of 'android.hardware.camera2.CameraDevice' appears in /system/framework/framework.jar), Config-Diff: Difference(deviceChanged=false, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
                                                                                                    java.lang.NoSuchMethodError: No virtual method createCaptureSessionByOutputConfigurations(Ljava/util/List;Landroid/hardware/camera2/CameraCaptureSession$StateCallback;Landroid/os/Handler;)V in class Landroid/hardware/camera2/CameraDevice; or its super classes (declaration of 'android.hardware.camera2.CameraDevice' appears in /system/framework/framework.jar)
                                                                                                    	at com.mrousavy.camera.extensions.CameraDevice_createCaptureSessionKt.createCaptureSession(CameraDevice+createCaptureSession.kt:60)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.configureOutputs(CameraSession.kt:374)
                                                                                                    	at com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:147)
                                                                                                    	at com.mrousavy.camera.core.CameraSession$configure$1.invokeSuspend(CameraSession.kt)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:739)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                                    	at android.os.Looper.loop(Looper.java:148)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:61)
2024-04-22 05:01:44.580 ١٨١٦٦-١٩٠٩٤ CameraView              com.mas.mashh                        E  invokeOnError(...):
2024-04-22 05:01:44.580 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  java.lang.NoSuchMethodError: No virtual method createCaptureSessionByOutputConfigurations(Ljava/util/List;Landroid/hardware/camera2/CameraCaptureSession$StateCallback;Landroid/os/Handler;)V in class Landroid/hardware/camera2/CameraDevice; or its super classes (declaration of 'android.hardware.camera2.CameraDevice' appears in /system/framework/framework.jar)
2024-04-22 05:01:44.580 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  	at com.mrousavy.camera.extensions.CameraDevice_createCaptureSessionKt.createCaptureSession(CameraDevice+createCaptureSession.kt:60)
2024-04-22 05:01:44.580 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  	at com.mrousavy.camera.core.CameraSession.configureOutputs(CameraSession.kt:374)
2024-04-22 05:01:44.580 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  	at com.mrousavy.camera.core.CameraSession.configure(CameraSession.kt:147)
2024-04-22 05:01:44.580 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  	at com.mrousavy.camera.core.CameraSession$configure$1.invokeSuspend(CameraSession.kt)
2024-04-22 05:01:44.581 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
2024-04-22 05:01:44.581 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
2024-04-22 05:01:44.581 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  	at android.os.Handler.handleCallback(Handler.java:739)
2024-04-22 05:01:44.581 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  	at android.os.Handler.dispatchMessage(Handler.java:95)
2024-04-22 05:01:44.581 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  	at android.os.Looper.loop(Looper.java:148)
2024-04-22 05:01:44.581 ١٨١٦٦-١٩٠٩٤ System.err              com.mas.mashh                        W  	at android.os.HandlerThread.run(HandlerThread.java:61)
```

### Camera Device

```json
{
  "hardwareLevel": "legacy",
  "minExposure": -71.99985600028799,
  "neutralZoom": 1,
  "supportsFocus": true,
  "formats": [],
  "supportsLowLightBoost": false,
  "hasTorch": true,
  "supportsRawCapture": false,
  "sensorOrientation": "landscape-right",
  "maxZoom": 6.059999942779541,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "isMultiCam": false,
  "id": "0",
  "position": "back",
  "minZoom": 1,
  "hasFlash": true,
  "maxExposure": 71.99985600028799,
  "name": "BACK (0)"
}
```

### Device

Android CT50 V6.0.1  api level 23 

### VisionCamera Version

3.6.17

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy

AladdinMohamed95, 2024-04-22:
i have updated to v 4.0.1 
 and  i got this error :

 {"error": "Camera.onError(session/invalid-output-configuration): Failed to configure the Camera Session because the output/stream configurations are invalid!"}  JSExceptionHandler

mrousavy, 2024-04-22:
There's nothing I can do without native (adb logcat) logs

AladdinMohamed95, 2024-04-22:
@mrousavy 
2024-04-22 07:36:57.816  ١١١٨-١٤١٦  CameraDeviceUserShim    com.mas.mashh                        E  createDefaultRequest - invalid templateId specified
2024-04-22 07:36:57.816  ١١١٨-١٤١٦  CaptureSession          com.mas.mashh                        W  Opening session with fail OPENING
                                                                                                    java.lang.IllegalArgumentException: Bad argument passed to camera service
                                                                                                    	at android.hardware.camera2.utils.CameraBinderDecorator.throwOnError(CameraBinderDecorator.java:114)
                                                                                                    	at android.hardware.camera2.utils.CameraBinderDecorator$CameraBinderDecoratorListener.onAfterInvocation(CameraBinderDecorator.java:73)
                                                                                                    	at android.hardware.camera2.utils.Decorator.invoke(Decorator.java:81)
                                                                                                    	at java.lang.reflect.Proxy.invoke(Proxy.java:393)
                                                                                                    	at $Proxy13.createDefaultRequest(Unknown Source)
                                                                                                    	at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureRequest(CameraDeviceImpl.java:628)
                                                                                                    	at androidx.camera.camera2.internal.Camera2CaptureRequestBuilder.buildWithoutTarget(Camera2CaptureRequestBuilder.java:213)
                                                                                                    	at androidx.camera.camera2.internal.CaptureSession.openCaptureSession(CaptureSession.java:334)
                                                                                                    	at androidx.camera.camera2.internal.CaptureSession.lambda$open$0$androidx-camera-camera2-internal-CaptureSession(CaptureSession.java:215)
                                                                                                    	at androidx.camera.camera2.internal.CaptureSession$$ExternalSyntheticLambda2.apply(D8$$SyntheticClass)
                                                                                                    	at androidx.camera.core.impl.utils.futures.ChainingListenableFuture.run(ChainingListenableFuture.java:203)
                                                                                                    	at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.workOnQueue(SequentialExecutor.java:231)
                                                                                                    	at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.run(SequentialExecutor.java:173)
                                                                                                    	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
                                                                                                    	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
                                                                                                    	at java.lang.Thread.run(Thread.java:818)
2024-04-22 07:36:57.817  ١١١٨-١٤١٦  Camera2CameraImpl       com.mas.mashh                        D  {Camera@dac031b[id=0]} Transitioning camera internal state: OPENED --> OPENED
2024-04-22 07:36:57.817  ١١١٨-١٤١٦  CameraStateMachine      com.mas.mashh                        D  New public camera state CameraState{type=OPEN, error=StateError{code=4, cause=java.lang.IllegalArgumentException: Bad argument passed to camera service}} from OPEN and StateError{code=4, cause=java.lang.IllegalArgumentException: Bad argument passed to camera service}
2024-04-22 07:36:57.817  ١١١٨-١٤١٦  CameraStateMachine      com.mas.mashh                        D  Publishing new public camera state CameraState{type=OPEN, error=StateError{code=4, cause=java.lang.IllegalArgumentException: Bad argument passed to camera service}}
2024-04-22 07:36:57.817  ١١١٨-١٤١٦  Camera2CameraImpl       com.mas.mashh                        E  Unable to configure camera Camera@dac031b[id=0]
                                                                                                    java.lang.IllegalArgumentException: Bad argument passed to camera service
                                                                                                    	at android.hardware.camera2.utils.CameraBinderDecorator.throwOnError(CameraBinderDecorator.java:114)
                                                                                                    	at android.hardware.camera2.utils.CameraBinderDecorator$CameraBinderDecoratorListener.onAfterInvocation(CameraBinderDecorator.java:73)
                                                                                                    	at android.hardware.camera2.utils.Decorator.invoke(Decorator.java:81)
                                                                                                    	at java.lang.reflect.Proxy.invoke(Proxy.java:393)
[... cut ...]
2024-04-22 07:36:57.864  ١١١٨-١١١٨  System.err              com.mas.mashh                        W  	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
2024-04-22 07:36:57.864  ١١١٨-١١١٨  System.err              com.mas.mashh                        W  	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
2024-04-22 07:36:57.864  ١١١٨-١١١٨  System.err              com.mas.mashh                        W  	at java.lang.Thread.run(Thread.java:818)

iqbaldev46, 2024-04-23:
Hi, facing same issue.

Device: vivo Y55L
react-native-vision-camera: 4.0.1

Logs below.

`04-23 20:38:10.487 16702 24125 E ACodec  : [OMX.qcom.video.encoder.avc] storeMetaDataInBuffers (output) failed w/ err -1010
04-23 20:38:10.487 16702 24125 W ACodec  : do not know color format 0x7fa30c04 = 2141391876
04-23 20:38:10.487 16702 24125 W ACodec  : do not know color format 0x7f000789 = 2130708361
04-23 20:38:10.497   658  6439 E OMX-VENC: ERROR: Unsupported H.264 level= 0
04-23 20:38:10.507   658  6438 E OMX-VENC: ERROR: Unsupported H.264 level= 0
04-23 20:38:10.527 16702 24125 W ACodec  : do not know color format 0x7f000789 = 2130708361
04-23 20:38:10.547   658  6439 E OMXNodeInstance: getParameter(bb:qcom.encoder.avc, ParamConsumerUsageBits(0x6f800004)) ERROR: UnsupportedIndex(0x8000101a)
04-23 20:38:10.647   658 24086 W AMessage: failed to post message as target looper for handler 0 is gone.
04-23 20:38:10.647   658 24092 W AMessage: failed to post message as target looper for handler 0 is gone.
04-23 20:38:10.647   658 24094 W AMessage: failed to post message as target looper for handler 0 is gone.
04-23 20:38:10.707   658  1738 E CameraService: getVivoLogCtrl:persist.sys.log.ctrl is 0
04-23 20:38:10.707   658  1738 E CameraService: getVivoLogCtrl:gLogLevel 0
04-23 20:38:10.727   658  1738 E QCamera :  yangqing g_cam_log[1][1] is 0
04-23 20:38:10.727   658  1738 E QCamera :  yangqing g_cam_log[1][2] is 0
04-23 20:38:10.727   658  1738 E QCamera :  yangqing g_cam_log[1][3] is 0
04-23 20:38:10.727   658  1738 E QCamera :  yangqing g_cam_log[1][4] is 0
04-23 20:38:10.727   658  1738 E QCamera :  yangqing g_cam_log[1][5] is 0
04-23 20:38:10.727   658  1738 E QCamera :  yangqing g_cam_log[1][6] is 0
04-23 20:38:10.787   687 24150 E mm-camera: <SENSOR><ERROR> s5k4h8_rear_eeprom_get_calibration_items: 21: Enter
04-23 20:38:10.787   687 24150 E mm-camera: <SENSOR><ERROR> s5k4h8_rear_eeprom_get_calibration_items: 31: Exit
04-23 20:38:10.787 16702 17201 E CameraDeviceUserShim: createDefaultRequest - invalid templateId specified
04-23 20:38:10.787 16702 17201 W CaptureSession: Opening session with fail OPENING
04-23 20:38:10.787 16702 17201 W CaptureSession: java.lang.IllegalArgumentException: Bad argument passed to camera service
04-23 20:38:10.787 16702 17201 W CaptureSession:        at android.hardware.camera2.utils.CameraBinderDecorator.throwOnError(CameraBinderDecorator.java:114)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at android.hardware.camera2.utils.CameraBinderDecorator$CameraBinderDecoratorListener.onAfterInvocation(CameraBinderDecorator.java:73)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at android.hardware.camera2.utils.Decorator.invoke(Decorator.java:81)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at java.lang.reflect.Proxy.invoke(Proxy.java:393)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at $Proxy15.createDefaultRequest(Unknown Source)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureRequest(CameraDeviceImpl.java:628)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at androidx.camera.camera2.internal.Camera2CaptureRequestBuilder.buildWithoutTarget(Camera2CaptureRequestBuilder.java:213)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at androidx.camera.camera2.internal.CaptureSession.openCaptureSession(CaptureSession.java:334)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at androidx.camera.camera2.internal.CaptureSession.lambda$open$0$androidx-camera-camera2-internal-CaptureSession(CaptureSession.java:215)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at androidx.camera.camera2.internal.CaptureSession$$ExternalSyntheticLambda2.apply(D8$$SyntheticClass)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at androidx.camera.core.impl.utils.futures.ChainingListenableFuture.run(ChainingListenableFuture.java:203)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.workOnQueue(SequentialExecutor.java:231)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.run(SequentialExecutor.java:173)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
04-23 20:38:10.787 16702 17201 W CaptureSession:        at java.lang.Thread.run(Thread.java:818)
04-23 20:38:10.797 16702 16702 E CameraView: invokeOnError(...):
04-23 20:38:10.797 16702 16702 W System.err: com.mrousavy.camera.core.InvalidOutputConfigurationError: Failed to configure the Camera Session because the output/stream configurations are invalid!
04-23 20:38:10.797 16702 16702 W System.err:    at com.mrousavy.camera.core.extensions.StateError_toCameraErrorKt.toCameraError(StateError+toCameraError.kt:23)
04-23 20:38:10.797 16702 16702 W System.err:    at com.mrousavy.camera.core.CameraSession$configureCamera$2.invoke(CameraSession.kt:452)
04-23 20:38:10.797 16702 16702 W System.err:    at com.mrousavy.camera.core.CameraSession$configureCamera$2.invoke(CameraSession.kt:440)
04-23 20:38:10.797 16702 16702 W System.err:    at com.mrousavy.camera.core.CameraSession.configureCamera$lambda$10(CameraSession.kt:440)
[... cut ...]
04-23 20:38:10.797 16702 16702 W System.err:    at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.run(SequentialExecutor.java:173)
04-23 20:38:10.797 16702 16702 W System.err:    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
04-23 20:38:10.797 16702 16702 W System.err:    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
04-23 20:38:10.797 16702 16702 W System.err:    at java.lang.Thread.run(Thread.java:818)`

mrousavy, 2024-04-29:
Interesting, so it seems like `qualityPrioritization="speed"` or `qualityPrioritization="quality"` works. Apparently the `balanced` (ZSL option) doesn't work on all devices, even though CameraX docs suggest it will automatically fallback.

felipejh, 2024-09-20:
I've been facing the issue on the Pixel 8 Pro. I have tried both `qualityPrioritization="speed"` and `qualityPrioritization="quality"`.

That is the issue caught on Sentry:

`Camera error: [session/invalid-output-configuration]: Failed to configure the Camera Session because the output/stream configurations are invalid! (caused by {"stacktrace":"java.lang.IllegalArgumentException: Surface was abandoned\n\tat android.hardware.camera2.utils.SurfaceUtils.getSurfaceSize(SurfaceUtils.java:135)\n\tat android.hardware.camera2.params.OutputConfiguration`

eckler-matthew, 2024-10-08:
I'm also facing this issue now

Nader-CS, 2024-11-14:
same issue

ics-india-tech, 2024-11-21:
Did anyone got the solution for this? I am looking to fix this from last 2 days...