margelo/react-native-vision-camera #2778: 🔧 Build Error with react-native-vision-camera Example App on Windows using VS Code

allorazz, 2024-04-22:
### How were you trying to build the app?

I was trying to build and run the example app for react-native-vision-camera on my Windows machine using Visual Studio Code. I intended to use Expo for development. I followed the instructions from the repository to set up the playground by running the following commands:

git clone https://github.com/mrousavy/react-native-vision-camera
cd react-native-vision-camera/package
yarn bootstrap

However, the build fails with a generic "Command failed with exit code 1," and I am unable to find relevant information regarding this error. 

### Full build logs

```tsx
PS C:\Users\carol\Desktop\VISION> git clone https://github.com/mrousavy/react-native-vision-camera
>> cd react-native-vision-camera/package
>> yarn bootstrap
Cloning into 'react-native-vision-camera'...
remote: Enumerating objects: 23835, done.
remote: Counting objects: 100% (7085/7085), done.
remote: Compressing objects: 100% (2578/2578), done.
remote: Total 23835 (delta 4364), reused 5966 (delta 3511), pack-reused 16750
Receiving objects: 100% (23835/23835), 102.22 MiB | 18.26 MiB/s, done.
Resolving deltas: 100% (15102/15102), done.
yarn run v1.22.22
$ yarn && cd example && yarn && yarn pods
[1/4] Resolving packages...
[2/4] Fetching packages...
[3/4] Linking dependencies...
warning "react-native > @react-native/codegen@0.72.7" has unmet peer dependency "@babel/preset-env@^7.1.6".
warning "react-native > @react-native/codegen > jscodeshift@0.14.0" has unmet peer dependency "@babel/preset-env@^7.1.6".
warning "react-native > @react-native-community/cli > @react-native-community/cli-plugin-metro > metro-react-native-babel-transformer@0.76.8" has unmet peer dependency "@babel/core@*".
warning " > react-native-reanimated@3.8.1" has unmet peer dependency "@babel/core@^7.0.0-0".
warning "react-native-reanimated > @babel/plugin-transform-arrow-functions@7.24.1" has unmet peer dependency "@babel/core@^7.0.0-0".        
warning "react-native-reanimated > @babel/plugin-transform-nullish-coalescing-operator@7.24.1" has unmet peer dependency "@babel/core@^7.0.0-0".
warning "react-native-reanimated > @babel/plugin-transform-optional-chaining@7.24.1" has unmet peer dependency "@babel/core@^7.0.0-0".      
warning "react-native-reanimated > @babel/plugin-transform-shorthand-properties@7.24.1" has unmet peer dependency "@babel/core@^7.0.0-0".   
warning "react-native-reanimated > @babel/plugin-transform-template-literals@7.24.1" has unmet peer dependency "@babel/core@^7.0.0-0".      
warning "react-native-reanimated > @babel/preset-typescript@7.24.1" has unmet peer dependency "@babel/core@^7.0.0-0".
warning "react-native-reanimated > @babel/preset-typescript > @babel/plugin-syntax-jsx@7.24.1" has unmet peer dependency "@babel/core@^7.0.0-0".
warning "react-native-reanimated > @babel/preset-typescript > @babel/plugin-transform-modules-commonjs@7.24.1" has unmet peer dependency "@babel/core@^7.0.0-0".
warning "react-native-reanimated > @babel/preset-typescript > @babel/plugin-transform-typescript@7.24.4" has unmet peer dependency "@babel/core@^7.0.0-0".
warning "react-native-reanimated > @babel/preset-typescript > @babel/plugin-transform-typescript > @babel/helper-create-class-features-plugin@7.24.4" has unmet peer dependency "@babel/core@^7.0.0".
warning "react-native-reanimated > @babel/preset-typescript > @babel/plugin-transform-typescript > @babel/plugin-syntax-typescript@7.24.1" has unmet peer dependency "@babel/core@^7.0.0-0".
warning "react-native-reanimated > @babel/preset-typescript > @babel/plugin-transform-typescript > @babel/helper-create-class-features-plugin > @babel/helper-replace-supers@7.24.1" has unmet peer dependency "@babel/core@^7.0.0".
[4/4] Building fresh packages...
[1/4] Resolving packages...
[2/4] Fetching packages...
[3/4] Linking dependencies...
[4/4] Building fresh packages...
$ cd ios && bundle install && bundle exec pod install
Bundler 2.5.9 is running, but your lockfile was generated with 2.3.22. Installing Bundler 2.3.22 and restarting using that version.
Fetching gem metadata from https://rubygems.org/.
Fetching bundler 2.3.22
Installing bundler 2.3.22
Fetching gem metadata from https://rubygems.org/.......
Fetching minitest 5.17.0
Fetching concurrent-ruby 1.2.0
Fetching public_suffix 4.0.7
Fetching rexml 3.2.5
Fetching zeitwerk 2.6.7
Fetching httpclient 2.8.3
Installing zeitwerk 2.6.7
Installing rexml 3.2.5
Installing public_suffix 4.0.7
Installing minitest 5.17.0
Installing httpclient 2.8.3
Installing concurrent-ruby 1.2.0
Using json 2.6.3
Fetching atomos 0.1.3
Installing atomos 0.1.3
[... 264 lines cut ...]
```

### VisionCamera Version

4.0.1

### Target platforms

iOS

### Operating system

Windows

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

allorazz, 2024-04-22:
Thank you for your prompt response, I was hoping to build with Expo EAS after unpacking the project files.

mrousavy, 2024-04-22:
```
[!] Invalid `Podfile` file: [!] Unable to locate the executable `mkdir`.

 #  from C:/Users/carol/Desktop/VISION/react-native-vision-camera/package/example/ios/Podfile:26
 #  -------------------------------------------
 #    end
 >  end
 #  # Resolve react_native_pods.rb with node to allow for hoisting
 #  -------------------------------------------
```

that path is a Windows path, so you are building on a Windows machine. I don't even know how you managed to run `pod install`?

allorazz, 2024-04-22:
Yes, I am running the project on a windows PC. I didn't run pod install, the only commands I gave were: 

git clone https://github.com/mrousavy/react-native-vision-camera
cd react-native-vision-camera/package
yarn bootstrap

Any advice you have for using EAS to build the Example app would be greatly appreciated!