margelo/react-native-vision-camera #2783: 🔧 Error compiling VisionCameraProxy.kt in RN 0.74.0 on react-native-vision-camera version 4.0.1

aardvarkk, 2024-04-22:
### How were you trying to build the app?

`yarn android`

### Full build logs

```tsx
e: file:///home/user/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:31:79 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///home/user/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:36:19 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///home/user/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:36:47 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///home/user/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:75:73 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change. FAILURE: Build failed with an exception. * What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
```
```

### Project dependencies

```json
"dependencies": {
    "@aws-amplify/auth": "^5",
    "@aws-amplify/storage": "^5",
    "@react-native-async-storage/async-storage": "^1.23.1",
    "@tanstack/react-query": "^4",
    "axios": "^1.6.8",
    "i18next": "^23.11.2",
    "immer": "^10.0.4",
    "lottie-react-native": "^6.7.2",
    "moment-timezone": "^0.5.45",
    "phosphor-react-native": "^2.0.0",
    "react": "18.2.0",
    "react-i18next": "^14.1.0",
    "react-native": "0.74.0",
    "react-native-barcode-builder": "^2.0.0",
    "react-native-bluetooth-classic": "^1.73.0-rc.8",
    "react-native-paper": "^5.12.3",
    "react-native-signature-capture": "^0.4.12",
    "react-native-svg": "^15.1.0",
    "react-native-vision-camera": "^4.0.1",
    "rn-fetch-blob": "^0.12.0",
    "uuid": "^9.0.1",
    "zustand": "^4.5.2"
  },
```
```

### VisionCamera Version

4.0.1

### Target platforms

Android

### Operating system

Linux

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-22:
VisionCamera does not yet support react-native 0.74.

I've been working sleeplessy the past few weeks to get VisionCamera 4.0.0 released which contains a full rewrite of the Android codebase and Skia Frame Processors, and now I'm off to a vacation for a week.

For react-native 0.74 I want to migrate VisionCamera to new arch (so TurboModule + Fabric + CodeGen), but this is a huge effort and also I am blocked: I have one prop (`frameProcessor`) which needs custom JSI conversion - I don't want the `jsi::Function` to be converted to an `RCTDirectEventBlock`, but rather to a custom type (`RNWorklet`).
I am not sure if that is supported by CodeGen/Fabric/TurboModules, so currently (for Bridge/Paper) I have a very nasty solution where I inject a `global.setFrameProcessor(viewTag, frameProcessor)` function. 
More info: <https://github.com/reactwg/react-native-new-architecture/discussions/167#discussioncomment-8938596>

Once that is resolved or if facebook can give me any pointers on how I could implement this clean, I could try to migrate VisionCamera to new arch / react-native 0.74.
But I don't know _when_ I'm gonna do that.

If someone wants to pay me to get VisionCamera migrated to new arch / react-native 0.74, contact me on https://margelo.io.

aardvarkk, 2024-04-22:
Thanks for the quick response! For now I am downgrading back to 0.73.7. Enjoy your vacation!

AlixH, 2024-04-23:
Well deserved rest, thank you for your work !

mrousavy, 2024-04-23:
see https://github.com/mrousavy/react-native-vision-camera/issues/2614

usyadav007, 2024-04-26:
> Well deserved rest, thank you for your work !

how to downgrade reacct native

usyadav007, 2024-04-26:
simply downgrade your react native version with npm install react-native@0.73.6 from 0.74.0 and it really works 26-04-2024

hengkx, 2024-05-06:
Are there any other temporary solutions besides downgrading?

danieloprado, 2024-05-07:
Workaround: https://github.com/mrousavy/react-native-vision-camera/issues/2851#issuecomment-2099325462

mrousavy, 2024-05-08:
Hey - this is a duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/2614.

New architecture support (which also implies react-native 0.74 support) is tracked in that issue. If you need this, please sponsor the issue/fund the development of that major refactor/migration by donating to the Polar pool mentioned in the issue above. 

Thank you.

mrousavy, 2024-05-08:
The workaround is fine as a patch btw if you need it to build on RN 74 now.

closed 2024-05-08 by mrousavy

BulilKhan77, 2024-05-21:
> The Best Solution is to simply downgrade your react native version otherwise you will stuck in a deep well and will not be able to escape. Just downgrade with npm install react-native@0.73.6 from 0.74.0 and it really works 21-05-2024 >
Just run these commands
1.  cd android 
2. ./gradlew clean 
3. cd .. 
4. npx react-native start --reset-cache 
5. and in another terminal type , **npm run android**
> **Alternate Option**
1. type ./android/gradlew clean
2. npx react-native start --reset-cache
3. close everthing and type **npm run android** or **npx react-native run-android**
>It works

coderjinhui, 2024-05-22:
so why this information not in the document?

mrousavy, 2024-05-22:
What document? I explained right here that VisionCamera doesn't work with RN 0.74 yet.
Also there's a pinned issue about this on top of the issue page.

GeekyCoder25, 2024-05-23:
[image]

danieloprado, 2024-05-23:
> I've added the lines for the patch but it still won't work, pls is there any workaround for this
> 
> ![image](https://private-user-images.githubusercontent.com/96173120/333004871-32e02669-1bf8-4a89-a8c6-4dc4351dc791.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0NDc4NjcsIm5iZiI6MTcxNjQ0NzU2NywicGF0aCI6Ii85NjE3MzEyMC8zMzMwMDQ4NzEtMzJlMDI2NjktMWJmOC00YTg5LWE4YzYtNGRjNDM1MWRjNzkxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIzVDA2NTkyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2Mjg1ZDYzMjgwM2UxM2EzYjFlZWU5MDliZWQ5NmM3ODI2NDM4MTVjNDRlNGQxMWFhZjFkZjcxYjZlYjAyZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yWRJGD17ZRHLPIj_s1nHBD_8FTRfvm0OKI2K_PdKISc)

Remove the + before the @OptIn