margelo/react-native-vision-camera #2786: Android Build with EAS Fails with V4.0.1 and vision-camera-resize-plugin🐛

AndyRC6, 2024-04-23:
### What's happening?

After upgrading to version 4, creating an Android development build using EAS fails as long as the resize plugin is included in the project. If I downgrade back to version 3, it builds just fine. It also builds successfully with version 4 if the resize plugin is removed from the project.

This is an Expo managed project using Development Builds. I had great success implementing Version 3 along with the tflite/resize plugin. This is the only issue I've encountered so far while upgrading to version 4 that's blocking me.

### Reproduceable Code

```tsx
//package.json relevant package versions
"react-native": "0.73.6",
"react-native-vision-camera": "^4.0.1",
"react-native-worklets-core": "^1.1.1",
"react-native-fast-tflite": "^1.2.0",
"vision-camera-resize-plugin": "^2.1.1"

//Build.gradle
buildscript {
    ext {
        buildToolsVersion = findProperty('android.buildToolsVersion') ?: '34.0.0'
        minSdkVersion = Integer.parseInt(findProperty('android.minSdkVersion') ?: '26')
        compileSdkVersion = Integer.parseInt(findProperty('android.compileSdkVersion') ?: '34')
        targetSdkVersion = Integer.parseInt(findProperty('android.targetSdkVersion') ?: '34')
        kotlinVersion = findProperty('android.kotlinVersion') ?: '1.8.10'

        ndkVersion = "25.1.8937393"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath('com.android.tools.build:gradle')
        classpath('com.facebook.react:react-native-gradle-plugin')
        classpath("org.jetbrains.kotlin:kotlin-gradle-plugin")
    }
}
```

### Relevant log output

```shell
> Task :vision-camera-resize-plugin:compileDebugKotlin
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt:9:28 Unresolved reference: frameprocessor
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt:10:28 Unresolved reference: frameprocessor
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt:11:28 Unresolved reference: frameprocessor
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt:12:28 Unresolved reference: frameprocessor
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt:16:39 Unresolved reference: VisionCameraProxy
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt:16:60 Unresolved reference: FrameProcessorPlugin
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt:48:3 'callback' overrides nothing
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt:48:32 Unresolved reference: Frame
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt:159:12 Unresolved reference: SharedArray
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/VisionCameraResizePluginPackage.kt:8:28 Unresolved reference: frameprocessor
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/VisionCameraResizePluginPackage.kt:13:7 Unresolved reference: FrameProcessorPluginRegistry
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/VisionCameraResizePluginPackage.kt:13:72 Cannot infer a type for this parameter. Please specify it explicitly.
e: file:///home/expo/workingdir/build/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/VisionCameraResizePluginPackage.kt:13:79 Cannot infer a type for this parameter. Please specify it explicitly.
> Task :vision-camera-resize-plugin:compileDebugKotlin FAILED
> Task :expo-modules-core:compileDebugJavaWithJavac
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
> Task :app:mergeExtDexDebug
FAILURE: Build failed with an exception.
* What went wrong:
Execution failed for task ':vision-camera-resize-plugin:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details
* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.
BUILD FAILED in 6m 6s
```

### Camera Device

```json
N/A
```

### Device

N/A

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

AndyRC6, 2024-04-23:
Note: I realize I used the bug template when I should have used the build error template. I believe I included all the necessary information in my edit. Apologies.

fuekiin, 2024-04-23:
Facing the same problem

TheGhostFish, 2024-04-24:
It's just due to renaming the packages in vision camera version 4.
You can temporarily use this patch for resize plugin to solve the issue.

```
diff --git a/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt b/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt
index 8c8c0c5..16a440a 100644
--- a/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt
+++ b/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/ResizePlugin.kt
@@ -6,10 +6,10 @@ import android.util.Log
 import androidx.annotation.Keep
 import com.facebook.jni.HybridData
 import com.facebook.jni.annotations.DoNotStrip
-import com.mrousavy.camera.frameprocessor.Frame
-import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin
-import com.mrousavy.camera.frameprocessor.SharedArray
-import com.mrousavy.camera.frameprocessor.VisionCameraProxy
+import com.mrousavy.camera.frameprocessors.Frame
+import com.mrousavy.camera.frameprocessors.FrameProcessorPlugin
+import com.mrousavy.camera.frameprocessors.SharedArray
+import com.mrousavy.camera.frameprocessors.VisionCameraProxy
 import java.nio.ByteBuffer
 
 @Suppress("KotlinJniMissingFunction") // We're using fbjni
diff --git a/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/VisionCameraResizePluginPackage.kt b/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/VisionCameraResizePluginPackage.kt
index 7174bbe..a0147ab 100644
--- a/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/VisionCameraResizePluginPackage.kt
+++ b/node_modules/vision-camera-resize-plugin/android/src/main/java/com/visioncameraresizeplugin/VisionCameraResizePluginPackage.kt
@@ -5,7 +5,7 @@ import com.facebook.react.bridge.NativeModule
 import com.facebook.react.bridge.ReactApplicationContext
 import com.facebook.react.module.model.ReactModuleInfo
 import com.facebook.react.module.model.ReactModuleInfoProvider
-import com.mrousavy.camera.frameprocessor.FrameProcessorPluginRegistry
+import com.mrousavy.camera.frameprocessors.FrameProcessorPluginRegistry
 
 class VisionCameraResizePluginPackage : TurboReactPackage() {
   companion object {
```

rayronvictor, 2024-04-24:
Same problem using the same packages:

```js
//package.json relevant package versions
"react-native": "0.73.6",
"react-native-vision-camera": "^4.0.1",
"react-native-worklets-core": "^1.1.1",
"vision-camera-resize-plugin": "^2.1.1"
```

AndyRC6, 2024-04-24:
> It's just due to renaming the packages in vision camera version 4. You can temporarily use this patch for resize plugin to solve the issue.
>
> ```
> diff --git [...]

Thank you, applying this patch seems to have fixed the problem for now and allowed me to build successfully.

mrousavy, 2024-05-06:
fixed now

closed 2024-05-06 by mrousavy

mgulfam0722, 2025-04-14:
This issue arises in almost every plugin!!

```
e: file:///C:/Users/Musab/COPS-mobile-app/node_modules/vision-camera-ocr/android/src/main/java/com/visioncameraocr/OCRFrameProcessorPlugin.kt:16:28 Unresolved reference: frameprocessor
e: file:///C:/Users/Musab/COPS-mobile-app/node_modules/vision-camera-ocr/android/src/main/java/com/visioncameraocr/OCRFrameProcessorPlugin.kt:18:32 Unresolved reference: FrameProcessorPlugin
e: file:///C:/Users/Musab/COPS-mobile-app/node_modules/vision-camera-ocr/android/src/main/java/com/visioncameraocr/OCRFrameProcessorPlugin.kt:99:5 'callback' overrides nothing
e: file:///C:/Users/Musab/COPS-mobile-app/node_modules/vision-camera-ocr/android/src/main/java/com/visioncameraocr/VisionCameraOcrPackage.kt:7:28 Unresolved reference: frameprocessor
e: file:///C:/Users/Musab/COPS-mobile-app/node_modules/vision-camera-ocr/android/src/main/java/com/visioncameraocr/VisionCameraOcrPackage.kt:12:9 Unresolved reference: FrameProcessorPlugin
```