margelo/react-native-vision-camera #2790: 🐛 iOS Pause Output Incorrect

xHeinrich, 2024-04-24:
### What's happening?

When taking a video and pausing/resuming the video on iOS, the audio and video are chopped into different segments rather than one continuous video. For example, I start recording and say

one two three

pause recording and say four five six

resume recording and say seven eight nine

stop recording

Example video output:

https://github.com/mrousavy/react-native-vision-camera/assets/7674587/c50c74ab-23be-404e-a81f-22128227475a

### Reproduceable Code

```tsx
const camera = useRef<Camera>(null)

    let device = useCameraDevice('back', {
        physicalDevices: [
            'ultra-wide-angle-camera',
            'wide-angle-camera',
            'telephoto-camera'
        ]
    })
 
    const [targetFps, setTargetFps] = useState(30)

    let filters = [
        {fps: targetFps},
        {videoStabilizationMode: 'auto'},
        {
            videoResolution: {
                width: 1280,
                height: 720
            },
        },
        {
            photoResolution: {
                width: 1280,
                height: 720
            },
        }
    ];

    const format = useCameraFormat(device, filters)
    const [torchOn, setTorchOn] = useState('off')
    const onError = useCallback((error: any) => {
        console.error(error)
    }, [])
                      
                   <Camera device={device}
                                zoom={device.neutralZoom}
                                ref={camera}
                                format={format}
                                enableZoomGesture={true}
                                exposure={0}
                                style={StyleSheet.absoluteFill}
                                isActive={isActive}
                                torch={torchOn}
                                orientation={'portrait'}
                                audio={micPermissionState.hasPermission}
                                photo={permissionState.hasPermission}
                                video={permissionState.hasPermission}
                                onError={onError}
                        />

[... 114 lines cut ...]
 "position": "back",
 "hardwareLevel": "full",
 "hasTorch": true
}
```

### Device

iPhone 13 Pro

### VisionCamera Version

4.0.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

thanhtungkhtn, 2024-05-07:
Any update on this? facing similar issue

    "react": "18.2.0",
    "react-native": "0.74.1",
    "react-native-vision-camera": "^4.0.3"

xHeinrich, 2024-05-08:
Think it must be some timing issue with the asset writer but I don't know enough swift to find actual issue.

lee-byeoksan, 2024-05-31:
Found relevant PR but it's closed.
https://github.com/mrousavy/react-native-vision-camera/pull/1546

mrousavy, 2024-06-08:
Hey all! 

I just spent a few days on thinking about a battleproof timestamp synchronization solution, and I came up with a great idea.
I built a `TrackTimeline` helper class which represents a video or audio track - it can be started & stopped, paused & resumed, and even supports nesting pauses without issues.
- The total duration of the video is summed up from the difference between the first and the last actually written timestamps, minus the total duration of all pauses between a video. No more incorrect `video.duration`! 🥳 
- Whereas before I just had a 4 second timeout if no frames arrive, I now just wait twice the frame latency (a few milliseconds) to ensure no frames are left out at maximum! 🎉 
- A video can be stopped while it is paused without any issues, as a pause call is taken into consideration before stopping 💪 
- A video file's session now exactly starts at the start() timestamp, and ends at the exact timestamp of the last video frame - this ensures there can never be any blank frames in the video, even if the audio track is longer 🤩 

This was really complex to built as I had to synchronize timestamps between capture sessions, and the entire thing is a producer model - a video buffer can come like a second or so later than the audio buffer, but I need to make sure the video track starts before the audio track starts, and ends after the audio track ends - that's a huge brainf*ck! 🤯😅 

There's also no helper APIs for this on iOS, and it looks like **no other Camera framework** (not even native Swift/ObjC iOS Camera libraries) **support this** - they all break when timestamps have a delay (e.g. video stabilization enabled) (or dont even support delays at all) ; so I had to build the thing myself.

Check out this PR and try if it fixes the issue for you; https://github.com/mrousavy/react-native-vision-camera/pull/2948

Thanks! ❤️

closed 2024-06-10 by mrousavy