margelo/react-native-vision-camera #2791: 💭 Apple Privacy Info

ISnowFoxI, 2024-04-24:
### Question

Does `react-native-vision-camera` have the newly added `PrivacyInfo.xcprivacy` file or is it something we have to handle it ourselves?

### What I tried

Not sure what to try for this, as the API used by the library is not clear to me. 

### VisionCamera Version

v4.0.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2024-04-25:
I don't know.

ISnowFoxI, 2024-04-25:
RN v0.73.7 added a new `PrivacyInfo.xcprivacy` file to the ios directory. All libs and SDKs need to have privacy manifest as far as I can tell. For example, Sentry added one to their SDK. I think RNVC needs this kind of manifest file because it uses file system APIs and such.

Official Apple documentation is here: 
https://developer.apple.com/documentation/bundleresources/privacy_manifest_files

mrousavy, 2024-04-25:
I see, well if someone contributes / submits a PR for that we can take it from there

zzz08900, 2024-04-26:
Just took a look at Apple's doc of Privacy manifest files and I don't feel like RNVC needs that.
In a nut shell that manifest is for libs that collects data about the person using apps, which RNVC apparently is not capable of.

ISnowFoxI, 2024-04-29:
Aren't photo and video files from the camera considered user data? It is either that, or we have to specify the usage of it. It has to be either in the SDK or the project itself. It makes more sense to have it in the project itself as the usage could change from app to app.

ISnowFoxI, 2024-04-29:
Well kinda. I am not sure. We know `VC` can use location data too on top of video, photo and audio. This link describes those:
https://developer.apple.com/documentation/bundleresources/privacy_manifest_files/describing_data_use_in_privacy_manifests
[image]

mrousavy, 2024-04-29:
ah yea;

[image]

ISnowFoxI, 2024-04-29:
So if I want to make a PR, I only need to add this privacy file to `package/ios`, right? pretty new to the whole contribution stuff.

mrousavy, 2024-04-29:
I'm sorry I am not sure if this works, I have not worked with the privacy files at all. But if you tried this on your end by just adding that file to the react-native-vision-camera node modules package locally and it works, then yes, just add it here to `package/ios`, and also add it in the files section in `package.json` so it gets published to npm.

zzz08900, 2024-04-29:
> Well, `VC` doesn't collect any data without the app. This confuses me. Why can't we describe it in one place and be done with it?

Some SDKs contains proprietary codes and do shady stuff in the background without the developer of the App even knowing. 
Probably Apple want to tight that end too, so they can be clear who is there to blame.

zzz08900, 2024-04-29:
Maybe its a little late but from what I understand, you only need to declare that manifest if you are  trying to "collect" location data, which should mean read location data, store it on the phone or send it to a server for later processing. 

Simply write location data to photo/video's meta info is not considered as "collecting location data" as I see it.

Well I'm kinda stuck on V2 now so I'll try publish new versions to App Store without the privacy manifest and I'll let you guys know if Apple does not approve my new versions.

ISnowFoxI, 2024-04-29:
Thank you, but starting May 1st Apple will reject all apps that don't have privacy manifest files. That is why rn v0.73.7 adds this manifest file when you install pods

zzz08900, 2024-04-29:
> Thank you, but starting May 1st Apple will reject all apps that don't have privacy manifest files. 

Yeah sure I'll update my app to include that manifest, along with other "suspicious" libs including RNFS and react-native-device-info. 

I ran a scan of RNVC V2 using https://github.com/Wooder/ios_17_required_reason_api_scanner
It didn't find anything.

Now I have all declaration of use of required reason APIs (at least I covered everything mentioned by Apple in the mail) in one of my apps. 

Sad we have to wait until May 1st to see how this goes.

ISnowFoxI, 2024-05-02:
Quick update. We sent an update to App Store Connect and it did get approved. Not sure if this means no manifest file is required or if they just haven't caught up. I'll try to look into a bit more this weekend if I can.

chunghe, 2024-05-03:
We're using expo(v50.0.17) + react-native-vision-camera(v4.0.1), here's our result:

1. do nothing: after upload to app connect, will receive an email about "NSLocationWhenInUseUsageDescription" privacy not explaining clearly
2. add `enableLocation: false` in app.config.js, also got the warning email
3. use `yarn patch` to modify `VisionCamera.podspec` directly, not receiving any warning email. Hope that is helpful.

patch file
```
diff --git a/VisionCamera.podspec b/VisionCamera.podspec
index 23ddf8658abd2526e9c8f99fa511cb94e3646bf4..c41aa49808544026a62c20efb993b7e735465045 100644
--- a/VisionCamera.podspec
+++ b/VisionCamera.podspec
@@ -6,7 +6,7 @@ nodeModules = File.join(File.dirname(`cd "#{Pod::Config.instance.installation_ro

 Pod::UI.puts "[VisionCamera] Thank you for using VisionCamera ❤️"

-enableLocation = true
+enableLocation = false
 if defined?($VCEnableLocation)
   Pod::UI.puts "[VisionCamera] $VCEnableLocation is set to #{$VCEnableLocation}!"
   enableLocation = $VCEnableLocation
@@ -14,7 +14,7 @@ else
   Pod::UI.puts "[VisionCamera] $VCEnableLocation is not set, enabling CLLocation APIs by default..."
 end

-enableFrameProcessors = true
+enableFrameProcessors = false
 if defined?($VCEnableFrameProcessors)
   Pod::UI.puts "[VisionCamera] $VCEnableFrameProcessors is set to #{$VCEnableFrameProcessors}!"
   enableFrameProcessors = $VCEnableFrameProcessors
```

mrousavy, 2024-05-03:
The change in Podfile has nothing to do with `NSLocationWhenInUseUsageDescription`, so this is just coincidence.

`NSLocationWhenInUseUsageDescription` is set by your app in `Info.plist`. Maybe you just didn't put an explanation in there.

closed 2024-05-03 by mrousavy

edmbn, 2024-05-05:
I'm also seeing the same issue, receiving apple `NSLocationWhenInUseUsageDescription` everytime even if I have `"enableLocation": false`. The thing is, I even tried to have have `"enableLocation": false` in my `app.config.js` and also <Camera enableLocation={true} /> and it is still working. Clearly something in app.config.js plugin configuration is not working.

mrousavy, 2024-05-05:
Yes. It was using the wrong name.
If you used TypeScript you should've noticed the issue on your end, but apparently not everyone uses TS.

It's fixed on main now.

edmbn, 2024-05-05:
I'm using `app.config.ts`, using version 4.0.1 and when I type "enable" it isn't showing the hint or error for "enableLocation", thats why I was trying to follow blindly the docs. Any Idea of why typescript cannot detect this on my end?

edmbn, 2024-05-05:
@mrousavy I understand that typescript works if you use "withVisionCamera" plugin. But if you add the plugin through Expo via:

```
{
  "name": "my app",
  "plugins": [
    [
      "react-native-vision-camera",
      {
        // ...
        "enableLocation": false,
      }
    ]
  ]
}
```
There aren't typescript hint or error helpers, right? I know this isn't the issue to talk about this. I would be happy to open a new issue or something else in order to help. I just want to help because if people use 4.0.1 from NPM and follows the documentation will face the same issue since they are out of sync. Current documentation is for the main branch not for the release on NPM, right? If I can do anything else to help just tell me. Thank you for your time.

mrousavy, 2024-05-06:
Ah gotcha yea through JSON it might not work. Maybe app.config.js? I don't know.

Either way will release an update soon