margelo/react-native-vision-camera #2792: ✨ Greyscale, B&W and more filters

DEMONCELL1, 2024-04-24:
### What feature or enhancement are you suggesting?

I think we can use the Filter property in the camera for getting videos with simple filters like Greyscale, B&W, polaroid etc.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

I am currently trying to do this with the Skia but I am not able to achieve a certain result.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2024-04-25:
You can add such filters by using Skia Frame Processors. Read the docs

closed 2024-04-25 by mrousavy

DEMONCELL1, 2024-04-29:
Can you help me with some example if possible?

RutvinBhanderi, 2024-05-21:
> You can add such filters by using Skia Frame Processors. Read the docs

But when we capture the photo then it is not taking that effect. Can you help me with that?

mrousavy, 2024-05-21:
See this section in the Skia FP docs.

[image]