margelo/react-native-vision-camera #2795: 💭 Any Frame differences for Android from v2 to v4

eledahl, 2024-04-24:
### Question

I'm running a pose detector that is lining up fine for drawing the pose on screen with SVG on v2. I'm attempting to upgrade to v4. The pose draws fine with SVG in v4 on iOS but Android seems to be behaving much different. Pretty hard to get it to line up with the body. 

With the change to CameraX/Camera2 for Android my guess is that the frame is coming through differently and I need to process it in my plugin a different way. Is there any info on changes to the frame for Android or am I way off with that assumption?

I can post the drawing code if it helps but I'm more just trying to locate any differences in what is being used since my code worked on v2

### What I tried

I've tried Skia but it slows the camera down like crazy when using useSkiaFrameProcessor and when I render Skia as a canvas the points lag like crazy. They weren't aligned either though. SVG seems to be best performance wise on Android.

### VisionCamera Version

4.0.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22💭+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

eledahl, 2024-04-24:
Looks like printing out frame.height and frame.width in v2 gives me 3024 x 3024 but format is  720 x 1280

On v4, the frame matches the format size and prints 720 x 1280

eledahl, 2024-04-24:
Bumped it up to 4.0.1 and same issue.

I ended up switching it to a 4:3 format to see if it would help on android and nada. When I console log frame.orientation, I get landscape-left and the format does have a higher width at 640 vs height at 480. But we have it set as orientation="portrait"

``` 
<Camera
            ref={camera}
            format={format}
            style={StyleSheet.absoluteFill}
            device={device}
            fps={30}
            hdr={enableHdr}
            lowLightBoost={device.supportsLowLightBoost && enableNightMode}
            isActive={isActive}
            onInitialized={onInitialized}
            onError={onError}
            enableZoomGesture={false}
            photo={false}
            video={true}
            audio={false}
            frameProcessor={frameProcessor}
            orientation="portrait"
          />
```

mrousavy, 2024-04-25:
Use a format with video resolution 3024x3024 then.

eledahl, 2024-04-25:
@mrousavy Yeah no 3024 x 3024 format available or any 1:1 formats. In v2, I was using a 16:9 format (1280 x 720) and it gave me the 1:1 frame (3024 x 3024). So was more curious if you knew why it did that in v2 but the frame changes in v4 to be 16:9 and matches the format. But I'll look into it

mrousavy, 2024-04-25:
Can you log all formats of device.formats?

eledahl, 2024-04-26:
Sure he is the the format chosen in v2:

```
{"autoFocusSystem": "none", "colorSpaces": ["jpeg"], "fieldOfView": 66.55955142399227, "frameRateRanges": [{"maxFrameRate": 30, "minFrameRate": 1}, {"maxFrameRate": 15, "minFrameRate": 15}, {"maxFrameRate": 24, "minFrameRate": 7}, {"maxFrameRate": 24, "minFrameRate": 24}, {"maxFrameRate": 30, "minFrameRate": 8}, {"maxFrameRate": 30, "minFrameRate": 30}], "isHighestPhotoQualitySupported": false, "maxISO": 3200, "maxZoom": 8, "minISO": 50, "photoHeight": 910, "photoWidth": 1920, "pixelFormat": "420v", "supportsPhotoHDR": true, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": ["off", "auto", "standard", "cinematic"], "videoWidth": 1280}
```

Here's the format chosen for v4:

```
{"autoFocusSystem": "contrast-detection", "fieldOfView": 78.7353670985857, "maxFps": 30, "maxISO": 3200, "minFps": 7, "minISO": 50, "photoHeight": 3024, "photoWidth": 4032, "supportsDepthCapture": false, "supportsPhotoHdr": false, "supportsVideoHdr": false, "videoHeight": 720, "videoStabilizationModes": ["off", "cinematic"], "videoWidth": 1280}
```

And here are all the formats for v4:

```
[
    {
        "autoFocusSystem": "contrast-detection",
        "fieldOfView": 78.7353670985857,
        "maxFps": 30,
        "maxISO": 3200,
        "minFps": 7,
        "minISO": 50,
        "photoHeight": 3024,
        "photoWidth": 4032,
        "supportsDepthCapture": false,
        "supportsPhotoHdr": true,
        "supportsVideoHdr": false,
        "videoHeight": 480,
        "videoStabilizationModes": [
            "off",
            "cinematic"
        ],
        "videoWidth": 720
    },
    {
        "autoFocusSystem": "contrast-detection",
        "fieldOfView": 78.7353670985857,
        "maxFps": 30,
        "maxISO": 3200,
        "minFps": 7,
        "minISO": 50,
        "photoHeight": 2268,
        "photoWidth": 4032,
        "supportsDepthCapture": false,
        "supportsPhotoHdr": true,
        "supportsVideoHdr": false,
        "videoHeight": 480,
        "videoStabilizationModes": [
            "off",
            "cinematic"
        ],
        "videoWidth": 720
    },
    {
        "autoFocusSystem": "contrast-detection",
        "fieldOfView": 78.7353670985857,
        "maxFps": 30,
        "maxISO": 3200,
        "minFps": 7,
        "minISO": 50,
        "photoHeight": 1908,
        "photoWidth": 4032,
        "supportsDepthCapture": false,
        "supportsPhotoHdr": true,
        "supportsVideoHdr": false,
        "videoHeight": 480,
        "videoStabilizationModes": [
            "off",
            "cinematic"
[... 1728 lines cut ...]
            "cinematic"
        ],
        "videoWidth": 3840
    },
    {
        "autoFocusSystem": "contrast-detection",
        "fieldOfView": 78.7353670985857,
        "maxFps": 30,
        "maxISO": 3200,
        "minFps": 7,
        "minISO": 50,
        "photoHeight": 144,
        "photoWidth": 176,
        "supportsDepthCapture": false,
        "supportsPhotoHdr": true,
        "supportsVideoHdr": false,
        "videoHeight": 2160,
        "videoStabilizationModes": [
            "off",
            "cinematic"
        ],
        "videoWidth": 3840
    }
]
```

mrousavy, 2024-04-29:
Yea so the formats definitely might have changed a bit- previously VisionCamera also included formats that aren't necessarily compatible with the MediaCodec, causing an error sometimes when trying to record. Now visionCamera only includes the safe profiles, so in your case just use 

```json
    {
        "autoFocusSystem": "contrast-detection",
        "fieldOfView": 78.7353670985857,
        "maxFps": 30,
        "maxISO": 3200,
        "minFps": 7,
        "minISO": 50,
        "photoHeight": 144,
        "photoWidth": 176,
        "supportsDepthCapture": false,
        "supportsPhotoHdr": true,
        "supportsVideoHdr": false,
        "videoHeight": 2160,
        "videoStabilizationModes": [
            "off",
            "cinematic"
        ],
        "videoWidth": 3840
    }
```
There is no 1x1 format from a Camera. 4:3 and 16:9 are the most adopted ones.