margelo/react-native-vision-camera #2797: 🐛 4! FFPS!???, Frame Processor, Synchronous

R4tch3t, 2024-04-24:
### What's happening?

This instead of improving or optimizing it comes out worse, why frame processor in synchronous run at 4 FPS when in previously versions run at full speed in the same phone and configuration, I used useState I don't chance for change to useSharedValue maybe is for that, but in first person it's a radical change and showing like a bug.
[image]

### Reproduceable Code

```tsx
let frameProcessor = useFrameProcessor((frame) => {
    'worklet'    
    
      //runAsync(frame,()=>{
        //'worklet'
          try{
            
            const values: any = textRecognizerPlugin(frame,{phoneWidth,phoneHeight,isLandFrame});                  
            
            if(values?.TextRecognized!){
                values.width=parseFloat(values.width+"")
                values.height=parseFloat(values.height+"")
                values.x=parseFloat(values.x+"")
                values.y=parseFloat(values.y+"")
                values.top=parseFloat(values.top+"")
                values.left=parseFloat(values.left+"")               
                
                const {top, left} = values
                setBoundsJS({origin: {top, bottom:0, left, right:0}, size: {width: values.width, height: values.height}})
                setEvalTextJS(values.TextRecognized)
                
              }else{
                setBoundsJS({origin: {top: 0, left: 0, right: 0, bottom: 0,  }, size: {width: 0, height: 0}})
                setEvalTextJS(null)     
              }

            }catch(e){
              console.log("ErrorFrame: ",e)
            }
      //})
  }, [phoneWidth, phoneHeight, isLandFrame])
```

### Relevant log output

```shell
none
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 6,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -6,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

motorola edge 20

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

gbark, 2024-04-24:
Show some respect for the people maintaining and developing this amazing library that you can you for **free**.

R4tch3t, 2024-04-24:
> Show some respect for the people maintaining and developing this amazing library that you can you use for **free**.

All framework and languages are free what's the point of your comment? I sponsored, and only opening thread because maybe not I am with the same situation...

gbark, 2024-04-24:
Don't think I can make my point much clearer. Not sure the same could be said for you though...

R4tch3t, 2024-04-25:
> Don't think I can make my point much clearer. Not sure the same could be said for you though...

Nope, I'm here to clarify this and see if it's a problem or not, and you just show up to throw hate.

mrousavy, 2024-04-25:
> I sponsored

With which account? I don't see a sponsorship under this account.

> All framework and languages are free

One of the most naive statements I've heard so far 👍
You pay $100/year for Apple Membership, $4k+ for MacBook and $1k+ for an iPhone. Do you really think Apple's frameworks are "free"?

> Relevant log output
> 
> `none`

Well, no log output means I can't help you. :)

closed 2024-04-25 by mrousavy

R4tch3t, 2024-04-25:
ok, thanks...