margelo/react-native-vision-camera #2798: This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change🔧

a289459798, 2024-04-25:
### How were you trying to build the app?

android build

### Full build logs

```tsx
e: file:///Users/zhangzy/workspace/android/5ichong/ichong_buy_new/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:31:79 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///Users/zhangzy/workspace/android/5ichong/ichong_buy_new/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:36:19 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///Users/zhangzy/workspace/android/5ichong/ichong_buy_new/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:36:47 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
e: file:///Users/zhangzy/workspace/android/5ichong/ichong_buy_new/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/frameprocessors/VisionCameraProxy.kt:75:73 This API is provided only for React Native frameworks and not intended for general users. This API can change between minor versions in alignment with React Native frameworks and won't be considered a breaking change.
```

### Project dependencies

```json
"dependencies": {
    "react": "18.2.0",
    "react-native": "0.74.0",
    "react-native-vision-camera": "^4.0.1"
  }
```

### VisionCamera Version

4.0.1

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

No, I cannot build the Example app either

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-25:
see #2614. RN 0.74 does not work yet I need to migrate to new arch. support the efforts by sponsoring me if you want.

closed 2024-04-25 by mrousavy

GeekyCoder25, 2024-05-22:
I'm also experiencing this error while trying to build the app both locally and on eas, I'm on react Native v0.74.1 and react-native-vision-camera v4.0.5
[image]