margelo/react-native-vision-camera #2801: [V4][Android] - Camera freeze after a few .takePhoto() calls 🐛

ldalzottomp, 2024-04-25:
### What's happening?

Hello!

In my recent try to update to the latest v4.0.1.
My devices encounter a freeze issue after taking a few photos.
 
Note: I am waiting for the photo captured to be completed every time.
Note2: I don't reproduce in the example app. But I suspect that the session is restarted between every photo (as I hear my camera clicking every time).

### Reproduceable Code

```tsx
https://github.com/ldalzottomp/rnvc-beta-test/blob/main/App.tsx
```

### Relevant log output

```shell
import { useEffect, useRef, useState } from 'react';
import { Button, Image, ScrollView, StyleSheet, Text, View } from 'react-native';
import { Camera, Orientation, useCameraDevice } from 'react-native-vision-camera';

type PhotoModel = {
    uri: string;
    visionCameraWidth: number;
    visionCameraHeight: number;
    imageWidth: number;
    imageheight: number;
};

export default App = () => {
    const [init, setInit] = useState(false);
    useEffect(() => {
        (async () => {
            const newCameraPermission = await Camera.requestCameraPermission();
            setInit(true);
        })();
    }, []);

    const device = useCameraDevice('back');
    const cameraRef = useRef<Camera>(null);

    const [orientation, setOrientation] = useState<Orientation>('portrait');
    const [images, setImages] = useState<PhotoModel[]>([]);

    if (!init) {
        return <></>;
    }
    if (device == null) return <></>;
    return (
        <View style={StyleSheet.absoluteFill}>
            <Camera
                ref={cameraRef}
                style={StyleSheet.absoluteFill}
                device={device}
                isActive={true}
                photo={true}
                orientation={orientation}
            />
            <Button
                title="Take photo"
                onPress={async () => {
                    const photoResult = await cameraRef.current?.takePhoto({});
                    if (photoResult) {
                        Image.getSize('file:///' + photoResult?.path, (w, h) => {
                            setImages((prev) => {
                                return [
                                    ...prev,
                                    {
                                        uri: 'file:///' + photoResult.path,
                                        visionCameraWidth: photoResult.width,
                                        visionCameraHeight: photoResult.height,
                                        imageWidth: w,
                                        imageheight: h,
                                    },
                                ];
                            });
                        });
                    }
                }}
            />
            <Button
                title="Focus"
                onPress={async () => {
                    const photoResult = await cameraRef.current?.focus({ x: 0, y: 0 });
                }}
            />
            <View style={{ backgroundColor: 'red', height: 30 }}>
                <View style={{ flex: 1, flexDirection: 'row' }}>
                    <Button
                        title="LandLeft"
                        onPress={() => {
                            setOrientation('landscape-left');
                        }}
                    />
                    <Button
                        title="LandRight"
                        onPress={() => {
                            setOrientation('landscape-right');
                        }}
                    />
                    <Button
                        title="Port"
                        onPress={() => {
                            setOrientation('portrait');
                        }}
                    />
                    <Button
                        title="PortDown"
                        onPress={() => {
                            setOrientation('portrait-upside-down');
                        }}
                    />
                </View>
            </View>
            <ScrollView
                style={{
                    position: 'absolute',
                    bottom: 0,
                    left: 0,
                    right: 0,
                    height: 200,
                    backgroundColor: 'black',
                }}
                horizontal={true}
            >
                {images.map((image, index) => {
                    return (
                        <View key={index} style={{ flex: 1 }}>
                            <Image
                                style={{ flex: 1 }}
                                source={{
                                    uri: image.uri,
                                }}
                                height={200}
                                width={200}
                            />

                            <Text
                                style={{
                                    position: 'absolute',
                                    bottom: 0,
                                    left: 0,
                                    right: 0,
                                    height: 20,
                                    color: 'red',
                                }}
                            >
                                VC: {image.visionCameraWidth}x{image.visionCameraHeight}. RN:{' '}
                                {image.imageWidth}x{image.imageheight}
                            </Text>
                        </View>
                    );
                })}
            </ScrollView>
        </View>
    );
};
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Galaxy S9 - SM-G960F (Android 10, One UI 2.5)

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ldalzottomp, 2024-05-23:
I am not reproducing this issue on all my testing android devices.
Thanks for the fix. I think that I can close this issue.

closed 2024-05-23 by ldalzottomp

mrousavy, 2024-05-27:
Awesome!