margelo/react-native-vision-camera #2802: [V4][Android] Orientation prop landscape-left and landscape-right seems to produce inverted images 🐛

ldalzottomp, 2024-04-25:
### What's happening?

Hello!

In my recent try to update to the latest v4.0.1.
My device have the captured image flipped for landscape-left and landscape-right orientation modes.
 
Note: I am not rotating the UI, the UI is locked in portrait. The four buttons above my app change the orientation props.
Note2: I don't reproduce in the example app. Because the orientation prop is fixed to portrait.

### Reproduceable Code

```tsx
TODO
```

### Relevant log output

```shell
TODO
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Galaxy S9 - SM-G960F (Android 10, One UI 2.5)

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jkaufman, 2024-04-25:
I am seeing the same behavior. Going from v2 to v4, I had to reverse `landscape-left` and `landscape-right` orientations on Android to yield the correct output.

mrousavy, 2024-05-27:
Yea, maybe I messed something up. Landscape left means the home-button is on the left side, which clearly is correct in this video - but the image is inverted... I am probably doing something wrong here then.

closed 2024-05-30 by mrousavy