margelo/react-native-vision-camera #2803: [V4][Android] - The returned captured image dimensions is different from the output of `Image.getSize` 🐛

ldalzottomp, 2024-04-25:
### What's happening?

Hello!

In my recent try to update to the latest v4.0.1.
I am encountering a weird issue when the orientation prop is set to landscape-left.
The image dimension produced by camera.takePhoto() doesn't match the output of the react-native api Image.getSize. I suspect that wrong metadata are set in this case?
All ohter values for the orientation props are okay.
 
You can check the following video that showcase the issue. Writtent in red : 
* VC : Is the output of the camera.takePhoto
* RN : Is the output of the Image.getSize

Note: I reproduce in the example app if I change the orientation prop and console log the Image.getSize output.

### Reproduceable Code

```tsx
TODO
```

### Relevant log output

```shell
TODO
```

### Camera Device

```json
{
  "formats": [],
  "sensorOrientation": "landscape-left",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "maxExposure": 20,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "minFocusDistance": 10,
  "minExposure": -20,
  "name": "0 (BACK) androidx.camera.camera2",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

Galaxy S9 - SM-G960F (Android 10, One UI 2.5)

### VisionCamera Version

4.0.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2024-04-29:
I use `BitmapFactory.decodeFile` to get the width and height of the photo after capturing.

According to [`BitmapFactory.Options.outHeight`](https://developer.android.com/reference/android/graphics/BitmapFactory.Options.html#outHeight);

> The resulting height of the bitmap. If [inJustDecodeBounds](https://developer.android.com/reference/android/graphics/BitmapFactory.Options#inJustDecodeBounds) is set to false, this will be height of the output bitmap after any scaling is applied. If true, it will be the height of the input image without any accounting for scaling.

We set `inJustDecodeBounds` to `true`, so this should be "the height of the input image without any accounting for scaling." - does `Image.getSize` maybe perform display scaling already?

ldalzottomp, 2024-04-30:
@mrousavy I tried to debug both source code and didn't any valid conclusion.
I noted indeed that when react load the Image there is a resize done only for the LandscapeLeft picture. (I lost the place in the source code). But why would it happen only for this orientation ? Other three looks fine. It has to be something else I feel.

The same app login in V3 works fine.

I noted that when react load the exif to get the orientation, I have the following results. 

`
public static final int ORIENTATION_NORMAL = 1;
public static final int ORIENTATION_ROTATE_180 = 3;
public static final int ORIENTATION_ROTATE_270 = 8;
public static final int ORIENTATION_ROTATE_90 = 6;

ORIENTATION_ROTATE_90 -> Portrait
ORIENTATION_ROTATE_NORMAL -> LandscapeLeft
ORIENTATION_ROTATE_180 -> LandscapeRight
ORIENTATION_ROTATE_270 -> PortraitUpsideDown
`
Which is different from V3 : 
`
ORIENTATION_ROTATE_180 -> LandscapeLeft
ORIENTATION_ROTATE_NORMAL -> LandscapeRight

`

mrousavy, 2024-04-30:
hmm, interesting. Orientation is still something that's not fully supported (See #1891) - so I don't really have a lot of insights, other than that I'd be confident that CameraX' EXIF writer is probably a bit more battletested than the one I wrote in V3. So V4 _should_ be correct, anything that's incorrect should probably be reported in the Google Issue Tracker, they test on a lot of devices. I only have 3 phones.

ldalzottomp, 2024-05-23:
Okay, I will dig more into it and check with a sample CameraX native app.
The root cause of this issue may be the same as for https://github.com/mrousavy/react-native-vision-camera/issues/2802

ldalzottomp, 2024-05-24:
I even feel that it's a bug in react-native, because loading the image without even using the camera cause the issue.
I looked for multiple issues on github of people having this issue.
I also realised that I had this issue even before V4.
So, closing the issue. Thanks for your time on this one!

closed 2024-05-24 by ldalzottomp